moveit/moveit_task_constructor #442: Double free or corruption (out)

leander2189, 2023-03-16:
### Environment:
Both ROS2 Galactic & Humble

### Error description:
When planning, the node crashes and returns this error message:
```bash
double free or corruption (out)
```

Tracking the exception leads to the function ConnectingPrivate::compute()
```cpp
void ConnectingPrivate::compute() {
	const StatePair& top = pending.pop(); // This line throws the exception, as pending has 0 elements
	const InterfaceState& from = *top.first;
	const InterfaceState& to = *top.second;
	assert(from.priority().enabled() && to.priority().enabled());
	static_cast<Connecting*>(me_)->compute(from, to);
}
```

I've been able to work around this by simply adding this line
```cpp
if (pending.size() == 0) return;
```

My question is: 
**Is this a bug in MTC or this error is generated by a bad definition of the stages?**

### Additional info
The stages are defined like this:
- Current State
- Parallel container (Alternatives)
    - Connect
    - Serial Container
        -  MoveTo
        - Connect
- Generator pose

rhaschke, 2023-03-17:
Connect wasn't designed and tested for parallel containers. So, there might be a bug in MTC. Could you please provide a minimal example to be used as a new unit test? Thanks.

leander2189, 2023-03-20:
I am working on this minimal example, but so far I can't reproduce the error. I'll keep debugging this situation

rhaschke, 2023-05-04:
Can we close this?

leander2189, 2023-05-05:
Yes, I wasn't able to create a minimal example that reproduces this error, so probably better to close it for now

closed 2023-05-05 by leander2189