moveit/moveit_task_constructor #446: Combine MTC with core MoveIt packages?

schornakj, 2023-03-20:
I think it would make sense to merge the MoveIt Task Constructor packages in this repo with the core MoveIt packages in http://github.com/ros-planning/moveit and http://github.com/ros-planning/moveit2, so I wanted to check with other MTC developers and users (especially @rhaschke  and @v4hn ) to see what other folks thought about this.

Based on the structure of the C++ namespaces and the inclusion of some MTC-specific tutorials in the core MoveIt docs, I think this consolidation of packages was the original intent of the MTC authors.

Other reasons why I think this is a good idea:
- Lets us support MTC as a first-class feature of MoveIt instead of a 3rd-party add-on.
- Improves visibility for MTC issues and PRs.
- Makes it easier to keep MTC up to date with new MoveIt features.
- Conversely, makes it easier to keep MoveIt up to date with features added to improve MTC.

rhaschke, 2023-03-20:
I am, in principle, open for such a move, but I don't agree to the list of benefits. For example, regarding the visibility of issues and PRs, I prefer a standalone repo, because it's much easier to keep track of them here. Joining the two repos would make sense if we have strong interdependencies with each other, which I didn't yet face in the past.

AndyZe, 2023-03-20:
I'm moderately opposed to the move because I think it's too early to declare MTC as the right solution for task-planning-like things (I know it's not really a task planner). There may be other options emerging soon.