moveit/moveit_task_constructor #453: rotation around given point as task

TheUncivil, 2023-04-21:
How can I define a rotation around a given point as a stage of a task?

I'm currently attempting to configure a robot arm to pick apples, using the moveit task constructor. I'm defining the necessary movements as a task.

I want to achieve a specific picking motion, where the apple is rotated around the point of connection between the stem and the branch as shown below. I am able to determine the center of the rotation already.

However, I'm having trouble defining a rotation around a given point as a stage of my picking task. I though a MoveRelative task would work, but it only seems to perform translations in a straight line. It can do rotations at the same time, but that's not quite what I'm looking for.

[image]

rhaschke, 2023-04-21:
MoveRelative is your friend. It allows for performing an arbitrary twist motion. Just use the rotation velocity and zero the linear velocity part. It will be key to set the IK frame into the center of rotation!

TheUncivil, 2023-04-24:
I've set the IK frame to the point I want to rotate around, but now the task fails to plan, every time. The planner accepts the task, but fails to plan and aborts it. I've tried to execute this with CHOMP, OMPL, and the Pilz industrial motion planner, but none of these seem to make a difference.

TheUncivil, 2023-04-24:
I'm not sure how to set ``solvers::CartesianPath`` when defining my task, but I've enabled the "use cartesian path" option in the RVIZ interface. This hasn't solved my issue, however.

I don't think a collision could be causing this issue, considering the only objects in my planning scene at the moment are the UR robot arm and a single sphere representing an apple.

rhaschke, 2023-04-24:
> I'm not sure how to set `solvers::CartesianPath` when defining my task.

Just use this class instead of `PipelinePlanner`.

> I've enabled the "use cartesian path" option in the RVIZ interface.

This will only effect plans triggered from rviz.

> Could it be a problem to rotate around a frame that (only) is defined as a TF frame?

Yes, that's definitely a problem. MoveIt and MTC don't know about TF frames by default.

TheUncivil, 2023-04-24:
> > Could it be a problem to rotate around a frame that (only) is defined as a TF frame?
> 
> Yes, that's definitely a problem. MoveIt and MTC don't know about TF frames by default.

Gotcha, I missed that. I'll change my implementation and see if that fixes it.

TheUncivil, 2023-04-26:
When I set my IK frame to a subframe, the task is rejcted. I can set the frame of my twist to a subframe and execute it, but that doesn't work for relative movements.
When I plan a task that involves a subframe, I get the following error: 
`
[moveit_transforms.transforms]: Unable to transform from frame 'stem' to frame 'world'. Returning identity.
`
Do you know what might be causing this?

At the moment, I'm using targets loaded in from a .scene file that looks as follows:
```
(noname)+
* apple_0
-0.236968 -0.874544 0.419444
0 0 0 1
2
sphere
0.0568352
0 0 0
0 0 0 1
0.32887 0.115246 0.0434078 1
cylinder
0.011367 0.125037
0.011367 0 0
0 -0.707107 0 0.707107
0.32887 0.115246 0.0434078 1
2
crown
-0.0568352 0 0
0 0 0 1
stem
0.125037 0 0
0 0 0 1
* apple_1
-0.845401 -0.671953 0.47212
0 0 0 1
2
sphere
0.0605912
0 0 0
0 0 0 1
0.328593 0.107315 0.0458313 1
cylinder
0.0121182 0.133301
0.0121182 0 0
0 -0.707107 0 0.707107
0.328593 0.107315 0.0458313 1
2
crown
-0.0605912 0 0
0 0 0 1
stem
0.133301 0 0
0 0 0 1
.
```

rhaschke, 2023-04-26:
Due to a limitation in MoveIt, possible IK frames are limited currently. A PR to fix that is pending (#3388).
Here is some simple code to realize your intention: 
```python
from std_msgs.msg import Header
from geometry_msgs.msg import TwistStamped, Twist, Vector3, PoseStamped, Pose, Point
from moveit.task_constructor import core, stages
from math import pi
import time

from moveit_commander.roscpp_initializer import roscpp_initialize

roscpp_initialize("mtc_tutorial")

cartesian = core.CartesianPath()

task = core.Task()

task.add(stages.CurrentState("current state"))
move = stages.MoveRelative("rotate", cartesian)
move.group = "panda_manipulator"
header = Header(frame_id="panda_link0")
move.setDirection(TwistStamped(header=header, twist = Twist(angular=Vector3(0, 0, pi / 2.0))))
header.frame_id = "panda_hand_tcp"
move.ik_frame = PoseStamped(header=header, pose=Pose(position=Point(0.2, 0, 0)))
task.add(move)

if task.plan():
    task.publish(task.solutions[0])
time.sleep(100)
```

closed 2023-04-26 by TheUncivil