moveit/moveit_task_constructor #454: Approach picking target from different angles

TheUncivil, 2023-05-01:
I'm currently using the Connect task stage to move a robot arm into position to pick an object (a sphere in my case). I've noticed that MoveIt invariably approaches the sphere along its equator (along the ). Is it possible to consider alternate angles of approach, or to specify one that is not on the equator?

rhaschke, 2023-05-01:
You can customize all motions to your liking. Just introduce (or modify) the corresponding MoveRelative stage.
The Connect stage shouldn't be responsible for the approach motion.

TheUncivil, 2023-05-02:
I had understood that MoveRelative is used to plan movements relative to a fixed point or frame, not to move the end effector to a specific point.

To clarify, when I say 'approach', I mean move the end effector from any arbitrary position to a position where it can begin to pick its target. The attached video should hopefully convey my meaning clearly.

[pick_approach.webm](https://user-images.githubusercontent.com/46045333/235592754-547c2a1f-c042-4a45-a1a7-845b8fcc2f0f.webm)

rhaschke, 2023-05-02:
Sorry, but the video doesn't convey anything. There are multiple spheres (to be picked?). None of them is approached.

Have a look at the Pick+Place tutorial: There, the target object is approached along the hand's z-axis, planning backward  starting from the final pick pose:
https://github.com/ros-planning/moveit_task_constructor/blob/4781ed563657cb7ecfcca857649bdf9f1c8d95b9/demo/src/pick_place_task.cpp#L236-L252

TheUncivil, 2023-05-02:
I figured out what I was missing. As you say, the approach is planned backward from the pick pose; that's key to what I want to do. Thank you for your help!

closed 2023-05-02 by TheUncivil