moveit/moveit_task_constructor #455: moveit_task_constrctor failed when i colcon build

galversano, 2023-05-03:
hello, i follow the tutorial for pick and place, when i build moveit_task_constructor i get a lot of errors (moveit_core)

uctor/merge.h:42:10: fatal error: moveit/trajectory_processing/time_parameterization.h: No such file or directory
   42 | #include <moveit/trajectory_processing/time_parameterization.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:115: src/CMakeFiles/moveit_task_constructor_core.dir/merge.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/gal/ws_moveit2/src/moveit_task_constructor/core/include/moveit/task_constructor/task_p.h:42,
                 from /home/gal/ws_moveit2/src/moveit_task_constructor/core/src/introspection.cpp:38:
/home/gal/ws_moveit2/src/moveit_task_constructor/core/include/moveit/task_constructor/task.h:50:10: fatal error: moveit/utils/moveit_error_code.h: No such file or directory
   50 | #include <moveit/utils/moveit_error_code.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:89: src/CMakeFiles/moveit_task_constructor_core.dir/introspection.cpp.o] Error 1
In file included from /home/gal/ws_moveit2/src/moveit_task_constructor/core/src/container.cpp:39:
/home/gal/ws_moveit2/src/moveit_task_constructor/core/include/moveit/task_constructor/merge.h:42:10: fatal error: moveit/trajectory_processing/time_parameterization.h: No such file or directory
   42 | #include <moveit/trajectory_processing/time_parameterization.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:63: src/CMakeFiles/moveit_task_constructor_core.dir/container.cpp.o] Error 1
/home/gal/ws_moveit2/src/moveit_task_constructor/core/src/stage.cpp: In member function ‘virtual bool moveit::task_constructor::Connecting::compatible(const moveit::task_constructor::InterfaceState&, const moveit::task_constructor::InterfaceState&) const’:
/home/gal/ws_moveit2/src/moveit_task_constructor/core/src/stage.cpp:905:22: error: ‘using element_type = struct collision_detection::World::Object’ {aka ‘struct collision_detection::World::Object’} has no member named ‘pose_’
  905 |   if (!(from_object->pose_.matrix() - to_object->pose_.matrix()).isZero(1e-4)) {
      |                      ^~~~~
/home/gal/ws_moveit2/src/moveit_task_constructor/core/src/stage.cpp:905:50: error: ‘using element_type = const struct collision_detection::World::Object’ {aka ‘const struct collision_detection::World::Object’} has no member named ‘pose_’
  905 |   if (!(from_object->pose_.matrix() - to_object->pose_.matrix()).isZero(1e-4)) {
      |                                                  ^~~~~
/home/gal/ws_moveit2/src/moveit_task_constructor/core/src/stage.cpp:955:31: error: ‘const class moveit::core::AttachedBody’ has no member named ‘getShapePosesInLinkFrame’
  955 |   auto from_it = from_object->getShapePosesInLinkFrame().cbegin();
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~
/home/gal/ws_moveit2/src/moveit_task_constructor/core/src/stage.cpp:956:32: error: ‘const class moveit::core::AttachedBody’ has no member named ‘getShapePosesInLinkFrame’
  956 |   auto from_end = from_object->getShapePosesInLinkFrame().cend();
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~
/home/gal/ws_moveit2/src/moveit_task_constructor/core/src/stage.cpp:957:27: error: ‘const class moveit::core::AttachedBody’ has no member named ‘getShapePosesInLinkFrame’
  957 |   auto to_it = to_object->getShapePosesInLinkFrame().cbegin();
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:141: src/CMakeFiles/moveit_task_constructor_core.dir/stage.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:264: src/CMakeFiles/moveit_task_constructor_core.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

im using ros2 foxy. can anyone help me? 
thank you.

rhaschke, 2023-05-03:
Looks like you are using incompatible versions of various source packages. Please list the versions of MoveIt and MTC you are using from source.

galversano, 2023-05-04:
> Looks like you are using incompatible versions of various source packages. Please list the versions of MoveIt and MTC you are using from source.

hey, i follow this tutorial:
https://moveit.picknik.ai/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html

i source the MTC with this command:
git clone https://github.com/ros-planning/moveit_task_constructor.git -b ros2

and the version of moveit i used is with this command:
https://github.com/ros-planning/moveit2_tutorials/tree/foxy

im using ros2 foxy.
can you help please?

Eser-TU, 2024-03-14:
hello did you solve the problem?

rhaschke, 2024-03-14:
The old Foxy release doesn't work with the latest ros2 branch. Try to use the humble branch instead.

navdotnetreqs, 2024-09-16:
Good day,

I have the same or similar issue running Ubuntu 22.04 and ROS2 Iron.

```
git clone https://github.com/ros-planning/moveit_task_constructor.git -b ros2
colcon build
```

```
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Starting >>> hello_moveit
--- stderr: rviz_marker_tools                                                                                                                              
rviz_marker_tools: You did not request a specific build type: Choosing 'Release' for maximum performance
---
Finished <<< rviz_marker_tools [11.4s]
Finished <<< hello_moveit [23.8s]                                                                                    
Finished <<< moveit_task_constructor_msgs [28.4s]                       
Starting >>> moveit_task_constructor_core
[Processing: moveit_task_constructor_core]                             
[Processing: moveit_task_constructor_core]                                      
--- stderr: moveit_task_constructor_core                                        
moveit_task_constructor_core: You did not request a specific build type: Choosing 'Release' for maximum performance
Missing content of submodule pybind11: Use 'git clone --recurse-submodule' in future.
Checking out content automatically
Submodule 'core/python/pybind11' (https://github.com/pybind/pybind11) registered for path 'pybind11'
Cloning into '/home/nb5/myar_ws/src/moveit_task_constructor/core/python/pybind11'...
From https://github.com/pybind/pybind11
 * branch              f4bc71f981d4eb2dd780215fd3c5a7420f1f03aa -> FETCH_HEAD
/home/nb5/myar_ws/src/moveit_task_constructor/core/src/solvers/cartesian_path.cpp: In member function ‘virtual moveit::task_constructor::solvers::PlannerInterface::Result moveit::task_constructor::solvers::CartesianPath::plan(const PlanningSceneConstPtr&, const moveit::core::LinkModel&, const Isometry3d&, const Isometry3d&, const moveit::core::JointModelGroup*, double, robot_trajectory::RobotTrajectoryPtr&, const Constraints&)’:
/home/nb5/myar_ws/src/moveit_task_constructor/core/src/solvers/cartesian_path.cpp:123:42: error: ‘relative’ is not a member of ‘moveit::core::JumpThreshold’
  123 |             moveit::core::JumpThreshold::relative(props.get<double>("jump_threshold")), is_valid,
      |                                          ^~~~~~~~
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:230: src/CMakeFiles/moveit_task_constructor_core.dir/solvers/cartesian_path.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
/home/nb5/myar_ws/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp: In member function ‘moveit::task_constructor::solvers::PlannerInterface::Result moveit::task_constructor::solvers::PipelinePlanner::plan(const PlanningSceneConstPtr&, const moveit::core::JointModelGroup*, const Constraints&, double, robot_trajectory::RobotTrajectoryPtr&, const Constraints&)’:
/home/nb5/myar_ws/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:213:53: error: ‘const class moveit::core::MoveItErrorCode’ has no member named ‘message’
  213 |                 return { false, solution.error_code.message };
      |                                                     ^~~~~~~
/home/nb5/myar_ws/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:213:61: error: could not convert ‘{false, <expression error>}’ from ‘<brace-enclosed initializer list>’ to ‘moveit::task_constructor::solvers::PlannerInterface::Result’
  213 |                 return { false, solution.error_code.message };
      |                                                             ^
      |                                                             |
      |                                                             <brace-enclosed initializer list>
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:258: src/CMakeFiles/moveit_task_constructor_core.dir/solvers/pipeline_planner.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:364: src/CMakeFiles/moveit_task_constructor_core.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_task_constructor_core [1min 7s, exited with code 2]

Summary: 3 packages finished [1min 36s]
  1 package failed: moveit_task_constructor_core
  2 packages had stderr output: moveit_task_constructor_core rviz_marker_tools
  4 packages not processed

```

Any ideas?

navdotnetreqs, 2024-09-16:
Trying with branch "humble" and I get a little different compile errors..

```
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Starting >>> hello_moveit
Finished <<< hello_moveit [0.22s]                                                                                                           
Finished <<< moveit_task_constructor_msgs [5.58s]                                                                      
--- stderr: rviz_marker_tools                              
rviz_marker_tools: You did not request a specific build type: Choosing 'Release' for maximum performance
---
Finished <<< rviz_marker_tools [9.19s]
Starting >>> moveit_task_constructor_core
[Processing: moveit_task_constructor_core]                             
[Processing: moveit_task_constructor_core]                                     
--- stderr: moveit_task_constructor_core                                        
moveit_task_constructor_core: You did not request a specific build type: Choosing 'Release' for maximum performance
Missing content of submodule pybind11: Use 'git clone --recurse-submodule' in future.
Checking out content automatically
Submodule 'core/python/pybind11' (https://github.com/pybind/pybind11) registered for path 'pybind11'
Cloning into '/home/nb5/myar_ws/src/moveit_task_constructor/core/python/pybind11'...
From https://github.com/pybind/pybind11
 * branch              f4bc71f981d4eb2dd780215fd3c5a7420f1f03aa -> FETCH_HEAD
/home/nb5/myar_ws/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp: In member function ‘moveit::task_constructor::solvers::PlannerInterface::Result moveit::task_constructor::solvers::PipelinePlanner::plan(const PlanningSceneConstPtr&, const MotionPlanRequest&, robot_trajectory::RobotTrajectoryPtr&)’:
/home/nb5/myar_ws/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:213:22: error: ‘struct planning_interface::MotionPlanResponse’ has no member named ‘trajectory_’; did you mean ‘trajectory’?
  213 |         result = res.trajectory_;
      |                      ^~~~~~~~~~~
      |                      trajectory
/home/nb5/myar_ws/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:214:92: error: ‘struct planning_interface::MotionPlanResponse’ has no member named ‘error_code_’; did you mean ‘error_code’?
  214 |         return { success, success ? std::string() : moveit::core::error_code_to_string(res.error_code_.val) };
      |                                                                                            ^~~~~~~~~~~
      |                                                                                            error_code
/home/nb5/myar_ws/src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp:214:109: error: could not convert ‘{success, <expression error>}’ from ‘<brace-enclosed initializer list>’ to ‘moveit::task_constructor::solvers::PlannerInterface::Result’
  214 |         return { success, success ? std::string() : moveit::core::error_code_to_string(res.error_code_.val) };
      |                                                                                                             ^
      |                                                                                                             |
      |                                                                                                             <brace-enclosed initializer list>
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:258: src/CMakeFiles/moveit_task_constructor_core.dir/solvers/pipeline_planner.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:362: src/CMakeFiles/moveit_task_constructor_core.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_task_constructor_core [1min 6s, exited with code 2]

Summary: 3 packages finished [1min 15s]
  1 package failed: moveit_task_constructor_core
  2 packages had stderr output: moveit_task_constructor_core rviz_marker_tools
  4 packages not processed
```

rhaschke, 2024-09-16:
Unfortunately, MoveIt2 has many branches with incompatible APIs. Looks like, we need a separate branch for Iron as well.
I don't have access to an Iron distro. I suggest starting from the `ros2` branch and cherry-picking some of the latest commits on the `humble` branch to fix compiler issues. From the reported compiler logs, I guess you will need the following commits:

* 8d0708c1 Revert "Update to the more recent JumpThreshold API"
* e942e9c7 Revert "Enable parallel planning with PipelinePlanner"
* ba1d3df6 Revert "Fix generation of Solution msg: consider backward operation"

navdotnetreqs, 2024-09-17:
Hi @rhaschke and thank you for your reply,

I cherry picked those 3 commits, but the error remains.

rhaschke, 2024-09-17:
> I then went into src/moveit_task_constructor/core/src/solvers/pipeline_planner.cpp and renamed references to error_code_ to error_code and trajectory_ to trajectory, and it compiles. Maybe it works as well.

Then you started off the humble branch... I suggested to start from the ros2 branch ;-)

navdotnetreqs, 2024-09-17:
Hi, no, pretty sure I started from ros2. Anyway, seems to be working, so now I can focus on trying to get my code to work.. =D

[image]

closed 2024-09-17 by rhaschke