moveit/moveit_task_constructor #457: ModifyPlanningScene is not allowing all collision

LeroyR, 2023-05-04:
I am using the `ModifyPlanningScene` stage for a scene where an object is stuck inside other objects. 

[Allowing all collision](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/include/moveit/task_constructor/stages/modify_planning_scene.h#L107) does not disable collisions for the specified object.

My workaround is to get all objects out of the planning scene and explicitly disable them, e.g: 

```
auto stage = std::make_unique<stages::ModifyPlanningScene>("allow collision (object,others)");
//stage->allowCollisions(object, true); //does not work, collisions between object and other object
stage->allowCollisions(object, all_objects, true); //does work
```

rhaschke, 2023-05-04:
I cannot confirm this issue. I just added a corresponding test (#460) and it works as expected.
Is it possible that the list of `all_objects` comprises a name that is not (yet) part of the ACM? The corresponding method only works for _known_ objects.

LeroyR, 2023-05-05:
Using all objects is the working case after all.

Object gets attached in the previous stage, I'll try to produce a failed test/example.

LeroyR, 2023-05-05:
`all_objects` is the working case after all. 

The `object` is attached maybe that causes the problem?

With this patch:
```
diff --git a/demo/config/panda_config.yaml b/demo/config/panda_config.yaml
index 090cc6d..99bbca7 100644
--- a/demo/config/panda_config.yaml
+++ b/demo/config/panda_config.yaml
@@ -22,7 +22,7 @@ surface_link: "table"
 # CYLINDER object specifications
 object_name: "object"
 object_dimensions: [0.25, 0.02] # [height, radius]
-object_pose: [0.5, -0.25, 0.0, 0, 0, 0]
+object_pose: [0.5, -0.25, -0.05, 0, 0, 0]
 
 # Table model
 spawn_table: true
diff --git a/demo/src/pick_place_task.cpp b/demo/src/pick_place_task.cpp
index f9732dd..cebe06c 100644
--- a/demo/src/pick_place_task.cpp
+++ b/demo/src/pick_place_task.cpp
@@ -309,7 +309,8 @@ bool PickPlaceTask::init() {
 		 ***************************************************/
 		{
 			auto stage = std::make_unique<stages::ModifyPlanningScene>("allow collision (object,support)");
-			stage->allowCollisions({ object }, support_surfaces_, true);
+			//stage->allowCollisions({ object }, support_surfaces_, true);
+			stage->allowCollisions(object, true);
 			grasp->insert(std::move(stage));
 		}
``` 

The test fails
```
catkin test moveit_task_constructor_demo
SUMMARY
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 1

```

rhaschke, 2023-05-06:
As expected: the ACM doesn't yet know about both `object` and `table`. Calling `allowCollisions("object", True)` just enables collisions for object will all known links, but not with table...
#460 should fix that now.

closed 2023-05-25 by rhaschke