moveit/moveit_task_constructor #472: task that executes differently based on planning scene state

TheUncivil, 2023-06-22:
I'd like to have a single task that branches into two separate series of stages, then have the MoveIt task constructor select which of those two to follow based on other external conditions, in this case the state of the planning scene.

Does the MoveIt task constructor provide a good way to achieve this, or would it be better to make the decision when generating the task, before sending it to the MoveIt task constructor?

TheUncivil, 2023-06-23:
I intend to implement this using two Serial Containers containing my task's "branches", leading one with a predicate filter that determines wether the first branch is desirable, and then using a Fallback Container encapsulating those branches to make the system choose alternative options should the filter fail.

TheUncivil, 2023-06-28:
This did work, although I it refused to plan with Serial Containers for some reason. However, since I only really need to change up one stage in my task, that's not a problem for me right now.

closed 2023-06-28 by TheUncivil

rhaschke, 2023-06-28:
Indeed, the described approach, using a filter stage at the beginning of the first serial container should work.
What exactly do you mean with "planning with serial containers refused to work"? What happened?

TheUncivil, 2023-06-28:
For whatever reason, moveit failed to plan the task every single time when I used serial containers to place subsets of stages inside the fallback container. I do not know what exactly caused this issue, but was able to fix it by not using serial containers.

I've otherwise had no problems with serial containers.

TheUncivil, 2023-06-29:
I would like to make a minimal example based on the existing pick and place demo, I'm also working on a pick and place system so it is very similiar. This would allow me to easily show you the task description using the panda. 

However, there are no instructions for running the demo. Simply launching pickplace.launch.py from moveit_task_constructor_demo does not work:

`
[pick_place_demo-1] [WARN] [1688022106.079600529] [moveit_ros.planning_scene_interface.planning_scene_interface]: service '/get_planning_scene' not advertised yet. Continue waiting...
`

I assumed this meant I first had to launch demo.launch.py from the same package to set up the robot model and controllers and planning scene, but that throws an error:

`
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: 'robot_description_planning'
`

How do I run the pick and place demo? Once that's operational, I can attempt to make a minimal example. Also, should I open a separate issue for this?

TheUncivil, 2023-06-30:
I was working on a minimal example based on the demo as described in your tutorials, and found that you organized the stages inside your serial containers differently from me; specifically, I tend to prioritize keeping the number of stages inside the containers to a minimum, while you seem to place a generate stage in the container, then surround it in propagator stages until a connect stage is needed, which you keep outside the container. When I reorganized my stages to mimic this, the system worked as intended. Should I just not have connect stages inside containers?

My task previously looked like as follows (I hope this is somewhat legible):

-> CurrentState
-> Connect
-> ModifyScene (Allow Collision)
-> GenerateGraspPose
-> ModifyScene (Attach target)
-> Fallback container
->> Serial container
->>> Predicate Filter (check target orientation)
->>>> MoveRelative (picking target of known orientation)
->>> MoveRelative (retreat)
->>> Connect
->>> MoveRelative (lower)
->>> GeneratePlacePose
->>Serial container
->>>MoveRelative (picking target of unknown orientation)
->>> MoveRelative (retreat)
->>> Connect
->>> MoveRelative (lower)
->>> GeneratePlacePose
-> ModifyScene (detach target)
-> MoveRelative (retreat)
-> ModifyScene (remove target from scene)

It now looks like this:

-> CurrentState
-> Connect
-> Fallback container
->> Serial container
->>> ModifyScene (Allow Collision)
->>> GenerateGraspPose
->>> ModifyScene (Attach target)
->>>Predicate Filter (check target orientation)
->>>> MoveRelative (picking target of known orientation)
->>> MoveRelative (retreat)
->> Serial container
->>> ModifyScene (Allow Collision)
->>> GenerateGraspPose
->>> ModifyScene (Attach target)
->>> MoveRelative (picking target of unknown orientation)
->>> MoveRelative (retreat)
-> Connect
-> MoveRelative (lower)
-> GeneratePlacePose
-> ModifyScene (detach target)
-> MoveRelative (retreat)
-> ModifyScene (remove target from scene)

rhaschke, 2023-06-30:
Connect stages inside SerialContainers should work w/o problems. Usually, you should avoid duplicate planning work.
Hence, if your stages inside the serial containers perform identical planning work, you should better move them outside:
```
CurrentState
Connect
ModifyScene (Allow Collision)
GenerateGraspPose
ModifyScene (Attach target)
Fallback container
   Serial container
      Predicate Filter (check target orientation)
         MoveRelative (picking target of known orientation)
   Serial container
      MoveRelative (picking target of unknown orientation)
MoveRelative (retreat)
Connect
MoveRelative (lower)
GeneratePlacePose
ModifyScene (detach target)
MoveRelative (retreat)
ModifyScene (remove target from scene)
```

TheUncivil, 2023-07-03:
At that stage, surely there is no need to use serial containers anymore, as each only contains a single stage. I already tried that and it worked.

The reason I had so many stages inside the container to begin with was because the two different picking motions end up in different positions and I assumed that would make a meaningful difference for the planning that came after, but the system's success without using serial containers proves that assumption wrong.