moveit/moveit_task_constructor #473: failure recovery during runtime

TheUncivil, 2023-06-29:
I'm currently working on a system that performs multiple pick-and-place tasks continuously; to ensure this happens as swiftly as possible, I plan tasks ahead of time based on the end state of the previous task. A problem with this approach is that all task plans are dependent on the previous tasks finishing correctly;  in the event of unanticipated failure during runtime, the task ends early and all future plans have to be discarded as a consequence.

Originally, I had intended to use a Fallback container to define a 'recovery'  plan in the event of failure by returning to a defined starting position. By then planning all tasks from that position, I would avoid having to discard all future plans. However, as far as I understand, the Fallback container only works during the planning phase of the task, and cannot help with runtime issues.

Naturally planning recovery trajectories for any potential failure state during runtime is untenable, but might it be possible to define potentially problematic stages during which failure could be reasonably expected ahead of time, then plan recovery from those stages? In my case, that would be the moment the gripper attempts to grasp the target, since the target's position may be slightly off, which would cause the gripper to miss its mark. Since the arm otherwise does not move during this stage, planning a recovery motion would be relatively simple.

Is there a method to do this in the MoveIt Task Constructor, or is there any intent to add such a function in the future?

rhaschke, 2023-06-29:
> However, as far as I understand, the Fallback container only works during the planning phase of the task, and cannot help with runtime issues.

This is correct.
To recover from failed grasping and return the arm to a default pose, I suggest to have a corresponding _separate_ MTC plan prepared and running it in case of failure. There is no error handling or recovery available in MTC itself. I consider this the task of a higher-level execution engine.

TheUncivil, 2023-06-30:
> There is no error handling or recovery available in MTC itself. I consider this the task of a higher-level execution engine.

That makes sense, I'll see if I can find a solution at higher levels.

> To recover from failed grasping and return the arm to a default pose, I suggest to have a corresponding separate MTC plan prepared and running it in case of failure.

That's actually the alternative I was considering already. I'll see if there are any other options, but may well use this solution.

closed 2023-06-30 by TheUncivil