moveit/moveit_task_constructor #475: ExecuteTaskSolution action returns no feedback

TheUncivil, 2023-07-04:
According to the ExecuteTaskSolution action definition, it should provide the IDs and numbers of finished subtrajectories (stages?) as feedback while executing the action. I'm attempting to print that information like so:

```
void MtcClient::cbFeedback(
  rclcpp_action::ClientGoalHandle<moveit_task_constructor_msgs::action::ExecuteTaskSolution>::SharedPtr p_goal_handle,
  const std::shared_ptr<const moveit_task_constructor_msgs::action::ExecuteTaskSolution::Feedback> p_feedback)
{
  RCLCPP_INFO(
    p_node_->get_logger(), "Received feedback: sub_id %i,  sub_no %i", p_feedback->sub_id,
    p_feedback->sub_no);
}
```

However, nothing is being printed. As far as I can tell, I'm receiving no feedback at all, even as the action executes normally.

What is the reason for this? Is feedback not yet implemented, or should I enable it somehow?

rhaschke, 2023-07-04:
Unfortunately feedback is not yet implemented:
https://github.com/ros-planning/moveit_task_constructor/blob/c605a0059a3030bd1095202d0662c6f8e43e345e/capabilities/src/execute_task_solution_capability.cpp#L171

rhaschke, 2025-02-11:
Closed via #653.