moveit/moveit_task_constructor #477: Record and Playback Solutions

swiz23, 2023-07-17:
Hi,

This is a feature request. I'm wondering if it would be possible to record plans (using rosbag) published to the `/solution` topic by MTC, then play them back later for debugging purposes. This would be particularly helpful in situations where you are testing hardware in the field and don't have time to debug failed plans on the spot. I tried doing this already by only recording the '/solution'` topic using the rosbag utility. Next, I started up MoveIt, Rviz, ros2 control, etc... in simulation and tried to playback the recorded messages from my bagfile one at a time. I could ros2 topic echo the `/solution` topic and see the message appear, but it didn't look like Rviz was subscribing to the messages at all (doing a ros2 topic info -v /solution showed no subscribers). I feel like this should be possible, but maybe I'm missing something? Like some sort of initialization step?

rhaschke, 2023-09-22:
Sorry for the late reply. 
Later inspection of MTC only works partially: It is possible to record and replay solutions, but you need to record 3 different topics: `rosbag record /mtc/solution  /mtc/description /mtc/statistics`
While the first one comprises the actually published solution(s), the second and third topics, provide the basic task structure and progress statistics respectively. Without these, rviz cannot interpret the solution.

Limitation:
You can only inspect solutions that were explicitly published by your MTC node. By default, only the best solution is published. The interactive inspection of other solutions is "on demand", employing a client-server approach: If you select a particular solution in rviz, rviz will actively ask the MTC node for that solution. Obviously, that scheme will not work in hindsight anymore.

swiz23, 2023-09-22:
Thanks!

Just tried this. I noticed that for successful plans, I was able to see them populate the Motion Planning Tasks Rviz plugin. I was also able to click on different stages in the Rviz plugin, then the cost, and see the ghost version of the manipulator move based on it.

For plans that were unsuccessful, I was able to see them populate the Motion Planning Tasks Rviz plugin as well. I could also see at which stage the cost was infinite. However, when I tried to click on the failed stage (or even a successful stage within the failed task), I did not see the ghost version of the manipulator move. Is that because a full solution was never published for that task? Is the limitation you described? In a normal scenario (no rosbags, just using mtc directly), when you click on a stage within a failed task in the Motion Planning Tasks Rviz plugin, does that try to call the MTC node 'live' for info on it?

rhaschke, 2023-09-25:
> Is that because a full solution was never published for that task?

Exactly. Those unsuccessful solutions were never published.

> When you click on a stage within a failed task in the Motion Planning Tasks Rviz plugin, does that try to call the MTC node 'live' for info on it?

Yes.

swiz23, 2023-09-27:
Is there a way to publish those unsuccessful solutions anyway using MTC?

rhaschke, 2023-09-27:
No, currently there is no way to access those solutions in hindsight.
Remember that the inspection stuff is intended for the development phase: It is assumed that you can keep your MTC node alive during development, thus allowing for interactive inspection. In a production system, you should disable storing of non-solutions anyway ;-)

swiz23, 2023-09-27:
Issue has been resolved to the extent possible.

closed 2023-09-27 by swiz23