moveit/moveit_task_constructor #481: No rule to make target moveit_task_constructor_msgs - moveit tutorial unable to build

vivithemage, 2023-07-31:
Hello, after following the [moveit2 humble tutorial,](https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html#install-ros-2-and-colcon) there is an error when running:

`colcon build --mixin release`

The error is:

```
Starting >>> moveit_task_constructor_msgs
--- stderr: moveit_task_constructor_msgs                             
gmake[1]: *** No rule to make target 'moveit_task_constructor_msgs__py/CMakeFiles/moveit_task_constructor_msgs__py.dir/all', needed by 'CMakeFiles/moveit_task_constructor_msgs__rosidl_generator_py.dir/all'. Stop.
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2

Failed   <<< moveit_task_constructor_msgs [0.32s, exited with code 2]

Summary: 0 packages finished [0.80s]
  1 package failed: moveit_task_constructor_msgs
  1 package had stderr output: moveit_task_constructor_msgs
```

Moveit Task Constructor seems to be the only package which breaks the setup of the moveit tutorial on humble.
Any help would be appreciated. 

Thanks in advance.

rhaschke, 2023-08-01:
Building MTC works without issues for me. Please try cleaning your workspace and rebuilding.

vivithemage, 2023-08-06:
Thanks for testing and your reply. I've started again with a new workspace and followed the [moveit steps](https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html#install-ros-2-and-colcon).

The error is showing with a fresh workspace. I'm not sure what's missing/incorrect to result in this error.

Thanks again

```
vivi@precision:~$ cd ~/ws_moveit2/src
git clone https://github.com/ros-planning/moveit2_tutorials -b humble --depth 1
Cloning into 'moveit2_tutorials'...
remote: Enumerating objects: 510, done.
remote: Counting objects: 100% (510/510), done.
remote: Compressing objects: 100% (474/474), done.
remote: Total 510 (delta 42), reused 344 (delta 20), pack-reused 0
Receiving objects: 100% (510/510), 44.51 MiB | 3.25 MiB/s, done.
Resolving deltas: 100% (42/42), done.
vivi@precision:~/ws_moveit2/src$ vcs import < moveit2_tutorials/moveit2_tutorials.repos
.......
=== ./launch_param_builder (git) ===
Cloning into '.'...
=== ./moveit2 (git) ===
Cloning into '.'...
=== ./moveit_resources (git) ===
Cloning into '.'...
=== ./moveit_task_constructor (git) ===
Cloning into '.'...
=== ./moveit_visual_tools (git) ===
Cloning into '.'...
=== ./rosparam_shortcuts (git) ===
Cloning into '.'...
=== ./srdfdom (git) ===
Cloning into '.'...
vivi@precision:~/ws_moveit2/src$ sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
[sudo] password for vivi: 
Hit:1 http://gb.archive.ubuntu.com/ubuntu jammy InRelease
Get:2 http://gb.archive.ubuntu.com/ubuntu jammy-updates InRelease [119 kB]                     
Get:3 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]                      
Hit:4 https://linux.teamviewer.com/deb stable InRelease                                        
Hit:5 https://brave-browser-apt-release.s3.brave.com stable InRelease                          
Hit:6 http://packages.ros.org/ros2/ubuntu jammy InRelease                                      
Hit:7 https://dl.google.com/linux/chrome/deb stable InRelease                                  
Get:8 http://gb.archive.ubuntu.com/ubuntu jammy-backports InRelease [109 kB]                   
Get:9 https://mega.nz/linux/repo/xUbuntu_20.04 ./ InRelease [2,961 B]                          
Hit:10 http://packages.osrfoundation.org/gazebo/ubuntu-stable focal InRelease    
Hit:11 https://esm.ubuntu.com/apps/ubuntu jammy-apps-security InRelease
Hit:12 https://esm.ubuntu.com/apps/ubuntu jammy-apps-updates InRelease
Hit:13 https://esm.ubuntu.com/infra/ubuntu jammy-infra-security InRelease
Hit:14 https://esm.ubuntu.com/infra/ubuntu jammy-infra-updates InRelease
Hit:15 https://esm.ubuntu.com/realtime/ubuntu jammy InRelease
Fetched 341 kB in 1s (242 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
57 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Skipping acquisition of configured file 'main/binary-i386/Packages', as repository 'https://esm.ubuntu.com/realtime/ubuntu jammy InRelease' doesn't support architecture 'i386'
#All required rosdeps installed successfully
vivi@precision:~/ws_moveit2/src$ cd ~/ws_moveit2
colcon build --mixin release
Starting >>> moveit_common
Starting >>> srdfdom
Starting >>> launch_param_builder
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_task_constructor_msgs
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_resources_prbt_support
Finished <<< moveit_common [1.69s]                                                          
Starting >>> rviz_marker_tools                                                             
Finished <<< moveit_resources_prbt_support [1.75s]                                         
Starting >>> rosparam_shortcuts
Finished <<< moveit_resources_fanuc_description [1.82s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_panda_description [1.89s]                                    
Starting >>> moveit_resources_panda_moveit_config
--- stderr: launch_param_builder                                                           
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< launch_param_builder [2.09s]
Finished <<< moveit_resources_pr2_description [2.15s]        
Finished <<< moveit_resources_fanuc_moveit_config [1.52s]        
Finished <<< moveit_resources_panda_moveit_config [1.60s]        
Starting >>> moveit_resources                                    
Finished <<< moveit_resources [1.35s]                            
--- stderr: moveit_task_constructor_msgs                           
gmake[1]: *** No rule to make target 'moveit_task_constructor_msgs__py/CMakeFiles/moveit_task_constructor_msgs__py.dir/all', needed by 'CMakeFiles/moveit_task_constructor_msgs__rosidl_generator_py.dir/all'. Stop.
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_task_constructor_msgs [13.0s, exited with code 2]
Aborted  <<< rviz_marker_tools [15.8s]                                         
Aborted  <<< srdfdom [20.0s]                                                    
Aborted  <<< rosparam_shortcuts [26.8s]                                    
                                   
Summary: 9 packages finished [29.2s]
  1 package failed: moveit_task_constructor_msgs
  3 packages aborted: rosparam_shortcuts rviz_marker_tools srdfdom
  2 packages had stderr output: launch_param_builder moveit_task_constructor_msgs
  42 packages not processed
vivi@precision:~/ws_moveit2$
```

rhaschke, 2023-08-07:
I still can't reproduce the error on a fresh humble installation. As can be seen from the [moveit2_tutorial CI builds](https://github.com/ros-planning/moveit2_tutorials/actions?query=branch%3Ahumble), building works as expected there too. I'm afraid you screwed your system installation in some way.

brschettini, 2023-08-20:
I don't know if it works for you, but it seems that the .repos file is mismatched from the humble version.

Try to install with the branch `humble` instead of `ros2`.