moveit/moveit_task_constructor #484: Task is getting stuck in execution - action server not started (humble)

valantano, 2023-09-04:
I am trying to execute a planned task but the execution gets stuck inside task_.execute(*task_.solutions().front());

I think this is because the Action Server is not found/not started.

ros2 action info /execute_task_solution returns:
```
Action: /execute_task_solution
Action clients: 1
    /task_view_private
Action servers: 0
```

And clicking the Execute button in Rviz returns:
```
[rviz2-2] [ERROR] [.] [moveit_task_constructor_visualization.task_view]: Failed to connect to task execution action
```

This is the console output when trying to execute the task
```
[mover_task_node-4] arm, gripper,   1  - ←   1 →   -  1 / demo task
[mover_task_node-4]     1  - ←   1 →   -  0 / current
[mover_task_node-4]     -  0 →   1 →   -  0 / Default Position
[mover_task_node-4]     -  0 →   1 →   -  1 / Left Position
[mover_task_node-4]     -  1 →   1 ←   1  - / move to pick
[mover_task_node-4]     1  - ←   1 →   -  0 / grasp pose IK
[mover_task_node-4]       1  - ←   1 →   -  1 / generate pose
[mover_task_node-4]     -  0 →   1 →   -  0 / approach object
[mover_task_node-4]     -  0 →   1 →   -  0 / attach object
[mover_task_node-4]     -  0 →   1 →   -  0 / Take object
[mover_task_node-4]     -  0 →   1 →   -  1 / Default Position with Object
[mover_task_node-4] [INFO] [.] [mover_node]: Task planned.
[mover_task_node-4] [INFO] [.] [pick_task]: Executing solution trajectory
// Getting stuck in auto execute_result = task_.execute(*task_.solutions().front());
```

My code corresponds to the tutorial code, calling init(), plan() and execute().

I appreciate any help.
Thanks in advance.

rhaschke, 2023-09-04:
The error message `Failed to connect to task execution action` indeed states that the action server is missing.

valantano, 2023-09-04:
So any Idea why it is not started? Or is this something I need to ensure that it is started because it is not started by the task constructor itself?

rhaschke, 2023-09-04:
Yes, the action server is not part of MTC but is provided by your robot driver. See ros_control for details.