moveit/moveit_task_constructor #487: no IK found in grasp pose IK stage

paweat, 2023-10-01:
Hello, I'm following the tutorial of pick and place with MTC but use my robot model instead. I got problem in grasp pose IK stage , it seems that cannot found at least 1 solution of IK for any pose of gripper. I don't know what caused this, or maybe my robot model is not flexible enough. Does it possible to fix?

**Environment:**
    - ROS2 Humble
    - Ubuntu 22.04 on VMware
    - MTC humble branch
    
[image]

rhaschke, 2023-10-01:
This error message means that the robot cannot reach the desired pose (not yet considering collisions).
From the given picture, I cannot infer the kinematic structure of your robot, but it looks like the target pose is too far away from your robot. Use the rviz visualizations to validate the correctness of your target pose!

paweat, 2023-10-01:
Sorry, My robot has 5 DOF for arm and 1 DOF for gripper. I have tried to set the joints to reach the target pose of the gripper , and it seems similar to one of the target pose that was generated. Is this help ? 

Note: this is my first project of moveit , sorry  if I missed something 

Before  
[image]

After
[image]

rhaschke, 2023-10-01:
With a 5-DoF arm you can't reach all poses in 3d space (which would required 6-DoF). Hence, you can only go for approximate solutions: You need to relax one degree of freedom to render the problem feasible. I'm not aware that any MoveIt IK solver can actually do that currently. I started some work along that direction for KDL years ago, but didn't finish.

paweat, 2023-10-01:
from my understanding that mean I cannot continue with MTC to do pick and place right?

rhaschke, 2023-10-01:
That's not a problem of MTC, but of the IK plugin you are using. You need to find an IK solver that can relax the pose constraint on some axes. Maybe [BioIK](https://github.com/TAMS-Group/bio_ik) can do so.