moveit/moveit_task_constructor #500: Stage timeout not working

RoBeau, 2023-10-28:
Hi there

I'm working on a complicated multi stage pipeline using MTC. As part of a project that needs to evaluate 100's of possible tasks. 

To manage the execution time we are trying to set timeouts for the task as well as each individual stage so that if the stage times out the entire task fails regardless of the task timeout. Is this functionality supported?

My colleague has provided a reduced example of our project for a separate issue https://github.com/leander2189/dtc_test

In this pipeline we define 3 connector stages

[move_to_fiducial_registration](https://github.com/leander2189/dtc_test/blob/94dbf23a675ee4d6da89b19adc0bc88cae0be512/src/dynamic_task_planner.cpp#L406C1-L408) - timeout = 1.0 s
[move_to_fastener_registration](https://github.com/leander2189/dtc_test/blob/94dbf23a675ee4d6da89b19adc0bc88cae0be512/src/dynamic_task_planner.cpp#L476-L478) - timeout = 2.0 s
[move_to_fastener_pre_dock](https://github.com/leander2189/dtc_test/blob/94dbf23a675ee4d6da89b19adc0bc88cae0be512/src/dynamic_task_planner.cpp#L542-L544C66) - timeout = 2.0 s

But if you run the example you can see that the stages continue planning well past these timeouts

[image]

rhaschke, 2023-10-28:
Timeouts for individual stages are currently considered _per planning attempt_, not accumulated.
Only the top-level task container currently considers the accumulated time.

Given that we have introduced timeouts for the _individual_ planners as well, we could actually change that behavior and consider accumulated time in _all_ stages (also containers, which currently ignore the timeout).
@ros-planning/moveit-maintainers: What do you think about that idea?

RoBeau, 2023-10-29:
Thanks for the quick reply. That makes sense. We might have some bandwidth to help out with this soon. If you had any rough pointers on how you would like it implemented.

sjahr, 2023-10-31:
@rhaschke,@MarqRazz and I discussed that and we think considering accumulated time in all stages is a good idea :+1: It might be good though, to keep the ability to specify the max planning time per attempt, for example to grant a connect stage that solves a hard problem more time.
What about renaming the solver/stage `timeout` -> `max planning time` to highlight that it is per attempt and add accumulated times limits as `timeout`?

rhaschke, 2023-10-31:
@sjahr, as I pointed out before the planning time of an individual planning attempt can be defined via the timeout property of the PlannerInterface:
https://github.com/ros-planning/moveit_task_constructor/blob/e752284d688e3db9251b3cc72a92dcd7a0191839/core/src/solvers/planner_interface.cpp#L50
Hence, there is no need to add an _additional_ timeout parameter in the stage.

MarqRazz, 2023-10-31:
@rhaschke would setting the `timeout` parameter have global scope over the task or is it possible to only apply it to individual stages?

Example:
```
{
    auto stage = std::make_unique<stages::MoveTo>("Step 1", mtc_pipeline_planner);
    // default planning timeout
    ...
}
{
  auto stage = std::make_unique<stages::Connect>(
      "Constrained connect to", stages::Connect::GroupPlannerVector{ { planning_group_name,
      mtc_pipeline_planner } });

  stage->setTimeout(10.0);  // Set the accumulated timeout for this stage
  mtc_pipeline_planner->timeout(2.5);// set the planning timeout for this stage larger because the constraint takes longer to solve
  ...
}
{
    auto stage = std::make_unique<stages::MoveRelative>("Step 2", mtc_cartesian_planner);
    // default planning timeout
    ...
}
```

rhaschke, 2023-10-31:
Of course, each stage would have a separate timeout. By default, they will be infinity. In your example, the Connect stage would use a timeout of 2.5s per planning attempt and 10s overall.

MarqRazz, 2023-11-01:
Thanks for the clarification Robert! Given the user has the flexibility to set these values per stage I agree updating `stage->setTimeout` to consider accumulated time in all stages (including containers) is the right move.

sjahr, 2023-11-02:
> @sjahr, as I pointed out before the planning time of an individual planning attempt can be defined via the timeout property of the PlannerInterface:
> 
> https://github.com/ros-planning/moveit_task_constructor/blob/e752284d688e3db9251b3cc72a92dcd7a0191839/core/src/solvers/planner_interface.cpp#L50
> 
> 
> Hence, there is no need to add an _additional_ timeout parameter in the stage, but the existing one can be used as accumulated timeout.

@rhaschke Yes, I am aware of that. My point was about renaming the existing property to `max_planning_time` to differentiate more clearly between accumulated times (`timout`) and planning time per attempt (`max_planning_time`). Is it possible to use the same solver in different stages with different timeouts if there is not planning attempt timeout at stage level?

rhaschke, 2023-11-02:
> Is it possible to use the same solver in different stages with different timeouts if there is not planning attempt timeout at stage level?

You can reuse planners/solvers across stages. However, as the timeout property is part of the solver, they will share that property across the stages too.

MarqRazz, 2023-11-02:
> However, as the timeout property is part of the solver, they will share that property across the stages too.

What if we made `stage->setTimeout` to consider accumulated time and `stage->max_planning_time` set the max planning time? That way we have the flexibility to only adjust it for stages that require addition time and others that do not specify a value can use `my_pipeline_planner->timeout(x.y);`

rhaschke, 2023-11-02:
I intended to avoid/remove the current redundancy between planner and stage timeout. However, I see the point of reusability of planners in different stages.

Note that container stages would only have the accumulated timeout: a per-planning-attempt timeout doesn't make sense for them.

MarqRazz, 2023-11-02:
> a per-planning-attempt timeout doesn't make sense for them.

Why? I thought that was the original intent with `stage->setTimeout`. 

Here is the same use case from above that supports the need, where I would prefer to not make multiple pipeline planners just to specify different planning timeouts.

```
 mtc_pipeline_planner->max_planning_time(0.5);// my default planning timeout
{
    auto stage = std::make_unique<stages::MoveTo>("Step 1", mtc_pipeline_planner);
    // default planning timeout
    ...
}
{
  auto stage = std::make_unique<stages::Connect>(
      "Constrained connect to", stages::Connect::GroupPlannerVector{ { planning_group_name,
      mtc_pipeline_planner } });
  stage->setTimeout(10.0);  // Set the accumulated timeout for this stage
  stage->max_planning_time(2.5);// set the planning timeout for this stage larger because the constraint takes longer to solve
  ...
}
{
    auto stage = std::make_unique<stages::MoveTo>("Step 2", mtc_pipeline_planner);
    // default planning timeout
    ...
}
```

v4hn, 2023-11-03:
> To manage the execution time we are trying to set timeouts for the task as well as each individual stage so that if the stage times out the entire task fails regardless of the task timeout. Is this functionality supported?

No, and I don't think it's a reasonable thing to do. Stages correspond to temporal segments of trajectories which depend on the other segments. So if you, e.g., keep sampling bad grasp points in one stage, the computation for another segment (often Connect) will fail with timeout a lot. That is clearly no reason to stop planning though as long as there is still overall time left. The timeout per attempt is supposed to reflect "any valid solution path through the stage should have this segment solved within that maximum time. Any attempt taking longer is likely unsolvable/unwanted in a solution".

As something more reasonable in this direction, you could for example monitor solutions for some stages and stop the task if it did not receive *any* (still `ENABLED`) solutions for these stages after some lower timeout. This pretty much corresponds to "if you could not find a way to pick after half the time, stop trying to plan the whole pick&place task").
There is no direct support for this either, but it can be implemented with solution callbacks.

Sorry, but I fail to see how a timeout per stage that is independent of the planning attempt solves anything.
It just introduces one more potential problem. It means that if you get a number of infeasible state pairs in Connect and plan with OMPL until (per attempt) timeout for each of them, you will never even attempt to connect the "good" pairs after that, i.e., your solver becomes incomplete even with unlimited overall Task runtime.
The real underlying problem here (again) is the very limited sequential compute implementation.

> I intended to avoid/remove the current redundancy between planner and stage timeout

There is indeed no reason for the Planner timeout at all, as they will always be invoked through stages. :wink:

rhaschke, 2023-11-05:
@v4hn, I understand your reasoning for ensuring completeness. However, you name a reason to limit the overall planning time for a stage: 

> If you cannot find a way to pick after half the time, stop trying to plan the whole pick&place task.

I think this is a perfect reason to introduce an accumulated timeout. It's up to the users to apply them in a meaningful way. AFAIK, container stages (except the top-level task) do not consider the timeout property yet. For those, it doesn't make sense to handle a per-run timeout. Similar to the task container, all other containers only should handle an accumulated timeout.

v4hn, 2023-11-14:
Sorry, that took a while to write down.

There are two different concepts here:

- "timeout" in stages currently refers to the maximum amount of computation time to solve a concrete trajectory segment in a stage based on whatever was used as seed (fixed inputs, a monitored solution, or up to two `InterfaceState`s). Overall planning goes on when it is exceeded, but the attempt fails.

- "timeout" of the whole task. This is entirely unrelated but sadly uses the same property. It refers to the maximum time to plan the whole Task (unless sufficient solutions have been found earlier). In contrast to @sjahr and @MarqRazz, this is what I would actually call "max planning time", because "timeout" suggests that exceeding it leads to a failure. However, after the Task ran for this time, it might have generated already many solutions.

The clash in property semantics is unfortunate and could be changed.

Now, either interpretation can generalize to arbitrary containers with nuances.

For the first one:

- Any interface state/state pair triggering something in a container could have a limited overall planning time across the children. It would need more bookkeeping and it's not intuitive to set it to concrete values because you have to reason about the accumulated complexity of all children stages. I don't think we want that, but it's a natural generalization.
- Alternatively, it could refer to the planning time inside the container starting whenever it received solutions from its children. This is actually the semantics used in `ComputeIK` where the timeout is per-attempt for solving inverse kinematics for a single child solution. Other containers I could imagine that could/should use these semantics are, e.g., chomp/stomp/TrajOpt trajectory refinement, the `Merger` if it would adaptively merge parallel solutions instead of the current "attempt merge and check collisions, else serialize" approach. For containers which do not really compute a lot (or anything at all, like `SerialContainer` or `Fallbacks`) this concept is not useful though.

For the second one:

- As a mechanism to stop task planning if a part of the task does not yield a solution within some time, you can add a check to a stage to see if it yielded a solution after the task ran for some time. If there is no solution by then, abort Task planning. This adds just another check to abort, but otherwise could continue planning all stages as before. It's what I propose above.
- The same check could be implemented with aggregated time of the stage/container only and ignoring all other stage run times in the task. This is how @rhaschke interpreted my statement I guess. The main reason I prefer total run time of the task instead of only the container is because whether the checked container *can even* generate a solution within some stage-specific aggregated timeout depends on the rest of the task. In that sense such a timeout is not inherent to the stage/container, but depends on the whole task setup.
  - If only bad grasp points are selected, some subsequent stage might repeatedly run into per-attempt timeouts and never find a solution. But that doesn't mean that the stopping criterion is "this stage keeps timing out". I would argue it is actually "the generator keeps generating bad grasp points" which has nothing to do with this stage/container-specific aggregated timeout.
  - More importantly, it does not give you any guarantees over how long the Task will actually run if an aggregated timeout will be exceeded eventually. Depending on the task structure it might happen at around the timeout value in total planning time or it might mean that the whole task planner ran for *almost the sum of all defined aggregated timeouts* (+ runtime for everything without aggregated timeouts specified). That makes the property useless in my opinion.
- This variant could also be implemented with unconditional Task failure upon aggregated timeout. This is (as I understood it) what the original author asked for. However, I see no reason to fail the whole task if the stage that exceeds its aggregated timeout actually generated solutions.
- Alternatively, an aggregated timeout could implement *solution commitment*. That way overall planning would continue, but the stage/container's `canCompute()` will return false after that aggregated time, so all overall solutions of the task must use solutions from that stage/container that are already there. For the case where there is eventually no `ACTIVE` solution anymore running through the stage/container and there is no alternative path through the Task to get a full solution, some additional logic would need to be implemented that might be a bit more involved.

Independently of all this, there is also the discussion about the "timeout" of the planner. Multiple users (including me) treat it as a default value to optionally overwrite in stages that use the planner. I mainly don't want it to be the main interface for per-attempt timeouts because it only affects stages that use the PlannerInterface (namely `MoveTo` and `MoveRelative`, and `Connect`), but supporting *arbitrary* stages is vital to MTC. `Stage::timeout()` is available to *any* Stage which considers it.

Please point out your favorite mechanism from the list above @RoBeau @sjahr @MarqRazz @rhaschke .

rhaschke, 2023-11-15:
Your list nicely emphasizes the multitude of interpretations possible for a timeout parameter, which asks for a user-specific implementation of the desired behavior. I want to add some more points:
- The interpretation of the `timeout` in stages is stage-specific, i.e. not always referring to the planning time for a single planning attempt - that only holds for the listed stages `MoveTo` and `MoveRelative`, and `Connect`. 
  Other stages might implement different semantics. For example, `CurrentState` uses the timeout to wait for the `GetPlanningScene` service. `GenerateRandomPose` uses it per run attempt to generate as many solutions as possible within the given time.
- Most stages, namely containers, wrappers, and most generators, currently ignore the timeout parameter.
- I think, @RoBeau originally asked for the _solution commitment_ interpretation, i.e. stop generating new solutions after the accumulated planning time exceeds the threshold, but proceed with the existing ones. That's what I had in mind when talking about "accumulated timeouts" as well.

MarqRazz, 2023-11-15:
>I think, @RoBeau originally asked for the solution commitment interpretation, i.e. stop generating new solutions after the accumulated planning time exceeds the threshold, but proceed with the existing ones. That's what I had in mind when talking about "accumulated timeouts" as well.

I would also prefer the _solution commitment_ option.

RoBeau, 2023-11-20:
Hey all, thanks for such detailed engagement on this issue. I think I understand the main point being discussed here. @v4hn can you check this understanding?

Assuming that all stages are executing in parallel there's no point stopping a stage early (via a accumulative stage timeout) as the overall task will continue if other stages are still solving. If other stages are still solving then why not continue to solve solutions, as this would either improve your current best solution or reduce the chance of a stage timing out without any solutions causing the whole task to fail. 

If you add an accumulative timeout for every stage then you're basically just artificially lowering the overall task timeout. It would seem that using a `max_solutions` for each stage is a better way to get a task to complete early but still provide it with a long solve time it needs it for harder solutions.

RoBeau, 2023-12-18:
Hey team, any more progress on this?

RoBeau, 2024-01-30:
@v4hn any capacity to circle back to this?