moveit/moveit_task_constructor #518: Segfault calling `spawn()` from `ComputeIK` stage in latest ROS 2 version

sea-bass, 2023-12-15:
I've just upgraded to the latest version of MoveIt 2 and MTC and running into an issue I could use some help with.

I've partially isolated a new segfault to the `ComputeIK` stage calling `spawn()` in this line: https://github.com/ros-planning/moveit_task_constructor/blob/ros2/core/src/stages/compute_ik.cpp#L437

The error itself is:

```
 terminate called after throwing an instance of 'std::out_of_range'
   what():  vector::_M_range_check: __n (which is 12) >= this->size() (which is 12)
```

Any ideas that could help pinpoint the problem?

sea-bass, 2023-12-16:
I just saw https://github.com/ros-planning/moveit2/issues/2615 which may actually point to the root cause, if it's also happening outside MTC?

sea-bass, 2023-12-17:
Update: I've found that reverting https://github.com/ros-planning/moveit2/pull/2546 makes this issue go away. Seems to be related to a recent refactor of MoveIt's `RobotState` type and nothing to do with MTC.

marioprats, 2023-12-18:
This should be (partially) fixed by: https://github.com/ros-planning/moveit_task_constructor/pull/519
The real fix should happen on the MoveIt `RobotState` side.

rhaschke, 2023-12-18:
I don't see how #519 is related to `spawn()`.

sea-bass, 2023-12-18:
Right. I think it's an underlying MoveIt issue with the `RobotState` refactor that we ended up finding later. We can close this as it's not related to MTC.

Thank you for looking into tihs!

closed 2023-12-18 by sea-bass