moveit/moveit_task_constructor #520: Python wrapper issue - anonymous nodes and topics

matthias88, 2024-01-09:
Hi,
I didn't get an answer from the community at ros answers, so maybe it is an issue, or inexperience on my part :-):

I am using ROS noetic on Ubuntu 20.04 with MoveIt and MoveIt Task Constructor Framework. If I run one of the mtc tutorial examples from [MTC Tutorial](https://ros-planning.github.io/moveit_tutorials/doc/moveit_task_constructor/moveit_task_constructor_tutorial.html) the mtc solution topic and node is not of type anonymous, as you can see in the attached image:
`roslaunch moveit_task_constructor_demo demo.launch`
`rosrun moveit_task_constructor_demo cartesian`

[image]

If I use the python API (wrapper C++ class) the mtc node and topic is anonymous and can't be renamed or remapped:
`roslaunch moveit_task_constructor_demo demo.launch`
`rosrun moveit_task_constructor_demo cartesian.py __ns:=mtc_tutorial`

[image]

Finally I would like to get an easy access to the solution, so that I can execute e.g. subtrajectories via move group node (as suggested in #162  or time reparameterize the calculated solution. At least I haven't found a possiblity, to realize none of them directly with the [MTC Python API](https://ros-planning.github.io/moveit_task_constructor/_autosummary/pymoveit_mtc.core.html). 

Thanks in advance

rhaschke, 2024-01-09:
Running exactly the `cartesian.py` example from `master` branch, I get an `mtc_tutorial` node as expected:
[image]

rhaschke, 2024-01-09:
You can access the solutions of any stage (including tasks) via `stage.solutions`

matthias88, 2024-01-10:
> Running exactly the `cartesian.py` example from `master` branch, I get an `mtc_tutorial` node as expected: [...]

Thanks a lot for your effort testing it directly on your setup. In the meantime i made a fresh install of ROS noetic on WSL2 with Ubuntu 20.04 and built moveit + moveit_task_constructor from source. But i got the same result: the node is still anonymous. I am going to investigate it further...

matthias88, 2024-01-10:
> Running exactly the `cartesian.py` example from `master` branch, I get an `mtc_tutorial` node as expected: [...]

According to your screenshot, your task has a name "Cartesian Path", but the task name is not set in the cartesian.py example from the master branch - is it really exactly [this](https://github.com/ros-planning/moveit_task_constructor/blob/master/demo/scripts/cartesian.py) python script?

matthias88, 2024-01-10:
> You can access the solutions of any stage (including tasks) via `stage.solutions`

I also tried this approach to get access to the solution of a stage - contrary to task.solutions, it doesn't seem to work. I get the following error in the traceback: _"ValueError: Missing value for wrapped C++ type: Python instance was disowned."_
[image]

cambel, 2024-01-12:
Hi @rhaschke

I am facing the same issue. Could you explain in a bit more detail what you are doing on the working example?

My problem with the Python tutorial script is that the solution is published to an autogenerated topic namespace. Instead of publishing to `mtc_tutorial/solution`, it is published to `move_group_commander_wrappers_###/solution` where `###` is just a random 19-digit number that is different each time I execute the script. In the case of the cpp example, no such namespace is created and the solution is published to the expected topic.  

Do you perhaps know how to avoid that namespace prefix?

cambel, 2024-01-12:
I found that the method`roscpp_initialize` is the one that hardcodes the namespace "move_group_commander_wrappers". I tried using `roscpp_init from moveit_ros_planning_interface._moveit_roscpp_initializer` instead to define my own namespace "mtc_tutorial". 

``` python
from moveit_ros_planning_interface._moveit_roscpp_initializer import roscpp_init
roscpp_init('mtc_tutorial', [])
```

That works but unfortunately, I cannot get rid of the anonymity of the topic as that is hardcoded in the wrapper.

https://github.com/vinitha910/roscpp_initializer/blob/4b717021295bacd45ff6b7e6d3248f29d0a7b80c/src/roscpp_initializer.cpp#L85

So the mtc solutions are still published to `mtc_tutorial__1705037785642280345/solution`

matthias88, 2024-01-16:
Hi @cambel,

do you have any new findings about this issue?

cambel, 2024-01-17:
Hi @matthias88,

No, I just want confirmation that MTC with the Python API can work as expected without any significant change.

For now though, I modified the Python wrapper used to initialize roscpp in the moveit code base.
https://github.com/ros-planning/moveit/compare/master...cambel:moveit:allow-non-anonymous-roscpp-init

And then changed the code in the tutorial scripts as follows:
```python
from moveit_ros_planning_interface._moveit_roscpp_initializer import roscpp_init
roscpp_init('mtc_tutorial', [], False)
```

Somewhat hacky in my opinion, but It works fine for me with that.

rhaschke, 2024-03-05:
I factored out the python binding tools into a separate ROS package: https://github.com/ros-planning/py_binding_tools.
As soon as the release is accepted, you can directly use:
```python
from py_binding_tools import roscpp_init
roscpp_init("mtc_tutorial")
```

ron007d, 2024-04-15:
Getting the same issue, is there any update on the release ?

rhaschke, 2024-05-31:
The release is available now and I updated the master branch to use the new package.

closed 2024-05-31 by rhaschke