moveit/moveit_task_constructor #522: Building MTC does not set up Python properly

cambel, 2024-01-11:
I am trying to use MTC through the Python wrappers.

After successfully building from the source and trying to execute the demos from the tutorial, it cannot find the `moveit.task_constructor` Python module.

Checking the build log I found this warning:

```shell
Warnings   << moveit_task_constructor_core:symlink /root/ros_ws/logs/moveit_task_constructor_core/build.symlink.000.log
Warning: Cannot symlink from /root/ros_ws/devel/.private/moveit_task_constructor_core/lib/python3/dist-packages/moveit/__init__.py to existing file /root/ros_ws/devel/lib/python3/dist-packages/moveit/__init__.py
```

If I manually delete the existing file `.../moveit/__init__.py` and create the symbolic link, then I can find the MTC Python module.

I tried "catkin clean && catkin build" but I still get the same issue.

This issue seems related to #520

KejiaChen, 2024-01-19:
Hi cambel, I have the same issue. Could you please explain how you solved it?

cambel, 2024-01-22:
Hi @KejiaChen 

In my case, I basically execute these two commands:
```shell
rm /PATH-TO-ROS-WORKSPACE/devel/lib/python3/dist-packages/moveit/__init__.py
ln -s /PATH-TO-ROS-WORKSPACE/devel/.private/moveit_task_constructor_core/lib/python3/dist-packages/moveit/__init__.py  /PATH-TO-ROS-WORKSPACE/devel/lib/python3/dist-packages/moveit/__init__.py
```

rhaschke, 2024-01-25:
Try to an install build: `catkin config --install`. There are fundamental problems with the devel-space layout.

cambel, 2024-01-26:
@rhaschke 

Thanks, that works for me.

Just to clarify, after building, do not forget to source from the `install` folder. For example:
```shell
source /PATH_TO_ROS_WORKSPACE/nit/setup.bash
```

closed 2024-01-26 by cambel