moveit/moveit_task_constructor #526: Significant deviance in IK computation

denizkitu, 2024-01-23:
I've been trying to perform pick and place with MoveIt Task Constructor but the ik_frame and target_frame never seems to overlap as they should. I've made sure that the reference frame that ik solver uses is the gripper's frame. But i observed that none of the frames reach the target_frame.
[image]
This is my code and launch file:
`#! /usr/bin/env python
# -*- coding: utf-8 -*-

from moveit_commander.roscpp_initializer import roscpp_initialize
from moveit.task_constructor import core, stages
from moveit_commander import PlanningSceneInterface
from geometry_msgs.msg import PoseStamped, TwistStamped
import time

roscpp_initialize("pickplace")

# [pickAndPlaceTut1]
# Specify robot parameters
arm = "arm_controller"
eef = "hand"
# [pickAndPlaceTut1]

# [pickAndPlaceTut2]
# Specify object parameters
object_name = "grasp_object"
object_radius = 0.02

# Start with a clear planning scene
psi = PlanningSceneInterface(synchronous=True)
psi.remove_world_object()

# [initCollisionObject]
# Grasp object properties
objectPose = PoseStamped()
objectPose.header.frame_id = "world"
objectPose.pose.orientation.x = 1.0
objectPose.pose.position.x = 0.30702
objectPose.pose.position.y = 0.0
objectPose.pose.position.z = 0.285
# [initCollisionObject]

# Add the grasp object to the planning scene
psi.add_box(object_name, objectPose, size=[0.03, 0.03, 0.03])
# [pickAndPlaceTut2]

# [pickAndPlaceTut3]
# Create a task
task = core.Task("PandaPickPipelineExample")
task.enableIntrospection()
# [pickAndPlaceTut3]

# [pickAndPlaceTut4]
# Start with the current state
task.add(stages.CurrentState("current"))

# [initAndConfigConnect]
# Create a planner instance that is used to connect
# the current state to the grasp approach pose
pipeline = core.PipelinePlanner()
pipeline.planner = "RRTConnectkConfigDefault" # buraya bak bi tekrar
planners = [(arm, pipeline)]

# Connect the two stages
task.add(stages.Connect("connect", planners))
# [initAndConfigConnect]
# [pickAndPlaceTut4]

# [pickAndPlaceTut5]
# [initAndConfigGenerateGraspPose]
# The grasp generator spawns a set of possible grasp poses around the object
grasp_generator = stages.GenerateGraspPose("Generate Grasp Pose")
grasp_generator.angle_delta = 0.2
grasp_generator.pregrasp = "open"
grasp_generator.grasp = "close"
grasp_generator.setMonitoredStage(task["current"])  # Generate solutions for all initial states
# [initAndConfigGenerateGraspPose]
# [pickAndPlaceTut5]
  
# [pickAndPlaceTut6]
# [initAndConfigSimpleGrasp]
# SimpleGrasp container encapsulates IK calculation of arm pose as well as finger closing
simpleGrasp = stages.SimpleGrasp(grasp_generator, "Grasp")
# Set frame for IK calculation in the center between the fingers
ik_frame = PoseStamped()
ik_frame.header.frame_id = "gripper_link"
ik_frame.pose.position.z = 0.09 # burası nasıl işliyor?
ik_frame.pose.position.y = -0.02
simpleGrasp.setIKFrame(ik_frame)
# [initAndConfigSimpleGrasp]
# [pickAndPlaceTut6]

# [pickAndPlaceTut7]
# [initAndConfigPick]
# Pick container comprises approaching, grasping (using SimpleGrasp stage), and lifting of object
pick = stages.Pick(simpleGrasp, "Pick")
pick.eef = eef
pick.object = object_name

# Twist to approach the object
approach = TwistStamped()
approach.header.frame_id = "world"
approach.twist.linear.z = -1
#approach.twist.linear.y = 0.5
pick.setApproachMotion(approach, 0.04, 0.40)

# Twist to lift the object
lift = TwistStamped()
lift.header.frame_id = "gripper_link"
lift.twist.linear.z = -1.0
pick.setLiftMotion(lift, 0.03, 0.1)
# [pickAndPlaceTut7]

# [pickAndPlaceTut8]
# Add the pick stage to the task's stage hierarchy
task.add(pick)
# [initAndConfigPick]
# [pickAndPlaceTut8]

# [pickAndPlaceTut9]
# Connect the Pick stage with the following Place stage
# # [pickAndPlaceTut9]

# # [pickAndPlaceTut10]
# # [initAndConfigGeneratePlacePose]
# # Define the pose that the object should have after placing
# placePose = objectPose
# placePose.pose.position.y += 0.2  # shift object by 20cm along y axis

# # Generate Cartesian place poses for the object
# place_generator = stages.GeneratePlacePose("Generate Place Pose")
# place_generator.setMonitoredStage(task["Pick"])
# place_generator.object = object_name
# place_generator.pose = placePose
# # [initAndConfigGeneratePlacePose]
# # [pickAndPlaceTut10]

# # [initAndConfigSimpleUnGrasp]
# # The SimpleUnGrasp container encapsulates releasing the object at the given Cartesian pose
# # [pickAndPlaceTut11]
# simpleUnGrasp = stages.SimpleUnGrasp(place_generator, "UnGrasp")
# # [pickAndPlaceTut11]

# # [pickAndPlaceTut12]
# # [initAndConfigPlace]
# # Place container comprises placing, ungrasping, and retracting
# place = stages.Place(simpleUnGrasp, "Place")
# place.eef = eef
# place.object = object_name
# place.eef_frame = "gripper_link"
# # [initAndConfigSimpleUnGrasp]

# # Twist to retract from the object
# retract = TwistStamped()
# retract.header.frame_id = "world"
# retract.twist.linear.z = 1.0
# place.setRetractMotion(retract, 0.03, 0.1)

# # Twist to place the object
# placeMotion = TwistStamped()
# placeMotion.header.frame_id = "gripper_link"
# placeMotion.twist.linear.z = 1.0
# place.setPlaceMotion(placeMotion, 0.03, 0.1)

# # Add the place pipeline to the task's hierarchy
# task.add(place)
# [initAndConfigPlace]
# [pickAndPlaceTut12]

# [pickAndPlaceTut13]
if task.plan():
    task.publish(task.solutions[0])

# avoid ClassLoader warning
del pipeline
del planners
# [pickAndPlaceTut13]

# Prevent the program from exiting, giving you the opportunity to inspect solutions in rviz
time.sleep(3600)
`
Launch file:
`<?xml version="1.0"?>
<launch>
  
  <include file="$(find rake_robotic_arm_moveit_config)/launch/demo.launch">
    <arg name="use_rviz" value="false"/>
  </include>
  <param name="move_group/capabilities" value="move_group/ExecuteTaskSolutionCapability" />

  
  <include file="$(find rake_robotic_arm_moveit_config)/launch/moveit_rviz.launch">
    <arg name="rviz_config" value="$(find moveit_task_constructor_demo)/config/mtc.rviz" />
  </include>
  
  <node pkg="rake_robotic_arm_basic_packages" type="picktest.py" name="picktest" output="screen" />
</launch>`

v4hn, 2024-01-24:
This looks like a bug in the IK solver setup you configured for the joint model group arm_controller, which returns this pose as a successful solution.

We could add another check to validate the solution through forward kinematics in the ComputeIK stage, but the solver is already expected to ensure that whatever it returns with `success == true` is actually successful.

denizkitu, 2024-01-24:
> This looks like a bug in the IK solver setup you configured for the joint model group arm_controller, which returns this pose as a successful solution.
>
> We could add another check to validate the [...]

I have two possible reasons as to why this is happening:

I've added this line at kinematics.yaml because ik solver couldn't find any solutions at all with error "no IK found"
`  position_only_ik: true`

Two: My robotic arm model has 5 degrees of freedom (except the gripper) instead of 6.

v4hn, 2024-01-24:
both reasons are severe issues.

If you are restricted to 5dof in the arm you likely need to provide a custom grasp sampler that considers the reachable pose space of your arm.

rhaschke, 2024-01-24:
I tested your example code. However, for me _all_ IK solutions fail:
```
        0  - ←   0 →   -  0 / compute ik
         32  - ←  32 →   - 32 / Generate Grasp Pose
```
Which IK solver do you use? It looks like the 5dof arm cannot reach any of the requested poses (exactly enough). But, I am also confused that ComputeIK doesn't show any failure markers. Needs further investigation.

denizkitu, 2024-01-24:
> position_only_ik: true

Did you add this line to kinematics.yaml: `position_only_ik: true`
Also you have to use a 5 DoF robot model to reproduce the error.

denizkitu, 2024-01-24:
> I tested your example code. However, for me _all_ IK solutions fail:
>
> ```
>         0  - ←   0 →   -  0 / compute ik
>          32  - ←  32 →   - 32 / Generate Grasp Pose
> ```
>
> Which IK solver do you [...]

Did you add this line to kinematics.yaml: `position_only_ik: true`
Also you have to use a 5 DoF robot model to reproduce the error. I use the KDL ik solver. I tried to install IKFast but i got tons of errors.

rhaschke, 2024-01-25:
KDL is good, particularly for underactuated arms. You could even balance orientation vs. position error. For further investigation, please prepare a zip comprising all relevant packages (URDF, SRDF/moveit_config, your source, exact commands to reproduce.)

denizkitu, 2024-01-27:
> KDL is good, particularly for underactuated arms. You could even balance orientation vs. position error. For further investigation, please prepare a zip comprising all relevant packages (URDF, SRDF/moveit_config, your source, exact commands to reproduce.)

Here they are:
[mtc.zip](https://github.com/ros-planning/moveit_task_constructor/files/14072100/mtc.zip)

`roslaunch rake_robotic_arm_moveit_config mtc_launch.launch`

Let me know if there is anything necessary to run it is missing.

denizkitu, 2024-01-27:
Here they are:
[mtc.zip](https://github.com/ros-planning/moveit_task_constructor/files/14072122/mtc.zip)

You have to run: `roslaunch rake_robotic_arm_moveit_config mtc_launch.launch`

Let me know if anything necessary to run it is missing. Also keep in mind that the line `position_only_ik: true` is added to kinematics.yaml

denizkitu, 2024-01-27:
> KDL is good, particularly for underactuated arms. You could even balance orientation vs. position error. For further investigation, please prepare a zip comprising all relevant packages (URDF, SRDF/moveit_config, your source, exact commands to reproduce.)

https://github.com/ros-planning/moveit_task_constructor/issues/526#issuecomment-1910307961

denizkitu, 2024-01-27:
> KDL is good, particularly for underactuated arms. You could even balance orientation vs. position error. For further investigation, please prepare a zip comprising all relevant packages (URDF, SRDF/moveit_config, your source, exact commands to reproduce.)

Here they are:
[mtc.zip](https://github.com/ros-planning/moveit_task_constructor/files/14072128/mtc.zip)

You have to run: `roslaunch rake_robotic_arm_moveit_config mtc_launch.launch`

Let me know if anything necessary to run it is missing. Also keep in mind that the line `position_only_ik: true` is added to kinematics.yaml file

rhaschke, 2024-01-27:
Thanks. Note, that it will take some time before I can have a deeper look into this.

denizkitu, 2024-01-27:
> Thanks. Note, that it will take some time before I can have a deeper look into this.
Sure! Thanks in advance.

rhaschke, 2024-02-26:
You have configured your KDL-based IK solver with `position_only_ik: true`. This means that orientation is ignored.
Note that the wrist joint position is always reached in your examples. ComputeIK first back-projects the Cartesian goal pose using the IK frame offset you defined, such that the goal is represented w.r.t. the eef link (probably `gripper_link`).
The IK solver then operates on this pose, which is always perfectly reached (in position only as you commanded).
However, as there is the frame offset and a _random_ orientation, the goal frame is offset along that random orientation.

I don't think you can achieve your grasping task with a 5-dof arm.

closed 2024-04-18 by rhaschke