moveit/moveit_task_constructor #530: Deterministic task solutions

captain-yoshi, 2024-02-08:
I have some tasks, that needs to be deterministic.

### Inverse Kinematic
I use `TRAC-IK` because we don't have a closed form solution when accounting for the robot kinematic calibration. Given the same seed and the same timeout, do you think that it is deterministic ? Not had a chance to benchmark this.

### Planners
I use the `Pilz industrial motion planner` to be deterministic. When using a cartesian goals, the IK solver is used to create the trajectory (or maybe just do some checks, not really sure). If the IK solver is not deterministic or the seed is random, then in fact this planner is not really deterministic for cartesian goals...

### MTC
*Assuming I have a IK solver which is deterministic.*

If I want to have the same task solutions, I would need first to change the `ComputeIK` stage to have the same sequence of seeds. I would need to always pass goals in joint space to Pilz and let MTC compute the IK's.

Do you see anything else in the MTC framework that has non-deterministic behavior ? Or maybe you would have a different approach.

rhaschke, 2024-02-09:
I cannot comment on Trac-IK. The Pilz planner should be deterministic, yes. ComputeIK randomly samples a new seed state, but only if the initial one didn't yield as solution. This is the exception and only occurs close to singularities.
Otherwise, I couldn't find non-deterministic behavior in MTC itself. Of course, any sampling-based planner will introduce randomness as well.

captain-yoshi, 2024-02-21:
As it is currently implemented, if I need 8 IK samples, the first one is deterministic but all other calls are random.
```cpp
while (ik_solutions.size() < max_ik_solutions && remaining_time > 0) {
    if (tried_current_state_as_seed) {
	    sandbox_state.setToRandomPositions(jmg);
	    sandbox_state.update();
    }
    tried_current_state_as_seed = true;
    ...
}
```
Even if it would be implemented as you described, it would also occur every time the min distance is not respected, not just singularities.

---
I had the impression that we could get multiple IK solutions with just one call but it is not the case. By using the kinematic plugin, would I be able to request multiple IK solutions ?

rhaschke, 2024-02-22:
> If I need 8 IK samples, the first one is deterministic but all other calls are random.

Yes, that's what I wrote too. Note that all but the first solution will induce strong joint motions, because you will switch between IK solution branches, e.g. from elbow up to elbow down. I always recommend to not do that. Surely, if you need those different solution branches, because different sub tasks operated on different branches, you need to go that route.
Note, that MoveIt's IK API doesn't allow to ask for _all_ solutions in a single call. Hence, we have chosen the random sampling approach in ComputeIK.