moveit/moveit_task_constructor #533: Long computation time in `ComputeIK` makes the task planning time very long

captain-yoshi, 2024-02-22:
This issue might be related with the current limitations of the scheduler #312.

For my 2 cases I have a task which requires one ik solution.

### Trac-IK with `Speed` solver type (returns on the first successful ik)
Everything is good, the task is planned in 294 milliseconds.

[image]

### Trac-IK with `Distance` solver type (returns best solution after a timeout)
Holy macaroni ! It takes 37 seconds for planning the task. It took 2.8 seconds for returning an IK solution, which is normal because it stops after the timeout. While the IK is different from the other case, I don't think it's the bottleneck here.

[image]

---

Is there a way to speed up the process ?

captain-yoshi, 2024-02-23:
Pilz does not check collision when planning the path, so it is normally super fast (unless there is a bug).

I have managed to almost have the exact IK joint configuration as the one who gives 38 seconds to plan. Joint positions: (Left) Distance, (Right) Speed solve type.
<p align="middle">
[image]
  [image]
  
</p>

### Trac-IK with `Distance` solver type
Same as before.
[image]

### Trac-IK with `Speed` solver type
With almost the exact IK cost and joint configuration. It takes 11 seconds but I have 8 seconds of trying to find 32 IK's and the rest is for planning the combinations. Which gives about 3 seconds to plan multiple sub-tasks.
[image]

---

As you can see, when the `compute_ik` stage does not return fast (Distance solve type) it takes 36 seconds to plan 1 trajectory. When the `compute_ik` stage returns fast (Speed solve type) it takes 3 seconds to plan multiple trajectories.

captain-yoshi, 2024-02-23:
Oh well this one's on me...

Well if I set the IK to return after a timeout, this will also affect the other LIN Pilz planners of the same group because they rely on the IK solver. Mystery solved.

rhaschke, 2024-02-24:
Knowing that you use the Pilz planner for subsequent stages, this would have been my next hint :wink:
Can we close this issue?

closed 2024-02-25 by captain-yoshi