moveit/moveit_task_constructor #548: Meaning of num_planning_attempts

leander2189, 2024-03-20:
After some reverse engineering of the source code, I've noticed that the _num_planning_attempts_ parameter in **pipeline_planner** is  setting the number of threads used by OMPL planner. 

Why this change in parameter naming? In my opinion, this approach (combined with the lack of documentation on motion planners) makes planner configuration harder.

rhaschke, 2024-03-20:
We kept the name of the corresponding member in `MotionPlanRequest` message:
https://github.com/ros-planning/moveit_task_constructor/blob/b2e8ae129670a1888721a0ef81c7ca86243d537d/core/src/solvers/pipeline_planner.cpp#L149
That this parameter maps onto the number of parallel planning threads is OMPL-specific. 
I don't see that there was another parameter name used for that number in MoveIt.

closed 2024-03-20 by rhaschke

leander2189, 2024-03-20:
I see, thanks!