moveit/moveit_task_constructor #550: Raspberry Pi 4b install moveit appears “cannot install moveit/__ini__.py from different namespace packages”

zhazhaxia006, 2024-03-24:
dpkg: 处理归档 /var/cache/apt/archives/ros-noetic-moveit-task-constructor-core_0.1.1-1buster.20230216.092007_arm64.deb (--unpack)时出错：
 正试图覆盖 /opt/ros/noetic/lib/python3/dist-packages/moveit/__init__.py，它同时被包含于软件包 ros-noetic-moveit-core 1.1.11-1buster.20230216.041139
dpkg-deb: 错误: 粘贴 子进程被信号(断开的管道) 终止了

I saw someone upgraded to 0.1.3, but the installation package of this version cannot be found in the Raspberry Pi 4b system.

the linux system is Raspberry Pi OS(debain 10 buster)

Please help me solve the problem

rhaschke, 2024-03-30:
Sorry, but the description does not yet explain, what you did and what failed.

zhazhaxia006, 2024-03-31:
I entered sudo apt-get install ros-noetic-moveit-* in the terminal of the Raspberry Pi operating system. Then the system starts to download the software installation package. When installing to ros-noetic-moveit-task-constructor-core_0.1.1-1buster.20230216.092007_arm64.deb, it prompts that the file decompression failed. I saw that when Ubuntu installed this software, version 0.1.3 was installed. It may be that the software source of the buster system has not been updated.

rhaschke, 2024-03-31:
Indeed the binary packages for Buster are outdated. Focal has the newer ones - also as arm64:

[image]

You can also try to `--force` install the package. The error message indicates that a file is to be installed from two different packages. However, in this case, the file content is the same.