moveit/moveit_task_constructor #551: Change OMPL planner by default

nachoorp, 2024-03-25:
Hey everyone, 

Im new with MTC and Im trying to test different OMPL planners for my pick and place task (similar to pick and place tutorial in ros2 humble). I want to change the default planner RRTconect with other like PRM for example . I tried to changed it on the ompl_interface.cpp but seems to keep using the RRTconect. Any help could be useful thank you all!

leander2189, 2024-03-28:
MTC is using the "Pipeline planner", that is, whatever you have configured for the move group. You should have an ompl_planner.yaml where RRTConnect is setup, and this file is fed to the move_group node in the launcher. If you want to try other planners, change the settings of that yaml file

LeroyR, 2024-03-29:
`setPlannerid` should do that. Your ids are defined in the mentioned `ompl_planner.yaml`

```
auto pipeline = std::make_shared<solvers::PipelinePlanner>();
pipeline->setPlannerId("PRM");
```

closed 2024-03-31 by nachoorp