moveit/moveit_task_constructor #555: Vaccum Gripper support

Cavalletta98, 2024-04-05:
Hi, i'm using an ufactory 850 robot with a vaccum gripper. I'm facing some problems by defining the stage for this gripper using this tutorial: https://moveit.picknik.ai/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html
I tested with a normal gripper, and it works. With the vaccum one,the stages are not visible into Rviz. So, i cannot perform the Pick and Place task
Is it the gripper supported by Task Constructor?
Do you have a simple code to try it?

I'm using ROS2 Humble

Thanks

rhaschke, 2024-04-05:
What do you mean by "the stages are not visible in rviz"? Each stage should be shown in the stage list.

I guess you want to have stages to turn on and off vacuum? Such external device switching is not directly supported by MTC (and MoveIt), which only handle joint actuation natively. To work around this limitation, you could:
1. Execute solutions partially, i.e. everything until turning vacuum on, then turn vacuum on, and execute the remaining solution trajectories
2. Introduce a virtual joint to operate the vacuum via a custom controller.

Cavalletta98, 2024-04-05:
Executing the debug operation i can noticed that there is a problem in the stage that allow the collision into the pick serial container. In fact, i can notice a warning  in Rviz: no messages received in the task monitor. Below, i am going to send you a screenshot of the warning
[image]
Regarding the thing of "the stages are not visible in rviz", this is what it is happening:
[image]

rhaschke, 2024-04-05:
You probably listen to the wrong topic. If no messages are received, the task tree cannot show anything.

rhaschke, 2024-04-05:
Is there any configuration issue reported on the command line of the MTC node?

Cavalletta98, 2024-04-11:
Hi, we are justing try to get worked the task constructor with our robot and the normal gripper. Once it will works, we go back to the vacuum gripper. Thanks for the support. I will reopen the issue if necessary

closed 2024-04-11 by Cavalletta98