moveit/moveit_task_constructor #556: Add/Modify the `PropertyMap` stage on new solution

captain-yoshi, 2024-04-06:
I would like to be able to add a property (which cannot be computed in advance) after a solution is found. E.g. compute the transform between a robot link and a attached collision object subframe, which would be needed by a cartesian controller.

Here are my 4 approaches to fix this. You might have other ideas.

## Alpha
I could go down the `PredicateFilter` route and add a new wrapper that add's a callback for changing the `PropertyMap` on new solution.  

## Beta
It might be interesting to add an optional callback to the NoOp stage to #534.

## Gamma
It would be nice if a callback could be set to any stage.

## Delta
One could add the callback directly in the `PropertyMap`.  As I'm writing this, I think this is the cleanest solution. No change in the API and the callback can be triggered only for the chosen solution (post-planning), as opposed to be triggered on every possible solutions while planning.

rhaschke, 2024-04-07:
Probably adding a post-processing callback somewhere is the way to go. I don't yet get the idea behind your last proposal: What do you mean by adding a callback in PropertyMap? Storing a "post-processing" callback as a property?

captain-yoshi, 2024-04-07:
Yes storing a "post-processing" callback as a property.

rhaschke, 2024-04-08:
Then we are on the same page. This is a reasonable approach.