moveit/moveit_task_constructor #557: Trajectory execution with status ABORTED

Gabriella7778, 2024-04-10:
I am trying to do a pick and place task following the moveit tutorial and trying to adapt it to the uf850 ufactory manipulator. I am using the humble version of ROS2. The planning works and i can see all the solutions on Rviz but the execution fails as you can see in the following lines of code. Can you help me?

root@gabriella-ThinkPad-E14-Gen-2:/home/dev_ws# ros2 launch pick_place uf850launch.launch.py add_gripper:=true add_realsense_d435i:=true
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-04-10-14-53-42-221370-gabriella-ThinkPad-E14-Gen-2-2630741
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [2630749]
[INFO] [gzclient-2]: process started with pid [2630751]
[INFO] [robot_state_publisher-3]: process started with pid [2630753]
[INFO] [spawn_entity.py-4]: process started with pid [2630755]
[INFO] [rviz2-5]: process started with pid [2630757]
[INFO] [static_transform_publisher-6]: process started with pid [2630759]
[INFO] [move_group-7]: process started with pid [2630761]
[INFO] [TaskConstructor-8]: process started with pid [2630765]
[static_transform_publisher-6] [WARN] [1712760824.583475201] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-6] [INFO] [1712760824.592349024] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-6] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-6] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-6] from 'world' to 'link_base'
[rviz2-5] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[robot_state_publisher-3] [INFO] [1712760824.643317653] [robot_state_publisher]: got segment camera_color_frame
[robot_state_publisher-3] [INFO] [1712760824.644234809] [robot_state_publisher]: got segment camera_color_optical_frame
[robot_state_publisher-3] [INFO] [1712760824.644799406] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher-3] [INFO] [1712760824.645197641] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher-3] [INFO] [1712760824.645551876] [robot_state_publisher]: got segment camera_left_ir_frame
[robot_state_publisher-3] [INFO] [1712760824.645989781] [robot_state_publisher]: got segment camera_left_ir_optical_frame
[robot_state_publisher-3] [INFO] [1712760824.646470358] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-3] [INFO] [1712760824.646997800] [robot_state_publisher]: got segment camera_right_ir_frame
[robot_state_publisher-3] [INFO] [1712760824.647459520] [robot_state_publisher]: got segment camera_right_ir_optical_frame
[robot_state_publisher-3] [INFO] [1712760824.647919844] [robot_state_publisher]: got segment left_finger
[robot_state_publisher-3] [INFO] [1712760824.648329533] [robot_state_publisher]: got segment left_inner_knuckle
[robot_state_publisher-3] [INFO] [1712760824.648739292] [robot_state_publisher]: got segment left_outer_knuckle
[robot_state_publisher-3] [INFO] [1712760824.649123070] [robot_state_publisher]: got segment link1
[robot_state_publisher-3] [INFO] [1712760824.649439032] [robot_state_publisher]: got segment link2
[robot_state_publisher-3] [INFO] [1712760824.649686340] [robot_state_publisher]: got segment link3
[robot_state_publisher-3] [INFO] [1712760824.649931343] [robot_state_publisher]: got segment link4
[robot_state_publisher-3] [INFO] [1712760824.650220975] [robot_state_publisher]: got segment link5
[robot_state_publisher-3] [INFO] [1712760824.650575210] [robot_state_publisher]: got segment link6
[robot_state_publisher-3] [INFO] [1712760824.650956334] [robot_state_publisher]: got segment link_base
[robot_state_publisher-3] [INFO] [1712760824.651251902] [robot_state_publisher]: got segment link_eef
[robot_state_publisher-3] [INFO] [1712760824.651526448] [robot_state_publisher]: got segment link_tcp
[robot_state_publisher-3] [INFO] [1712760824.651914975] [robot_state_publisher]: got segment right_finger
[robot_state_publisher-3] [INFO] [1712760824.652295889] [robot_state_publisher]: got segment right_inner_knuckle
[robot_state_publisher-3] [INFO] [1712760824.652644188] [robot_state_publisher]: got segment right_outer_knuckle
[robot_state_publisher-3] [INFO] [1712760824.652980054] [robot_state_publisher]: got segment world
[robot_state_publisher-3] [INFO] [1712760824.653259420] [robot_state_publisher]: got segment xarm_gripper_base_link
[move_group-7] [INFO] [1712760824.669778837] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[move_group-7] [INFO] [1712760824.669948971] [moveit_robot_model.robot_model]: Loading robot model 'UF_ROBOT'...
[move_group-7] [INFO] [1712760824.669973345] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-5] amdgpu: os_same_file_description couldn't determine if two DRM fds reference the same file description.
[rviz2-5] If they do, bad things may happen!
[spawn_entity.py-4] [INFO] [1712760824.820594584] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1712760824.821116228] [spawn_entity]: Loading entity published on topic robot_description
[spawn_entity.py-4] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/qos.py:307: UserWarning: DurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL is deprecated. Use DurabilityPolicy.TRANSIENT_LOCAL instead.
[spawn_entity.py-4]   warnings.warn(
[spawn_entity.py-4] [INFO] [1712760824.824379492] [spawn_entity]: Waiting for entity xml on robot_description
[spawn_entity.py-4] [INFO] [1712760824.826925276] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1712760824.828617880] [spawn_entity]: Waiting for service /spawn_entity
[rviz2-5] [INFO] [1712760824.909753541] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1712760824.910634519] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-5] [INFO] [1712760824.926822959] [rviz2]: Stereo is NOT SUPPORTED
[move_group-7] [INFO] [1712760824.985959944] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-7] [INFO] [1712760824.986141741] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/joint_states' for joint states
[move_group-7] [INFO] [1712760824.986654655] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/joint_states'
[move_group-7] [INFO] [1712760824.987209195] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/move_group/planning_scene_monitor' for attached collision objects
[move_group-7] [INFO] [1712760824.987231614] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-7] [INFO] [1712760824.988156522] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/move_group/publish_planning_scene'
[move_group-7] [INFO] [1712760824.988180408] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-7] [INFO] [1712760824.988745634] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-7] [INFO] [1712760824.989319240] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[rviz2-5] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[... 394 lines cut ...]
[gzserver-1] [INFO] [1712760865.971864293] [uf850_gripper_traj_controller]: Received new action goal
[gzserver-1] [INFO] [1712760865.971920027] [uf850_gripper_traj_controller]: Accepted new action goal
[move_group-7] [INFO] [1712760865.972018573] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: uf850_gripper_traj_controller started execution
[... cut ...]
[move_group-7] [WARN] [1712760868.423133379] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'uf850_traj_controller' failed with error PATH_TOLERANCE_VIOLATED: Aborted due to path tolerance violation
[move_group-7] [WARN] [1712760868.423320274] [moveit_ros.trajectory_execution_manager]: Controller handle uf850_traj_controller reports status ABORTED
[move_group-7] [INFO] [1712760868.423336408] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...

rhaschke, 2024-04-10:
This is a duplicate of #478. See https://github.com/ros-planning/moveit_task_constructor/issues/478#issuecomment-1643866791 for a solution.

closed 2024-04-10 by rhaschke

rhaschke, 2024-04-11:
The message states, that the gripper controller is not running. So, check your setup!

Gabriella7778, 2024-04-11:
i notice that if i put the object in another position, the gripper controller run and the trajectory execution is in aborted status. 
I don't understand the reason. Below, the error that i have now.

[gzserver-1] [ERROR] [1712831752.517078377] [tolerances]: State tolerances failed for joint 5:
[gzserver-1] [ERROR] [1712831752.517114066] [tolerances]: Position Error: -1.052823, Position Tolerance: 1.000000
[gzserver-1] [WARN] [1712831752.517127545] [uf850_traj_controller]: Aborted due to state tolerance violation
[gzserver-1] [INFO] [1712831752.660170586] [uf850_gripper_traj_controller]: Goal reached, success!
[move_group-7] [INFO] [1712831752.665429564] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'uf850_gripper_traj_controller' successfully finished
[move_group-7] [WARN] [1712831752.665663742] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'uf850_traj_controller' failed with error PATH_TOLERANCE_VIOLATED: Aborted due to path tolerance violation
[move_group-7] [WARN] [1712831752.665744897] [moveit_ros.trajectory_execution_manager]: Controller handle uf850_traj_controller reports status ABORTED
[move_group-7] [INFO] [1712831752.665768783] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[TaskConstructor-8] [ERROR] [1712831752.761634808] [moveit_task_constructor_executor_104051241356768]: Goal was aborted or canceled
[TaskConstructor-8] [ERROR] [1712831752.767981285] [TaskConstructor]: Task execution failed

Thanks.