moveit/moveit_task_constructor #558: CI: rosdep installation of ros-humble-rmw-connextdds fails

rhaschke, 2024-04-14:
@ros-planning/moveit-maintainers: do you know how to fix the following rosdep install failure in humble-release:
```
  executing command [apt-get install -y -qq ros-humble-rmw-connextdds]
  rti-connext-dds-6.0.1: RTI license agreement for non-commercial and pre-commercial use NOT accepted.
  E: Sub-process /usr/bin/dpkg returned an error code (1)
  ERROR: the following rosdeps failed to install
    apt: command [apt-get install -y -qq ros-humble-rmw-connextdds] failed
```
Obviously, the installation of the DDS middleware requires explicit confirmation to the license agreement. The option `-y` doesn't work as expected, maybe because one needs to scroll through several pages first.
Where does this requirement come from?

sea-bass, 2024-04-14:
Don't have the full background, but if this has to do with a CI image we can control, there seems to be an environment variable that can be set.

https://stackoverflow.com/a/56108565

rhaschke, 2024-04-15:
Thanks for the hint! I filed https://github.com/ros-planning/moveit2/pull/2790 to fix CI.

rhaschke, 2024-04-15:
P.S.: I couldn't figure out yet, which dependency pulls in `rmw-connextdds`. MTC and MoveIt don't explicitly require it.

closed 2024-04-15 by rhaschke