moveit/moveit_task_constructor #561: Pick/Place demo fails when executing in humble

DaniGarciaLopez, 2024-04-18:
I was experiencing problems with `moveit_collision_detection` in my setup and decided to try the pick/place demo. Planning is successful, but when executing, there is a collision between the gripper and the object, and the execution is preempted. Additionally, there is an error regarding an empty JointState message.

In my setup, I also encounter a collision of the attached object with itself (Found a contact between 'object' (type 'Object') and 'object' (type 'Robot attached')), which should have already been fixed [here](https://github.com/ros-planning/moveit/pull/1899).

I'm building moveit and mtc from source using the latest humble branch. Any idea what's happening?

```
[move_group-4] [INFO] [1713480832.513559786] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-4] [INFO] [1713480832.513591921] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480832.513601982] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [ERROR] [1713480832.513651890] [moveit_robot_state.conversions]: Found empty JointState message
[move_group-4] [INFO] [1713480832.513866048] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480832.513871182] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[ros2_control_node-5] [INFO] [1713480832.514144977] [panda_hand_controller]: Received & accepted new action goal
[move_group-4] [INFO] [1713480832.564803810] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480832.564822959] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480832.564888270] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to panda_arm_controller
[ros2_control_node-5] [INFO] [1713480832.565200435] [panda_arm_controller]: Received new action goal
[ros2_control_node-5] [INFO] [1713480832.565265437] [panda_arm_controller]: Accepted new action goal
[move_group-4] [INFO] [1713480832.565467992] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: panda_arm_controller started execution
[move_group-4] [INFO] [1713480832.565484011] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-5] [INFO] [1713480834.293341539] [panda_arm_controller]: Goal reached, success!
[move_group-4] [INFO] [1713480834.316251404] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'panda_arm_controller' successfully finished
[move_group-4] [ERROR] [1713480834.316321089] [moveit_robot_state.conversions]: Found empty JointState message
[move_group-4] [INFO] [1713480834.316470961] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480834.316492694] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480834.316551981] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to panda_arm_controller
[ros2_control_node-5] [INFO] [1713480834.316678166] [panda_arm_controller]: Received new action goal
[ros2_control_node-5] [INFO] [1713480834.316734533] [panda_arm_controller]: Accepted new action goal
[move_group-4] [INFO] [1713480834.316795278] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: panda_arm_controller started execution
[move_group-4] [INFO] [1713480834.316798391] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-5] [INFO] [1713480834.983417550] [panda_arm_controller]: Goal reached, success!
[move_group-4] [INFO] [1713480835.017084896] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'panda_arm_controller' successfully finished
[move_group-4] [ERROR] [1713480835.017135495] [moveit_robot_state.conversions]: Found empty JointState message
[move_group-4] [INFO] [1713480835.017204221] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480835.017212213] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [ERROR] [1713480835.017223539] [moveit_robot_state.conversions]: Found empty JointState message
[move_group-4] [INFO] [1713480835.017249949] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480835.017252666] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480835.017314011] [moveit_ros.plan_execution]: Trajectory component '7/19' is invalid
[move_group-4] [INFO] [1713480835.017322304] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'object' (type 'Object') and 'panda_leftfinger' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[move_group-4] [INFO] [1713480835.017324905] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[move_group-4] [INFO] [1713480835.017327567] [moveit_ros.plan_execution]: Upcoming trajectory component '7/19' is invalid
[move_group-4] [INFO] [1713480835.017329728] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1713480835.017332353] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[ros2_control_node-5] [INFO] [1713480835.017435359] [panda_hand_controller]: Received & accepted new action goal
[move_group-4] [INFO] [1713480835.023644920] [moveit_ros.plan_execution]: Trajectory component '7/19' is invalid
[move_group-4] [INFO] [1713480835.023658700] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'object' (type 'Object') and 'panda_leftfinger' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[move_group-4] [INFO] [1713480835.023661704] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[move_group-4] [INFO] [1713480835.023665009] [moveit_ros.plan_execution]: Trajectory component '7/19' is invalid after scene update
[move_group-4] [INFO] [1713480835.023666826] [moveit_ros.plan_execution]: Stopping execution because the path to execute became invalid(probably the environment changed)
[move_group-4] [INFO] [1713480835.023675637] [moveit.simple_controller_manager.gripper_controller_handle]: Cancelling execution for panda_hand_controller
[ros2_control_node-5] [INFO] [1713480835.023768156] [panda_hand_controller]: Got request to cancel goal
[ros2_control_node-5] [INFO] [1713480835.023798970] [panda_hand_controller]: Canceling active action goal because cancel callback received.
[move_group-4] [INFO] [1713480835.023931403] [moveit_ros.trajectory_execution_manager]: Stopped trajectory execution.
[move_group-4] [INFO] [1713480835.067635284] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status PREEMPTED ...
```

Cavalletta98, 2024-04-23:
Have you solved It?

DaniGarciaLopez, 2024-04-24:
No, I'm still looking for a solution

Cavalletta98, 2024-05-02:
Write here when you found a solution. Thanks

rhaschke, 2024-05-03:
The default settings from `humble` branch (commit 085735196ca733f086826c3df7f11b98db4d0587) disable automatic execution:
https://github.com/moveit/moveit_task_constructor/blob/085735196ca733f086826c3df7f11b98db4d0587/demo/src/pick_place_demo_parameters.yaml#L1-L4
If I manually trigger execution from within rviz then, execution works just fine.
However, when enabling automatic execution, I do observe the exact same issue reported here as well.
Need to figure out why...

@DaniGarciaLopez, can you confirm this behavior? Did you use automatic execution to trigger the failure?

DaniGarciaLopez, 2024-05-06:
> @DaniGarciaLopez, can you confirm this behavior? Did you use automatic execution to trigger the failure?

Yes, @rhaschke, I found the issue when setting the `execute` parameter to `true`, so it uses [task_->execute(*task_->solutions().front())](https://github.com/moveit/moveit_task_constructor/blob/085735196ca733f086826c3df7f11b98db4d0587/demo/src/pick_place_task.cpp#L456C19-L456C62). Our project experiences the same problem, and we call the ExecuteTaskSolution action server directly, so maybe there is no problem within the `task_->execute` function but in the action server itself.

> If I manually trigger execution from within rviz then, execution works just fine.

I didn't know that it was possible to trigger the execution of a MTC task from Rviz. Using the MotionPlanning display and then clicking on execute? But how do you pass the task to it?

rhaschke, 2024-05-06:
MTC has its own `Motion Planning Tasks` display and panel:

[image]

Select a top-level solution and click the _execute_ button in the top-right corner.
It would be interesting to know whether execution works for you from there as well.

DaniGarciaLopez, 2024-05-06:
You're right! I've been using MTC for over a year, and I still can't believe I didn't pay attention to that button before.

I can confirm that with the rviz panel I cannot reproduce the error mentioned here, and the execution ends successfully.

rhaschke, 2024-05-08:
I pushed another commit to the humble branch reverting a fix from the master branch that is not yet supported by humble's MoveIt. This resolves the issue for me.

DaniGarciaLopez, 2024-05-08:
It's also working on my end. Thanks @rhaschke!

closed 2024-05-08 by DaniGarciaLopez

chiadikav, 2025-05-24:
I'm still having issues with this on humble. When I press execute in rviz, nothing happens

rhaschke, 2025-05-24:
@chiadikav: Did you correctly configure the task execution capability for move_group:
`<arg name="capabilities" default="move_group/ExecuteTaskSolutionCapability"/>`

chiadikav, 2025-05-24:
What file is that in? I have just been following all of the tutorials on the moveit website

chiadikav, 2025-05-24:
Ahh ok I understand now. I had to add that to my launch file, since the mtc_demo launch file also uses it. Thanks. I'm realizing that there's quite a few errors that slipped through the crack when they published the walk through on their website lol

rhaschke, 2025-05-25:
What other issues did you notice? Please feel free to fix the shortcomings of the tutorial by contributing a PR at https://github.com/moveit/moveit2_tutorials.

chiadikav, 2025-06-01:
In the demo for using the task constructor, we create the object's start pose wrt the world and then we create the goal pose wrt to the object itself. I could be missing something, but that seems to be an easy way to lead to confusion. I've also found that most of the included files are .hpp, but I can build them unless I include them as .h files. I am quite new to ROS2 and c++ in general, so there's a good chance my confusion stems from my lack of knowledge. I've also never contributed to any pen source work before, so I'm not too sure of what that process looks like.