moveit/moveit_task_constructor #563: ROS2 MTC Python API

matthias88, 2024-04-27:
Hey there, 

i am trying to port my ROS1 noetic project to ROS2 humble. The only show-stopper is, that there is no working Python API for Moveit task constructor, am I right? At least I wasn't able to find and work with it. 

In the ROS2 and Humble branches the Python example scripts are not updated and modules from ROS1 are imported.

Is there a workaround, except using CPP?

Many thanks

rhaschke, 2024-04-28:
The Python API is on my todo list for a long time, but I couldn't find time for it yet. Remember that this development is an voluntary action.

matthias88, 2024-04-28:
> The Python API is on my todo list for a long time, but I couldn't find time for it yet. Remember that this development is an voluntary action.

Thank you for the quick response. I am aware that this library is a voluntary action and I want to tell you what a great package you have created with MTC. Is there already a ROS2 pybind preparatory work on which I could build? Is it enough to adjust the namespaces/headers of `https://github.com/ros-planning/moveit_task_constructor/tree/master/core/python/bindings,` or are there any breaking changes to consider?

karanchahal, 2024-07-12:
Hi I was wondering the same thing, so ROS2 is not supported by Move it task constructor, correct ?

rhaschke, 2024-07-13:
I just finished porting the Python API. As the node needs to access ROS parameters, one needs to run it via a launch script:
`ros2 launch moveit_task_constructor_demo run.launch.py exe:=cartesian.py`

It also needs the latest version of [py_binding_tools](https://github.com/ros-planning/py_binding_tools.git).

closed 2024-07-13 by rhaschke