moveit/moveit_task_constructor #571: Path constraints and other stages

samraise, 2024-05-21:
I was wondering if there was a way I could structure a task so that it only moves certain joints in different stages. For example, lets say I have a 6 DOF arm with the current task:

←→ Current state
→← Connect
←← Move relative
←→ Compute IK

Now instead I want to move joints 1-3 and then move joints 4-6. Is there a way I could structure the task like this?

←→ Current state
→← Connect (path constraint for joints 4-6)
←→ [Not given state generator]
→← Connect (path constraint for joints 1-3)
←← Move relative
←→ Compute IK

after adding something in connect that allows for relative joint constraints. I understand that a state that isn't given would go against the idea of how a generator should work. I've tried just modifying a single connect with a path constraint but it has a mismatch between the goal constraints generated by the Move relative state coming after it (which makes sense).

I think I could as separate the arm into 2 different planning groups to use at different steps but I'm not sure how I could use both groups in the same tree.

rhaschke, 2024-05-22:
As you correctly pointed out, a generator stage needs to come up with a well-defined robot state.
To solve, your task, you could simply split the Connect task onto two move groups: Define groups for joints 1-3 and 4-6 and then call Connect like this:
```c++
stages::Connect::GroupPlannerVector planners = { { "group46", pipeline }, { "group13", pipeline } };
auto connect = std::make_unique<stages::Connect>("connect", planners);
connect->setProperty("merge_mode", Connect::MergeMode::SEQUENTIAL);
```

samraise, 2024-05-22:
That makes sense! Thanks for the help. What would be the best way to configure those properties with the Python API?

rhaschke, 2024-05-22:
The Python API works the same as the C++ API for this.

samraise, 2024-05-22:
Right now I'm running into this error setting the property:

`            self.stage.properties.update(
                {
                    "merge_mode": stages.MergeMode.SEQUENTIAL,
                }
            )
`

gives me

`TypeError: MergeMode is not a ROS message type`

samraise, 2024-05-23:
Awesome, thanks for the quick fix

closed 2024-05-23 by samraise