moveit/moveit_task_constructor #572: Grasp generator issue

ron007d, 2024-05-22:
Hi all I have been trying to understand the grasp generator.
Trying to grasp a box using # finger gripper but getting this grasp generations :point_down: 

[image]

where I want this grasp generation to get generated and executed :point_down:  , Some time its generated successfully and working but sometime its generating :point_up: this and executing. for which its not able to grab the object correctly in simulation.

[image]

```python
    grasp_generator = stages.GenerateGraspPose("Generate Grasp Pose")
    grasp_generator.angle_delta = 1.5708
    grasp_generator.pregrasp = "open"
    grasp_generator.grasp = "close"
    grasp_generator.setMonitoredStage(task["current"]) 
    grasp_generator.object = object_name
```
Thanks

rhaschke, 2024-05-22:
The `GenerateGraspPose` stage just creates Cartesian poses around the object's frame by rotating about the z-axis by default. Additionally some properties, like pregrasp and grasp posture for fingers are set, which can be used by _neighbouring_ stages to actually actuate the fingers.

ron007d, 2024-05-22:
Thanks @rhaschke  I will check it out, first I thought it was issue with the object size but then its showed the same for bigger object, 

there was a python doc for moveit task constructor, but it was showing 404,  can you help me with the link for that page ?

rhaschke, 2024-05-22:
The docs are available here: https://moveit.github.io/moveit_task_constructor/api.html

ron007d, 2024-05-23:
I figured out that jupyter notebook don't have the sourced directory thus **it was not able to get the model info**, now its fetching error on certain grasp,
- added gripper group name in grasp
- added object name in grasp
- added `source ...devel.bash` info the jupyter server.

[image]

closed 2024-05-23 by ron007d