moveit/moveit_task_constructor #574: WIP: python API for ROS2

rhaschke, 2024-05-23:
Hi @JafarAbdi,

I'm currently working on reenabling the Python API for the `ros2` branch of MTC, which you initially disabled when porting to ROS2. Unfortunately, I'm stuck with several issues. One of them is, that the `RobotModelLoader` cannot access the `robot_description_semantic` parameter, although it can access `robot_description`:

```
[ERROR] [1716476943.051554493] [mtc_tutorial]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.
Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
         at line 715 in ./src/model.cpp
[ERROR] [1716476943.053114764] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[ERROR] [1716476943.053302555] [moveit_task_constructor.task]: task pipeline: received invalid robot model
[ERROR] [1716476943.053314726] [moveit_exceptions.exceptions]: Task failed to construct RobotModel
Exception thrown.
Traceback (most recent call last):
  File "/homes/rhaschke/src/ros/src/moveit/mtc/demo/scripts/cartesian.py", line 30, in <module>
    task.loadRobotModel(node)
RuntimeError: Task failed to construct RobotModel
```

Any idea what is different between these parameters? Shouldn't the `rclcpp::Node` either be able to access both of them or none?

rhaschke, 2024-05-29:
Can any MoveIt2 @moveit/maintainers help on this issue? Any reason why `robot_description` is found, but `robot_description_semantic` not?

henningkayser, 2024-05-29:
The error is about the subscription which probably succeeds for the robot_description but not for the srdf. I assume none of the original parameters make it through the python bindings or launch system. This sounds oddly familiar with https://github.com/moveit/moveit2/issues/2219 and https://github.com/moveit/moveit2/pull/2393. What config are you working on?

Abishalini, 2024-05-29:
When you launch `move_group` node, did you set this parameter to True? - https://github.com/moveit/moveit2/blob/82ef3fc8dbf6770f309557924634ce0bc4ea6af8/moveit_configs_utils/moveit_configs_utils/launches.py#L216

rhaschke, 2024-05-29:
> What config are you working on?

I'm running the demo launch file of MTC: `ros2 launch demo/launch/demo.launch.py`
and then an example binary or python script:
- `ros2 run moveit_task_constructor_demo cartesian`
- `python3 demo/scripts/cartesian.py`

@Abishalini, your hint resolved the SRDF issue. Thanks!
Next issue is that the kinematics parameters are not found:
`[WARN] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!`

Is there any way to run simple binaries/scripts and pass the corresponding parameters? 
I don't want to provide a `.launch.py` for each and every example.

Wouldn't it make sense to fetch those common parameters from the `move_group` node using `ParametersClient` classes (see https://answers.ros.org/question/340600/how-to-get-ros2-parameter-hosted-by-another-node)? Did you consider that option?

MarqRazz, 2024-05-29:
> Next issue is that the kinematics parameters are not found:
[WARN] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!

I've been running into this same issue with some C++ examples I have. I kinda like the idea of getting them from the `move_group` parameter namespace, but what if my node is built on `moveit_cpp`? We could provide MTC a namespace that it could search for `robot_description_kinematics` or it would probably be best if MoveIt published the info on a topic like it does with `robot_description_semantic`.

matthias88, 2024-06-22:
I highly appreciate your effort for reenabling the Python API for ROS2. :-) Have you already progressed? Maybe this helps for getting a list of parameters from another node `https://answers.ros.org/question/327280/ros2-rclpy-list-parameters-from-node/`. If you need anybody for testing (humble, jazzy), I am happy to help.

rhaschke, 2024-07-12:
I had to resort to launch file loading.