moveit/moveit_task_constructor #587: Pybind11 smart_holder/v2.12 branch bug

samraise, 2024-06-17:
I can't build the moveit_task_constructor_core package because when I have the pybind11 submodule on the [smart holders branch](https://github.com/moveit/moveit_task_constructor/blob/8fa66a6d2c268a2a41e938a435d7701969b762dc/.gitmodules#L4) I can't build because of [these additions](https://github.com/moveit/moveit_task_constructor/blob/8fa66a6d2c268a2a41e938a435d7701969b762dc/core/python/bindings/src/core.cpp#L83) from 8fa66a6. I can't build on the master/v2.12 because of [these includes](https://github.com/moveit/moveit_task_constructor/blob/8fa66a6d2c268a2a41e938a435d7701969b762dc/core/python/bindings/src/module.cpp#L35).

rhaschke, 2024-06-18:
Looks like your pybind11 submodule is not really on the correct smart holder branch. Please double check!

rhaschke, 2024-07-11:
Closing issue due to inactivity.

closed 2024-07-11 by rhaschke