moveit/moveit_task_constructor #601: Action call in a stage

karanchahal, 2024-08-07:
Hi, I want to run a custom stage called attach object which runs things externally to MTC/Move it.

So I want to run this action call when task is being executed and robot is in the correct position, but when I add this stage into MTC (the stage is inherited from PropogatingForward stage), in the task.plan() stage it is being executed and the rest of my stack fails since robot is not in the correct position.

How do I add the constraint of only running code in the custom stage during task.execute() ?

rhaschke, 2024-08-08:
Currently, MTC doesn't allow for running code during task execution: it is primarily a framework planning trajectories. Side-effects cannot be transmitted to the `move_group` node for execution.
If you just want to attach a collision object to the robot, use the `ModifyPlanningScene` stage.

blooop, 2024-12-06:
I have a vacuum gripper and I'd like to call a ros service to turn on/off the vacuum gripper during execution. From the sounds of it, MTC doesn't support calling the service at the appropriate point during the execution?  Are there any workarounds for using MTC with a vacuum gripper?

Thanks

rhaschke, 2024-12-06:
Yes, that's currently not supported as the MTC node just _sends_ the planned trajectory for execution to the main `move_group` node.
I have some code lying around, changing that logic and directly using MoveIt's execution manager within the MTC node to execute the trajectory _with side effects_. However, I never found time to finalize and merge this into MoveIt+MTC.

blooop, 2024-12-06:
Thanks for the quick reply.

As a horrible horrible workaround, what about having a fake gripper joint that MTC activates as normal and then the gripper node subscribes to the state of the fake gripper joint to turn the vac on and off.

matthias88, 2024-12-06:
An approach that works is to execute the sub-trajectories up to the point where you need to call the gripper service, and then execute the remaining sub-trajectories afterward.

rhaschke, 2024-12-09:
Both of these approaches will work.

zhenjing, 2025-04-02:
I create multi sub-trajectories, than execute sub-trajectories . However it crash.

`       int sulution = 0;
        SolutionSequence pick_solution;  
        SolutionSequence place_solution;  
        SolutionSequence return_solution; 

        auto solutions = dynamic_cast<const SolutionSequence*>(task_->solutions().front().get());
        for(auto sub_solution : solutions->solutions()) {
            auto stage = sub_solution->creator();
            if(0 == sulution){
                pick_solution.push_back(*sub_solution);
            } else if(1 == sulution){
                place_solution.push_back(*sub_solution);
            } else {
                return_solution.push_back(*sub_solution);
            } 
            
            if(APPROACH_STAGE == stage->name() || LOWER_STAGE == stage->name()){
                sulution ++;
            }
        }

        execute_result = task_->execute(pick_solution);`

gdb stack:
`#0  0x00007ffff7c5ac64 in moveit::task_constructor::Stage::name[abi:cxx11]() const () from /home/rw/zhenjing/git/BagageBusiness/jaka_ws/install/moveit_task_constructor_cor
#1  0x00007ffff7c30117 in moveit::task_constructor::Introspection::solutionId(moveit::task_constructor::SolutionBase const&) () from /home/rw/zhenjing/git/BagageBusiness/j
#2  0x00007ffff7c6775b in moveit::task_constructor::SolutionBase::fillInfo(moveit_task_constructor_msgs::msg::SolutionInfo_<std::allocator<void> >&, moveit::task_construct
#3  0x00007ffff7c688ac in moveit::task_constructor::SolutionSequence::appendTo(moveit_task_constructor_msgs::msg::Solution_<std::allocator<void> >&, moveit::task_construct
#4  0x00007ffff7c66ce6 in moveit::task_constructor::SolutionBase::toMsg(moveit_task_constructor_msgs::msg::Solution_<std::allocator<void> >&, moveit::task_constructor::Int
#5  0x00007ffff7c72b1c in moveit::task_constructor::Task::execute(moveit::task_constructor::SolutionBase const&) () from /home/rw/zhenjing/git/BagageBusiness/jaka_ws/insta
#6  0x00005555555a65ef in jaka_mtc::PickPlaceTask::execute_multi_trajectory() ()`

> An approach that works is to execute the sub-trajectories up to the point where you need to call the gripper service, and then execute the remaining sub-trajectories afterward. Yes, its a workaround, but it fit my needs

zhenjing, 2025-04-02:
sub-trajectory SolutionSequence Object should setCreator 、setStartState、setEndState. Such as: 

        Stage* stage = task_->stages();
        pick_solution.setCreator(stage);  
        pick_solution.setStartState(*pick_solution.internalStart());
        pick_solution.setEndState(*pick_solution.internalEnd());

        place_solution.setCreator(stage);   
        place_solution.setStartState(*place_solution.internalStart());
        place_solution.setEndState(*place_solution.internalEnd());

        return_solution.setCreator(stage);
        return_solution.setStartState(*return_solution.internalStart());
        return_solution.setEndState(*return_solution.internalEnd());

Tobimane, 2025-04-16:
Hi, 

I am currently working on something that would benefit from this. Is there a tutorial somewhere on how to do it?

Cheers 
T

zhenjing, 2025-04-17:
An example：

`    
int PickPlaceTask::execute_multi_trajectory(std::shared_ptr<ModbusCommand> plc, int pauseTime) {
        LOG_INFO("Executing multi solution trajectory");
        const Stage* stageSolu = task_->solutions().front().get()->creator();  
        Stage* stage = task_->stages();

        moveit_msgs::msg::MoveItErrorCodes execute_result;

        int solution = 0;
        SolutionSequence pick_solution;   
        SolutionSequence place_solution;  
        SolutionSequence return_solution;
        
        auto solutions = dynamic_cast<const SolutionSequence*>(task_->solutions().front().get());
        for(auto sub_solution : solutions->solutions()) {
            auto stage = sub_solution->creator();
            if(0 == solution){
                pick_solution.push_back(*sub_solution);
            } else if(1 == solution){
                place_solution.push_back(*sub_solution);
            } else {
                return_solution.push_back(*sub_solution);
            } 
            
            if(APPROACH_STAGE == stage->name() || LOWER_STAGE == stage->name()){
                solution ++;
            }
        }
        
        // 设置SolutionSequence的关键参数，不设置会crash
        pick_solution.setCreator(stage);   // 沿用Task stages
        pick_solution.setStartState(*pick_solution.internalStart());
        pick_solution.setEndState(*pick_solution.internalEnd());

        place_solution.setCreator(stage);   
        place_solution.setStartState(*place_solution.internalStart());
        place_solution.setEndState(*place_solution.internalEnd());

        return_solution.setCreator(stage);
        return_solution.setStartState(*return_solution.internalStart());
        return_solution.setEndState(*return_solution.internalEnd());

        LOG_INFO("pick_solution: ");
        for(auto sub_solution : pick_solution.solutions()) {
            auto stage = sub_solution->creator();
            LOG_INFO("stage id: " << stage->introspectionId() << " name: " << stage->name() << " cost: " << sub_solution->cost());
        }

        execute_result = task_->execute(pick_solution);
        if (execute_result.val != moveit_msgs::msg::MoveItErrorCodes::SUCCESS) {
            LOG_ERROR("Task execution failed and returned: " << execute_result.val << " message: " 
                      << moveit::core::error_code_to_string(execute_result.val));
            return execute_result.val;
        }

        // TODO
            rclcpp::sleep_for(std::chrono::seconds(pauseTime)); 

        LOG_INFO("place_solution: ");
        for(auto sub_solution : place_solution.solutions()) {
            auto stage = sub_solution->creator();
            LOG_INFO("stage id: " << stage->introspectionId() << " name: " << stage->name() << " cost: " << sub_solution->cost());
        }
        execute_result = task_->execute(place_solution);
        if (execute_result.val != moveit_msgs::msg::MoveItErrorCodes::SUCCESS) {
            LOG_ERROR("Task execution failed and returned: " << execute_result.val << " message: " 
                      << moveit::core::error_code_to_string(execute_result.val));
            return execute_result.val;
        }
        
       // TODO
            rclcpp::sleep_for(std::chrono::seconds(pauseTime)); 

        execute_result = task_->execute(return_solution);
        if (execute_result.val != moveit_msgs::msg::MoveItErrorCodes::SUCCESS) {
            LOG_ERROR("Task execution failed and returned: " << execute_result.val << " message: " 
                      << moveit::core::error_code_to_string(execute_result.val));
            return execute_result.val;
        }

        return execute_result.val;
    }`

Tobimane, 2025-04-29:
Great, thanks! I'll give it a try when i find the time!