moveit/moveit_task_constructor #604: MTC is not performing any movements on real hardware

mitsav01, 2024-08-16:
I configured my real Franka Emika arm for MTC to perform some simple tasks in ROS 2. However, solution is generated in Motion Planning Tasks window in Rviz but it is not moving my robotic arm. Could any one help me with that?

[image]
 and given is the snippet from launch file:
```
     mtc_node = Node(
        package='franka_tc',
        executable='cartesian',
        name='mtc_tutorial',
        output='both',
        parameters=[moveit_config.robot_description,
                    moveit_config.robot_description_semantic,
                    moveit_config.joint_limits,
                    moveit_config.planning_pipelines,
                    moveit_config.planning_scene_monitor,
                    moveit_config.robot_description_kinematics],

    )
    
```
What changes I should do to move my arm?

rhaschke, 2024-08-16:
It would be helpful to see the console log of both the move_group node and the MTC node. Probably, there you can find a hint to the underlying problem. I guess, that you are missing the task execution capability, which needs to be configured in the move_group node.

mitsav01, 2024-08-19:
[image]
[image]

These are console logs for move_group node and cartesian. 
I have already added task execution capabilities in move_group node,  but it is not working on real robot. Also, it is not taking current position of robot into account

rhaschke, 2024-08-19:
This log is incomplete. Please don't send images, but paste text console output.

> Also, it is not taking current position of robot into account

What exactly do you mean by this?

mitsav01, 2024-08-19:
[image]
Since initialization, the actual position of the robot is different then one given in the planning.

Here is console output:

> ros2 launch franka_tc demo.launch.py robot_ip:=192.168.101.116 load_gripper:=true 
[INFO] [launch]: All log files can be found below /home/mitesh/.ros/log/2024-08-19-11-21-15-970291-mitesh-IAI-42926
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [42929]
[INFO] [cartesian-2]: process started with pid [42930]
[INFO] [robot_state_publisher-3]: process started with pid [42931]
[INFO] [move_group-4]: process started with pid [42932]
[INFO] [ros2_control_node-5]: process started with pid [42933]
[INFO] [joint_state_publisher-6]: process started with pid [42934]
[INFO] [franka_gripper_node-7]: process started with pid [42935]
[INFO] [ros2 run controller_manager spawner panda_arm_controller-8]: process started with pid [42936]
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-9]: process started with pid [42937]
[robot_state_publisher-3] [INFO] [1724059276.379871172] [robot_state_publisher]: Robot initialized
[franka_gripper_node-7] [INFO] [1724059276.437889866] [panda_gripper]: Trying to establish a connection with the gripper
[franka_gripper_node-7] [INFO] [1724059276.444175165] [panda_gripper]: Connected to gripper
[move_group-4] [INFO] [1724059276.477479574] [move_group.moveit.moveit.ros.rdf_loader]: Loaded robot model in 0.00745191 seconds
[move_group-4] [INFO] [1724059276.477555569] [move_group.moveit.moveit.core.robot_model]: Loading robot model 'panda'...
[move_group-4] [WARN] [1724059276.477663824] [move_group.moveit.moveit.core.robot_model]: Link panda_leftfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[move_group-4] [WARN] [1724059276.477684633] [move_group.moveit.moveit.core.robot_model]: Link panda_rightfinger has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[move_group-4] [INFO] [1724059276.495982026] [move_group.moveit.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'panda_arm': 1 1 1 1 1 1 1
[ros2_control_node-5] [INFO] [1724059276.498435837] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-5] [INFO] [1724059276.498505734] [controller_manager]: update rate is 200 Hz
[ros2_control_node-5] [INFO] [1724059276.502331480] [controller_manager]: Received robot description from topic.
[ros2_control_node-5] [INFO] [1724059276.510686228] [controller_manager.resource_manager]: Loading hardware 'FrankaHardwareInterface' 
[ros2_control_node-5] [INFO] [1724059276.530796734] [controller_manager.resource_manager]: Loaded hardware 'FrankaHardwareInterface' from plugin 'franka_hardware/FrankaHardwareInterface'
[ros2_control_node-5] [INFO] [1724059276.530846141] [controller_manager.resource_manager]: Initialize hardware 'FrankaHardwareInterface' 
[ros2_control_node-5] [INFO] [1724059276.530875746] [FrankaHardwareInterface]: Connecting to robot at "192.168.101.116" ...
[ros2_control_node-5] [INFO] [1724059276.553090993] [FrankaHardwareInterface]: Successfully connected to robot
[move_group-4] [INFO] [1724059276.555724603] [move_group.moveit.moveit.ros.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-4] [INFO] [1724059276.555875984] [move_group.moveit.moveit.ros.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-4] [INFO] [1724059276.559186211] [move_group.moveit.moveit.ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-4] [INFO] [1724059276.560836120] [move_group.moveit.moveit.ros.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-4] [INFO] [1724059276.560862203] [move_group.moveit.moveit.ros.planning_scene_monitor]: Stopping existing planning scene publisher.
[move_group-4] [INFO] [1724059276.561701620] [move_group.moveit.moveit.ros.planning_scene_monitor]: Stopped publishing maintained planning scene.
[move_group-4] [INFO] [1724059276.569656907] [move_group.moveit.moveit.ros.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-4] [INFO] [1724059276.569974635] [move_group.moveit.moveit.ros.planning_scene_monitor]: Starting planning scene monitor
[move_group-4] [INFO] [1724059276.575738289] [move_group.moveit.moveit.ros.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-4] [INFO] [1724059276.575765150] [move_group.moveit.moveit.ros.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-4] [INFO] [1724059276.579414704] [move_group.moveit.moveit.ros.planning_scene_monitor]: Listening to 'collision_object'
[move_group-4] [INFO] [1724059276.581646104] [move_group.moveit.moveit.ros.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[ros2_control_node-5] [INFO] [1724059276.584527605] [error_recovery_service_server]: Error recovery service started
[move_group-4] [WARN] [1724059276.587624168] [move_group.moveit.moveit.ros.occupancy_map_monitor]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[... cut ...]
[ros2 run controller_manager spawner panda_arm_controller-8] [INFO] [1724059277.673411355] [spawner_panda_arm_controller]: Configured and activated panda_arm_controller
[rviz2-1] [INFO] [1724059277.682111675] [rviz2.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'panda_arm': 1 1 1 1 1 1 1
[rviz2-1] [INFO] [1724059277.726946137] [rviz2.moveit.ros.planning_scene_monitor]: Starting planning scene monitor
[rviz2-1] [INFO] [1724059277.729691387] [rviz2.moveit.ros.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[INFO] [ros2 run controller_manager spawner panda_arm_controller-8]: process has finished cleanly [pid 42936]

rhaschke, 2024-08-19:
The log still looks incomplete. There is no logging about MTC planning or execution.

rhaschke, 2024-08-22:
The new log is unrelated to the previous one. But the previous one is incomplete!