moveit/moveit_task_constructor #605: not able to stop the retractig movement in the pick and place pipeline

ron007d, 2024-08-23:
I am currently using the pick and place pipeline of the `moveit task constructor`.

I want to change certain movement in the pipeline.

After moving to the place pose
1. Don't open the gripper.
2. Don't retract.

but if I disable the set retract motion its fetching error.

Tried other method like calculating the robot_tool0 pose based on the object's target pose with the frame_id = "world"

so that I can move the gripper to the pose where the object will be placed.  but not able to get the tf values of the robot_tool0 with the desired object location. I guess I am missing some calculation.

rhaschke, 2024-08-23:
> I want to change certain movement in the pipeline. After moving to the place pose
> 
> 1. Don't open the gripper.
> 2. Don't retract.

Simply remove the corresponding stages.

> Tried other method like calculating the robot_tool0 pose based on the object's target pose with the frame_id = "world"
> so that I can move the gripper to the pose where the object will be placed. But [...]

I don't understand what you want to do and what you tried. Please clarify.

ron007d, 2024-08-24:
sorry for giving a very small info.

Currently I am using this task constructor to pick and place a object
https://moveit.github.io/moveit_task_constructor/tutorials/pick-and-place.html

in this I had to pass two objects 
1. object name
2. place pose
and its working from there

the movement is like this
1. It goes over the object
2. approach the object
3. attach the object to the end effector with grippers touch links
4. grasp the object
5. picks the object
6. then its goes over the place position
7. place the object
8. ungrasp the object
9. detach the object from the end effector
10. retract from the position

in this I wanted to stop this last 3 operation
1. ungrasp the object
2. detach
3. and retract

but this all three are combined into the `pymoveit_mtc.stages.Place`.

after `generating place pose`  how can I use those pose to move to that position. 

so I was not able to fix that so I though out this mathematics

```
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "robot1_tool0"
pose: 
  position: 
    x: 4.117445095779876e-09
    y: 8.452217313319532e-09
    z: 0.15000000031112115
  orientation: 
    x: -1.933663352116923e-08
    y: -6.661140770257759e-09
    z: -0.707108080393473
    w: 0.7071054819772353
```

> ### this is the posestamp of the attach object with the frame_id `robot1_tool0` :point_up_2:

```
header: 
  seq: 0
  stamp: 
    secs: 1724403677
    nsecs: 878194093
  frame_id: "base_link"
pose: 
  position: 
    x: 0.2499999950542699
    y: 2.2709854219371554e-10
    z: 0.45000000017410197
  orientation: 
    x: -0.7071054810709814
    y: 0.7071080812997231
    z: -7.608165474450681e-09
    w: 2.082272085122174e-08
```
> ### this is the posestamp of the robot1_tool0 with the frame_id `base_link` :point_up_2:

```
frame_id: "base_link"
pose:
position: 
  x: 0.24999999263314437
  y: -1.0870961284455411e-09
  z: 0.2999999998629808
orientation: 
  x: 1.0000000000000002
  y: -1.2816341960898667e-09
  z: 1.7204706991680041e-09
  w: -3.811717107357996e-10
```  
> ### this is the object pose value with the frame_id `base_link` :point_up_2:

**if I want the `target pose of the object` to be**
```
frame_id : "base_link"
position: 
  x: 0.24999999263314437
  y: 1.3
  z: 0.0045
orientation: 
  x: -0.707
  y: 0.707
  z: 0.0
  w: 0.0
```
**with the frame_id `base_link`**

> ### then what will be the robot_tool0 pose with the frame_id `base_link`, which can be used to `set target pose`. :point_up_2:

rhaschke, 2024-08-24:
If you don't want to ungrasp the object, i.e. skip the last 3 stages, you need to replace `pymoveit_mtc.stages.Place` with individual stages to perform the place operation: a `PoseGenerator` wrapped by `ComputeIK` to generate the place pose and (optionally) a `MoveRelative` stage to place the object using a Cartesian motion.

There is no need to manually fiddle with transforms.

ron007d, 2024-08-24:
> If you don't want to ungrasp the object, i.e. skip the last 3 stages, you need to replace `pymoveit_mtc.stages.Place` with individual stages to perform the place operation: a `PoseGenerator` wrapped [...]

Cool its worked.
first I had added only compute IK then added the move relative before that I have planned to place the object directly at the position then in place generator added extra z value
then added a move relative Cartesian stage to move the object to the place pose

```python
def pick_place_object(object_name,
                      place_pose_x : Union[int,float],
                      place_pose_y : Union[int,float],
                      place_pose_z : Union[int,float],
                      place_orientation_roll : Union[int,float],
                      place_orientation_pitch : Union[int,float],
                      place_orientation_yaw : Union[int,float],
                      robot_arm_group_name : str = 'robot1',
                      robot_arm_gripper_group_name : str = 'robot1_gripper',
                      robot_arm_end_effector_link_name : str = 'robot1_tool0',
                      frame_id : str = 'base_link') -> Tuple[bool,core.Task]:
    arm = robot_arm_group_name
    end_effector = robot_arm_gripper_group_name
    eef_link = robot_arm_end_effector_link_name
    task = core.Task('pick_place_task')
    task.enableIntrospection()

    task.add(stages.CurrentState('current'))

    pipeline = core.PipelinePlanner()
    pipeline.planner = "RRTConnectkConfigDefault"
    pipeline.max_velocity_scaling_factor = 1.0
    pipeline.max_acceleration_scaling_factor = 1.0
    planners = [(arm, pipeline)]

    task.add(stages.Connect("connect1", planners))

    grasp_generator = stages.GenerateGraspPose("Generate Grasp Pose")
    grasp_generator.eef = end_effector
    grasp_generator.angle_delta = 1.5708
    grasp_generator.pregrasp = "open"
    grasp_generator.grasp = "close"
    grasp_generator.setMonitoredStage(task["current"]) 
    grasp_generator.object = object_name

    simpleGrasp = stages.SimpleGrasp(grasp_generator, "Grasp")

    ik_frame = PoseStamped()
    ik_frame.header.frame_id = eef_link
    ik_frame.pose.position.z = 0.15
    # ik_frame.pose.orientation.
    simpleGrasp.setIKFrame(ik_frame)

    pick = stages.Pick(simpleGrasp, "Pick")
    pick.eef = end_effector
    pick.object = object_name
    pick.eef_group = end_effector
    pick.eef_parent_group = arm
    print(pick.properties)
    
    # Twist to approach the object
    approach = TwistStamped()
    approach.header.frame_id = frame_id
    # approach.twist.angular.x = 1.5708
    approach.twist.linear.z = -2.0
    pick.setApproachMotion(approach, 0.1, 0.2)

    lift = TwistStamped()
    lift.header.frame_id = eef_link
    # lift.pose.position.z = 0.15
    lift.twist.linear.z = -1.0
    pick.setLiftMotion(lift, 0.1, 0.2)

    task.add(pick)

    task.add(stages.Connect("connect2", planners))

    placepose = PoseStamped()
    placepose.header.frame_id = frame_id
    quaternion_array = quaternion_from_euler(ai = math.radians(place_orientation_roll),
                                             aj = math.radians(place_orientation_pitch),
                                             ak = math.radians(place_orientation_yaw))
    placepose.pose.position.x = place_pose_x
    placepose.pose.position.y = place_pose_y
    placepose.pose.position.z = place_pose_z + 0.2
    placepose.pose.orientation.x = quaternion_array[0]
    placepose.pose.orientation.y = quaternion_array[1]
    placepose.pose.orientation.z = quaternion_array[2]
    placepose.pose.orientation.w = quaternion_array[3]

    place_generator = stages.GeneratePlacePose("Generate Place Pose")
    place_generator.setMonitoredStage(task["Pick"])
    place_generator.object = object_name
    place_generator.pose = placepose

    ## test code

    compute_ik = stages.ComputeIK('computeIK',place_generator)
    compute_ik.group = robot_arm_group_name
    compute_ik.ik_frame = PoseStamped(header=Header(frame_id=object_name))
    compute_ik.max_ik_solutions = 10
    # [propertyTut14]
    props = compute_ik.properties
    # derive target_pose from child's solution
    props.configureInitFrom(core.Stage.PropertyInitializerSource.INTERFACE, ["target_pose"])
    # [propertyTut14]

    task.add(compute_ik)

    cartesian = core.CartesianPath()
    move = stages.MoveRelative("z + 0.2",cartesian)
    move.group = robot_arm_group_name
    move.setDirection(Vector3Stamped(
        header= Header(
            frame_id=frame_id),
        vector=Vector3(0,0,-0.2)))
    
    task.add(move)
```

closed 2024-08-24 by ron007d