moveit/moveit_task_constructor #606: pick_place_demo: parameter 'table_name' is not initialized

KarthiAru, 2024-08-26:
Hi,

I'm getting the below error while executing the pick_place_demo example. I'm on the humble branch.

```
~/ws_moveit  ros2 launch moveit_task_constructor_demo run.launch.py exe:=pick_place_demo
[INFO] [launch]: All log files can be found below /home/ka/.ros/log/2024-08-26-18-09-08-824360-karthiaru-dl-rig-245644
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [pick_place_demo-1]: process started with pid [245645]
[pick_place_demo-1] terminate called after throwing an instance of 'rclcpp::exceptions::ParameterUninitializedException'
[pick_place_demo-1]   what():  parameter 'table_name' is not initialized
[ERROR] [pick_place_demo-1]: process has died [pid 245645, exit code -6, cmd '/home/ka/ws_moveit/install/moveit_task_constructor_demo/lib/moveit_task_constructor_demo/pick_place_demo --ros-args --params-file /tmp/launch_params_kwlzogib --params-file /tmp/launch_params_8f6n4gst --params-file /tmp/launch_params_o55rp0oh --params-file /tmp/launch_params_w6biha_v --params-file /tmp/launch_params_co89ejq_'].
```

KarthiAru, 2024-08-26:
#607 Fixes this issue

rhaschke, 2024-08-27:
Fixed via #607

closed 2024-08-27 by rhaschke