moveit/moveit_task_constructor #609: MTC node crashes with exit code -11 when trying to modify the pick and place tutorial with interbotix arm

e-tse, 2024-08-29:
I'm trying to take the MTC pick and place tutorial found [here](https://moveit.picknik.ai/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html) and implement it on the [WidowX 250s arm by Trossen Robotics](https://docs.trossenrobotics.com/interbotix_xsarms_docs/). 

Trossen supplies a launch file titled "xsarm_moveit.launch.py" which launches a move group node, a robot state publisher, a joint state publisher, and an rviz node.  I modified the supplied launch file slightly so that it contains `move_group_capabilities = {
        "capabilities": "move_group/ExecuteTaskSolutionCapability"
    }
` within the move group parameters.

I then created a new workspace that contains the moveit_task_constructor and a custom package called pick_and_place.  [See this link to access my pick_and_place code.](https://github.com/e-tse/pick_and_place.git).  This is primarily just the pick and place tutorial code with a modified launch file, and some changed goal names to reflect the different SRDF file (for example, a gripper pose of "Released" instead of "open").  

A few notes: The moveit_task_constructor is pointed to the humble branch.  Interbotix automatically installs moveit so I am using a binary install as opposed to installing from source.

Here are the steps I take to get to error:

1. I launch the supplied interbotix launch file using the command ```ros2 launch interbotix_xsarm_moveit xsarm_moveit.launch.py robot_model:=wx250s hardware_type:=fake```
2. Using the motion planning rviz GUI, I set the gripper pose to the gripping state, since for some reason it starts in an invalid state
3. In a separate window, I launch ```ros2 launch pick_and_place pick_and_place.launch.py```.  The interbotix workspace is sourced in my .bashrc.

After running the second launch file, I get the following console output:
[image]

One thing to note is, the node works if I only have the open hand stage implemented in code, i.e. if I comment out lines 142-221, I can get the robot to successfully open the gripper.  When I do that, the terminal output is as follows:

[image]

Please note that the error messages "Pre planning" "passed a" and "passed b" are simply logger messages I created to try and pinpoint the source of the error.  Using these, I seem to have tracked the issue down to be line 82, which is the `if (!task_.plan(1))` line.  I've set it to 1 instead of 5 (which is the number in the tutorial) because I saw a post somewhere saying that may fix it, but it still leads to the same error.

Please let me know if there's anything you can do to help.  Apologies if this is more of a question for Trossen than MTC.  Thank you in advance.

e-tse, 2024-08-29:
First output message as a text if easier for you:
[INFO] [launch]: All log files can be found below /home/etse8/.ros/log/2024-08-29-10-53-48-809070-ME02L0465GRD02L.me.gatech.edu-57480
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING:root:Cannot infer URDF from `/home/etse8/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit`. -- using config/wx250s.urdf
WARNING:root:Cannot infer SRDF from `/home/etse8/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit`. -- using config/wx250s.srdf
WARNING:root:"File /home/etse8/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/wx250s.urdf doesn't exist"
WARNING:root:The robot description will be loaded from /robot_description topic 
WARNING:root:trajectory_execution: `Parameter file_path is undefined and no matches for /home/etse8/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/*_controllers.yaml
[INFO] [pick_and_place-1]: process started with pid [57481]
[pick_and_place-1] [INFO] [1724943230.165513741] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0258015 seconds
[pick_and_place-1] [INFO] [1724943230.165557359] [moveit_robot_model.robot_model]: Loading robot model 'wx250s'...
[pick_and_place-1] [INFO] [1724943230.165564544] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[pick_and_place-1] [WARN] [1724943230.191596890] [pick_and_place_node]: Failed to find 'ompl.planning_plugin'. Attempting to load pipeline from old parameter structure. Please update your MoveIt config.
[pick_and_place-1] [INFO] [1724943230.205354899] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[pick_and_place-1] [INFO] [1724943230.208513546] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[pick_and_place-1] [INFO] [1724943230.208527883] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[pick_and_place-1] [INFO] [1724943230.208532373] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[pick_and_place-1] [INFO] [1724943230.208544086] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[pick_and_place-1] [INFO] [1724943230.208556150] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[pick_and_place-1] [INFO] [1724943230.208560928] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[pick_and_place-1] [INFO] [1724943230.208570621] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[pick_and_place-1] [INFO] [1724943230.208575314] [moveit_ros.fix_start_state_collision]: Param 'move_group.jiggle_fraction' was not set. Using default value: 0.020000
[pick_and_place-1] [INFO] [1724943230.208579488] [moveit_ros.fix_start_state_collision]: Param 'move_group.max_sampling_attempts' was not set. Using default value: 100
[pick_and_place-1] [INFO] [1724943230.208588875] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[pick_and_place-1] [INFO] [1724943230.208592950] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[pick_and_place-1] [INFO] [1724943230.208596162] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[pick_and_place-1] [INFO] [1724943230.208599140] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[pick_and_place-1] [INFO] [1724943230.208602139] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[pick_and_place-1] [ERROR] [1724943230.210533256] [pick_and_place]: Pre planning
[pick_and_place-1] [INFO] [1724943230.255891861] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'interbotix_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ERROR] [pick_and_place-1]: process has died [pid 57481, exit code -11, cmd '/home/etse8/ros2_code/arm_ws/install/pick_and_place/lib/pick_and_place/pick_and_place --ros-args --params-file /tmp/launch_params_x795jl45 --params-file /home/etse8/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit/config/kinematics.yaml --params-file /tmp/launch_params_x0poi4z5 -r /joint_states:=/wx250s/joint_states -r /robot_description:=/wx250s/robot_description'].

Second output as text:
[INFO] [launch]: All log files can be found below /home/etse8/.ros/log/2024-08-29-12-11-19-885908-ME02L0465GRD02L.me.gatech.edu-60087
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING:root:Cannot infer URDF from `/home/etse8/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit`. -- using config/wx250s.urdf
WARNING:root:Cannot infer SRDF from `/home/etse8/interbotix_ws/install/interbotix_xsarm_moveit/share/interbotix_xsarm_moveit`. -- using config/wx250s.srdf
[... cut ...]
[pick_and_place-1] [ERROR] [1724947881.319294688] [pick_and_place]: Pre planning
[pick_and_place-1] [INFO] [1724947881.368176255] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'interbotix_gripper' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_and_place-1] [ERROR] [1724947881.411945496] [pick_and_place]: passed a
[pick_and_place-1] [ERROR] [1724947881.412237920] [pick_and_place]: passed b

rhaschke, 2024-08-29:
I was able to reproduce the issue - after collecting all the required interbotix packages.
The culprit is the name of the link: `wx250s/ee_gripper_link`. As it contains a slash, the MoveIt code couldn't resolve the corresponding link. This is fixed in https://github.com/moveit/moveit2/pull/2985. Until this is merged and released, you need to build MoveIt from source to get that fix.

closed 2024-08-29 by rhaschke