moveit/moveit_task_constructor #612: Unable to use MTC with UR5

minku1219, 2024-09-10:
Hi, I am working upon MTC for using it with UR5.
I tried using cartesian executable wihth UR5 but it did'nt worked and showed me an error.

Error:-

[cartesian-1] Failing stage(s):
[cartesian-1] x +0.2 (0/1): missing ik_frame

Full log message :-

[INFO] [launch]: All log files can be found below /home/tcs/.ros/log/2024-09-10-12-00-02-734367-tcs-Precision-7710-68104
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING:root:Cannot infer URDF from `/opt/ros/humble/share/ur_moveit_config`. -- using config/ur.urdf
WARNING:root:Cannot infer SRDF from `/opt/ros/humble/share/ur_moveit_config`. -- using config/ur.srdf
[INFO] [cartesian-1]: process started with pid [68105]
[cartesian-1] [INFO] [1725949803.501420331] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0461322 seconds
[cartesian-1] [INFO] [1725949803.501527067] [moveit_robot_model.robot_model]: Loading robot model 'ur5'...
[cartesian-1] [INFO] [1725949803.501541152] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[cartesian-1] Plan!
[cartesian-1] PN6moveit16task_constructor11TaskPrivateE
[cartesian-1] Init!
[cartesian-1]   0  - ←   0 →   -  0 / Cartesian Path
[cartesian-1]     1  - ←   1 →   -  0 / initial state
[cartesian-1]     -  0 →   0 →   -  0 / x +0.2
[cartesian-1] Failing stage(s):
[cartesian-1] x +0.2 (0/1): missing ik_frame
[cartesian-1] -1
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[cartesian-1] [INFO] [1725949815.137223296] [rclcpp]: signal_handler(signum=2)
[INFO] [cartesian-1]: process has finished cleanly [pid 68105]

[image]

rhaschke, 2024-09-10:
Looks like your robot's move group doesn't have a unique end-effector frame. Thus, you need to specify the frame explicitly by setting the `ik_frame` property of the Cartesian `x +0.2` stage.

minku1219, 2024-09-10:
Could you please elaborate where and how to add this setIkFrame?
Thanks in advance!

rhaschke, 2024-09-10:
Just augment the stage definition:
```cpp
auto stage = std::make_unique<stages::MoveRelative>("x +0.2", cartesian_interpolation);
stage->setGroup(group);
```
with a line defining the ik frame:
```cpp
stage->setIKFrame("your robot's eef frame");
```

minku1219, 2024-09-10:
Thanks @rhaschke for the solution.
The error is gone but the robot is unable to move.
Could you please provide with some snippet to make it work with UR5

rhaschke, 2024-09-10:
How can I provide help if you don't specify the exact error you are observing?

minku1219, 2024-09-10:
Actually it is not showing any error, instead the robot i am working with is not moving.

minku1219, 2024-09-10:
I have'nt found any example for working with UR5 or any UR manipulator, so i am trying this on my own

rhaschke, 2024-09-10:
> Actually it is not showing any error, instead the robot i am working with is not moving.

The original code of `cartesian.cpp` is not attempting to execute the solution:
https://github.com/moveit/moveit_task_constructor/blob/5a4480814628db4207d9734d63dd907619ced93b/demo/src/cartesian.cpp#L142-L143
You need to execute it for the robot to move:
`task.execute(*task.solutions().front()); `

minku1219, 2024-09-10:
Added this and encountered an error!

[INFO] [launch]: All log files can be found below /home/tcs/.ros/log/2024-09-10-14-35-32-732590-tcs-Precision-7710-150197
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING:root:Cannot infer URDF from `/opt/ros/humble/share/ur_moveit_config`. -- using config/ur.urdf
WARNING:root:Cannot infer SRDF from `/opt/ros/humble/share/ur_moveit_config`. -- using config/ur.srdf
[INFO] [cartesian-1]: process started with pid [150198]
[cartesian-1] [INFO] [1725959133.689321885] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0616049 seconds
[cartesian-1] [INFO] [1725959133.689485639] [moveit_robot_model.robot_model]: Loading robot model 'ur5'...
[cartesian-1] [INFO] [1725959133.689501686] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[cartesian-1] Plan!
[cartesian-1] PN6moveit16task_constructor11TaskPrivateE
[cartesian-1] Init!
[cartesian-1] 1
[cartesian-1] [ERROR] [1725959134.262591040] [moveit_task_constructor_executor_140722771479248]: Failed to connect to the 'execute_task_solution' action server

minku1219, 2024-09-10:
# Copyright (c) 2021 PickNik, Inc.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
#
#    * Redistributions of source code must retain the above copyright
#      notice, this list of conditions and the following disclaimer.
#
#    * Redistributions in binary form must reproduce the above copyright
#      notice, this list of conditions and the following disclaimer in the
#      documentation and/or other materials provided with the distribution.
#
#    * Neither the name of the {copyright_holder} nor the names of its
#      contributors may be used to endorse or promote products derived from
#      this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
# AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
# IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
# ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
# LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
# CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
# SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
# INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
# CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
# ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
# POSSIBILITY OF SUCH DAMAGE.

#
# Author: Denis Stogl

import os

from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare
from ur_moveit_config.launch_common import load_yaml

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.conditions import IfCondition
from launch.substitutions import Command, FindExecutable, LaunchConfiguration, PathJoinSubstitution
from ament_index_python.packages import get_package_share_directory

def launch_setup(context, *args, **kwargs):

    # Initialize Arguments
    ur_type = LaunchConfiguration("ur_type")
    use_fake_hardware = LaunchConfiguration("use_fake_hardware")
    safety_limits = LaunchConfiguration("safety_limits")
    safety_pos_margin = LaunchConfiguration("safety_pos_margin")
    safety_k_position = LaunchConfiguration("safety_k_position")
    # General arguments
    description_package = LaunchConfiguration("description_package")
    description_file = LaunchConfiguration("description_file")
    _publish_robot_description_semantic = LaunchConfiguration("publish_robot_description_semantic")
    moveit_config_package = LaunchConfiguration("moveit_config_package")
    moveit_joint_limits_file = LaunchConfiguration("moveit_joint_limits_file")
    moveit_config_file = LaunchConfiguration("moveit_config_file")
    warehouse_sqlite_path = LaunchConfiguration("warehouse_sqlite_path")
    prefix = LaunchConfiguration("prefix")
    use_sim_time = LaunchConfiguration("use_sim_time")
    launch_rviz = LaunchConfiguration("launch_rviz")
    launch_servo = LaunchConfiguration("launch_servo")
    exe = LaunchConfiguration("exe")

    joint_limit_params = PathJoinSubstitution(
        [FindPackageShare(description_package), "config", ur_type, "joint_limits.yaml"]
    )
    kinematics_params = PathJoinSubstitution(
        [FindPackageShare(description_package), "config", ur_type, "default_kinematics.yaml"]
[... 302 lines cut ...]
            "use_sim_time",
            default_value="false",
            description="Make MoveIt to use simulation time. This is needed for the trajectory planing in simulation.",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument(
            "prefix",
            default_value='""',
            description="Prefix of the joint names, useful for "
            "multi-robot setup. If changed than also joint names in the controllers' configuration "
            "have to be updated.",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument("launch_rviz", default_value="true", description="Launch RViz?")
    )
    declared_arguments.append(
        DeclareLaunchArgument("launch_servo", default_value="true", description="Launch Servo?")
    )
    declared_arguments.append(
        DeclareLaunchArgument("exe", default_value="cartesian", description="Launch executable?")
    )

    return LaunchDescription(declared_arguments + [OpaqueFunction(function=launch_setup)])

minku1219, 2024-09-10:
Cartesian.cpp code:-

#include <moveit/task_constructor/task.h>

#include <moveit/task_constructor/stages/fixed_state.h>
#include <moveit/task_constructor/solvers/cartesian_path.h>
#include <moveit/task_constructor/solvers/joint_interpolation.h>
#include <moveit/task_constructor/stages/move_to.h>
#include <moveit/task_constructor/stages/move_relative.h>
#include <moveit/task_constructor/stages/connect.h>

#include <rclcpp/rclcpp.hpp>
#include <moveit/planning_scene/planning_scene.h>

using namespace moveit::task_constructor;

Task createTask(const rclcpp::Node::SharedPtr& node) {
	Task t;
	t.stages()->setName("Cartesian Path");

	const std::string group = "ur_manipulator";
	const std::string eef = "tool0";

	// create Cartesian interpolation "planner" to be used in various stages
	auto cartesian_interpolation = std::make_shared<solvers::CartesianPath>();
	// create a joint-space interpolation "planner" to be used in various stages
	auto joint_interpolation = std::make_shared<solvers::JointInterpolationPlanner>();

	// start from a fixed robot state
	t.loadRobotModel(node);
	auto scene = std::make_shared<planning_scene::PlanningScene>(t.getRobotModel());
	{
		auto& state = scene->getCurrentStateNonConst();
		state.setToDefaultValues(state.getJointModelGroup(group), "ready");

		auto fixed = std::make_unique<stages::FixedState>("initial state");
		fixed->setState(scene);
		t.add(std::move(fixed));
	}

	{
		auto stage = std::make_unique<stages::MoveRelative>("x +0.01", cartesian_interpolation);
		stage->setGroup(group);
		stage->setIKFrame("tool0");
		geometry_msgs::msg::Vector3Stamped direction;
		direction.header.frame_id = "world";
		direction.vector.x = 0.01;
		stage->setDirection(direction);
		t.add(std::move(stage));
	}

	// {
	// 	auto stage = std::make_unique<stages::MoveRelative>("y -0.01", cartesian_interpolation);
	// 	stage->setGroup(group);
	// 	stage->setIKFrame("ur5_tool0");

	// 	geometry_msgs::msg::Vector3Stamped direction;
	// 	direction.header.frame_id = "world";
	// 	direction.vector.y = -0.01;
	// 	stage->setDirection(direction);
	// 	t.add(std::move(stage));
	// }

	// {  // rotate about TCP
	// 	auto stage = std::make_unique<stages::MoveRelative>("rz +45°", cartesian_interpolation);
	// 	stage->setGroup(group);
	// 	stage->setIKFrame("ur5_tool0");

	// 	geometry_msgs::msg::TwistStamped twist;
	// 	twist.header.frame_id = "world";
	// 	twist.twist.angular.z = M_PI / 4.;
	// 	stage->setDirection(twist);
	// 	t.add(std::move(stage));
	// }

	// {  // perform a Cartesian motion, defined as a relative offset in joint space
	// 	auto stage = std::make_unique<stages::MoveRelative>("joint offset", cartesian_interpolation);
	// 	stage->setGroup(group);
	// 	stage->setIKFrame("ur5_tool0");

	// 	std::map<std::string, double> offsets = { { "ur5_wrist_3_joint", M_PI / 6. }, { "ur5_wrist_1_joint", -M_PI / 6 } };
	// 	stage->setDirection(offsets);
	// 	t.add(std::move(stage));
	// }

	// {  // move from reached state back to the original state, using joint interpolation
	// 	stages::Connect::GroupPlannerVector planners = { { group, joint_interpolation } };
	// 	auto connect = std::make_unique<stages::Connect>("connect", planners);
	// 	t.add(std::move(connect));
	// }

	// {  // final state is original state again
	// 	auto fixed = std::make_unique<stages::FixedState>("final state");
	// 	fixed->setState(scene);
	// 	t.add(std::move(fixed));
	// }

	return t;
}

int main(int argc, char** argv) {
	rclcpp::init(argc, argv);
	auto node = rclcpp::Node::make_shared("mtc_tutorial");
	std::thread spinning_thread([node] { rclcpp::spin(node); });

	auto task = createTask(node);
	try {
		if (task.plan()){
			task.introspection().publishSolution(*task.solutions().front());
			task.execute(*task.solutions().front());
		}
	} catch (const InitStageException& ex) {
		std::cerr << "planning failed with exception" << std::endl << ex << task;
	}

	// keep alive for interactive inspection in rviz
	spinning_thread.join();
	return 0;
}

rhaschke, 2024-09-10:
To allow execution, you also need to enable the task execution capability in your `move_group` launch file:

https://github.com/ros-planning/moveit_task_constructor/blob/cb867aeddfdca6e3791a4ede998fe23ce9958754/demo/launch/demo.launch.py#L19-L21

https://github.com/ros-planning/moveit_task_constructor/blob/cb867aeddfdca6e3791a4ede998fe23ce9958754/demo/launch/demo.launch.py#L27-L30

minku1219, 2024-09-10:
Thanks @rhaschke .
I'll just give it a try.

minku1219, 2024-09-10:
On adding this to the capabilities:-

moveit_config.move_group_capabilities = {
        "capabilities": """pilz_industrial_motion_planner/MoveGroupSequenceAction \
            pilz_industrial_motion_planner/MoveGroupSequenceService \
                move_group/ExecuteTaskSolutionCapability"""
    }
    
move_group_node = Node(
        package="moveit_ros_move_group",
        # namespace="ur5",
        executable="move_group",
        output="screen",
        parameters=[
            moveit_config.robot_description,
            moveit_config.robot_description_semantic,
            moveit_config.robot_description_kinematics,
            moveit_config.joint_limits,
            planning_pipelines_param,
            moveit_config.pilz_cartesian_limits,
            moveit_config.move_group_capabilities,
            moveit_config.trajectory_execution,
            moveit_config.planning_scene_monitor,
            {"use_sim_time": use_sim_time},
            warehouse_ros_config,
        ],
        remappings=remappings,
    )
    
Same error encountered:-

[INFO] [cartesian-1]: process started with pid [163881]
[cartesian-1] [INFO] [1725961230.361890400] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0540685 seconds
[cartesian-1] [INFO] [1725961230.361966498] [moveit_robot_model.robot_model]: Loading robot model 'ur5'...
[cartesian-1] [INFO] [1725961230.361975480] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[cartesian-1] Plan!
[cartesian-1] PN6moveit16task_constructor11TaskPrivateE
[cartesian-1] Init!
[cartesian-1] 1
[cartesian-1] [ERROR] [1725961230.932017126] [moveit_task_constructor_executor_140722062763376]: Failed to connect to the 'execute_task_solution' action server

minku1219, 2024-09-10:
[image]

Hope this helps!

rhaschke, 2024-09-10:
Did you restarted your `move_group` node? 
It should state `Loading 'move_group/ExecuteTaskSolutionCapability'...` in its startup log.
Please correctly [quote code using triple backticks](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code)!

minku1219, 2024-09-10:
Yes! i started the move_group node.

Complete log is:-

```
[INFO] [launch]: All log files can be found below /home/tcs/.ros/log/2024-09-10-15-38-09-813656-tcs-Precision-7710-171263
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [171276]
[INFO] [controller_stopper_node-3]: process started with pid [171278]
[INFO] [ur_ros2_control_node-1]: process started with pid [171274]
[INFO] [urscript_interface-4]: process started with pid [171280]
[INFO] [robot_state_publisher-5]: process started with pid [171282]
[INFO] [spawner-6]: process started with pid [171284]
[INFO] [spawner-7]: process started with pid [171286]
[INFO] [spawner-8]: process started with pid [171288]
[INFO] [spawner-9]: process started with pid [171291]
[INFO] [spawner-10]: process started with pid [171293]
[INFO] [spawner-11]: process started with pid [171296]
[controller_stopper_node-3] [INFO] [1725962890.713165343] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[dashboard_client-2] [INFO] [1725962890.717344348] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[ur_ros2_control_node-1] [WARN] [1725962890.814710877] [ur5.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-1] [INFO] [1725962890.815450780] [resource_manager]: Loading hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1725962890.818591969] [resource_manager]: Initialize hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1725962890.818695611] [resource_manager]: Successful initialization of hardware 'ur5'
[ur_ros2_control_node-1] [INFO] [1725962890.818991597] [resource_manager]: 'configure' hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1725962890.819006971] [resource_manager]: Successful 'configure' of hardware 'ur5'
[ur_ros2_control_node-1] [INFO] [1725962890.819055130] [resource_manager]: 'activate' hardware 'ur5' 
[ur_ros2_control_node-1] [INFO] [1725962890.819073033] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1725962890.819092053] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1725962890.823825220] [UR_Client_Library:ur5_]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[robot_state_publisher-5] [INFO] [1725962890.831832775] [ur5.robot_state_publisher]: got segment ur5_base
[robot_state_publisher-5] [INFO] [1725962890.832904654] [ur5.robot_state_publisher]: got segment ur5_base_link
[robot_state_publisher-5] [INFO] [1725962890.833714417] [ur5.robot_state_publisher]: got segment ur5_base_link_inertia
[robot_state_publisher-5] [INFO] [1725962890.833744543] [ur5.robot_state_publisher]: got segment ur5_flange
[robot_state_publisher-5] [INFO] [1725962890.833762145] [ur5.robot_state_publisher]: got segment ur5_forearm_link
[robot_state_publisher-5] [INFO] [1725962890.833779817] [ur5.robot_state_publisher]: got segment ur5_ft_frame
[robot_state_publisher-5] [INFO] [1725962890.833795862] [ur5.robot_state_publisher]: got segment ur5_shoulder_link
[robot_state_publisher-5] [INFO] [1725962890.833817815] [ur5.robot_state_publisher]: got segment ur5_tool0
[robot_state_publisher-5] [INFO] [1725962890.833843922] [ur5.robot_state_publisher]: got segment ur5_tool_2f85
[robot_state_publisher-5] [INFO] [1725962890.833860395] [ur5.robot_state_publisher]: got segment ur5_tool_2f85_tactile
[robot_state_publisher-5] [INFO] [1725962890.833875854] [ur5.robot_state_publisher]: got segment ur5_tool_calib
[robot_state_publisher-5] [INFO] [1725962890.833891794] [ur5.robot_state_publisher]: got segment ur5_tool_epick
[robot_state_publisher-5] [INFO] [1725962890.833910550] [ur5.robot_state_publisher]: got segment ur5_upper_arm_link
[robot_state_publisher-5] [INFO] [1725962890.833935394] [ur5.robot_state_publisher]: got segment ur5_wrist_1_link
[robot_state_publisher-5] [INFO] [1725962890.833948184] [ur5.robot_state_publisher]: got segment ur5_wrist_2_link
[robot_state_publisher-5] [INFO] [1725962890.833961460] [ur5.robot_state_publisher]: got segment ur5_wrist_3_link
[robot_state_publisher-5] [INFO] [1725962890.833975286] [ur5.robot_state_publisher]: got segment world
[ur_ros2_control_node-1] [INFO] [1725962891.157190397] [UR_Client_Library:ur5_]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1725962891.158036672] [UR_Client_Library:ur5_]: Setting up RTDE communication with frequency 125.000000
[ur_ros2_control_node-1] [WARN] [1725962892.229232770] [UR_Client_Library:ur5_]: DEPRECATION NOTICE: Setting the keepalive count has been deprecated. Instead use the RobotReceiveTimeout, to set the timeout directly in the write commands. Please change your code to set the read timeout in the write commands directly. This keepalive count will overwrite the timeout passed to the write functions.
[ur_ros2_control_node-1] [WARN] [1725962892.229276883] [UR_Client_Library:ur5_]: DEPRECATION NOTICE: Setting the keepalive count has been deprecated. Instead you should set the timeout directly in the write commands. Please change your code to set the read timeout in the write commands  [... cut ...]
[robot_control-15] [INFO] [1725962901.082751023] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[robot_control-15] [INFO] [1725962901.119212286] [move_group_interface]: Ready to take commands for planning group ur5_manipulator.
[robot_control-15] [INFO] [1725962901.119919361] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[robot_control-15] [INFO] [1725962901.125733270] [xrm_robot_node]: Reference frame: world
[robot_control-15] [INFO] [1725962901.125854049] [xrm_robot_node]: End effector link: ur5_tool0
```

rhaschke, 2024-09-10:
The log states that the capability is correctly loaded.

minku1219, 2024-09-10:
I think you should notice this in the log
```
[move_group-12] [ERROR] [1725963244.479810848] [move_group.move_group]: Exception while loading move_group capability 'move_group/ExecuteTaskSolutionCapability': According to the loaded plugin descriptions the class move_group/ExecuteTaskSolutionCapability with base class type move_group::MoveGroupCapability does not exist. Declared types are  move_group/ApplyPlanningSceneService move_group/ClearOctomapService move_group/MoveGroupCartesianPathService move_group/MoveGroupExecuteService move_group/MoveGroupExecuteTrajectoryAction move_group/MoveGroupGetPlanningSceneService move_group/MoveGroupKinematicsService move_group/MoveGroupMoveAction move_group/MoveGroupPlanService move_group/MoveGroupQueryPlannersService move_group/MoveGroupStateValidationService move_group/TfPublisher pilz_industrial_motion_planner/MoveGroupSequenceAction pilz_industrial_motion_planner/MoveGroupSequenceService
```

rhaschke, 2024-09-10:
In this case, you probably didn't sourced your MTC workspace correctly.

minku1219, 2024-09-10:
Ah! Now i got the issue.
thanks for your valuable and prompt response!

minku1219, 2024-09-10:
Upon calling the execution, I had an error in Robot node side:-

```
[move_group-12] [WARN] [1725963972.080310394] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '3' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-12] [INFO] [1725963972.080372726] [moveit_task_constructor_visualization.execute_task_solution]: Executing TaskSolution
[move_group-12] [INFO] [1725963972.080424543] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-12] [INFO] [1725963972.080456608] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-12] [INFO] [1725963972.080560695] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-12] [INFO] [1725963972.084257908] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-12] [INFO] [1725963972.084367379] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-12] [INFO] [1725963972.084410606] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-12] [INFO] [1725963972.084949653] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-12] [INFO] [1725963972.084991267] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-12] [INFO] [1725963972.085282172] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to /ur5/scaled_joint_trajectory_controller
[ur_ros2_control_node-1] [INFO] [1725963972.085649461] [ur5.scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [INFO] [1725963972.085702989] [ur5.scaled_joint_trajectory_controller]: Accepted new action goal
[move_group-12] [INFO] [1725963972.085841741] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: /ur5/scaled_joint_trajectory_controller started execution
[move_group-12] [INFO] [1725963972.085899744] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ur_ros2_control_node-1] [WARN] [1725963972.091499784] [ur5.scaled_joint_trajectory_controller]: Aborted due to state tolerance violation
[move_group-12] [WARN] [1725963972.136140538] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller '/ur5/scaled_joint_trajectory_controller' failed with error PATH_TOLERANCE_VIOLATED: 
[move_group-12] [WARN] [1725963972.136223351] [moveit_ros.trajectory_execution_manager]: Controller handle /ur5/scaled_joint_trajectory_controller reports status ABORTED
[move_group-12] [INFO] [1725963972.136258886] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
```

Log in cartesian executable side:-

```
[cartesian-1] [INFO] [1725963972.010288576] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.055288 seconds
[cartesian-1] [INFO] [1725963972.010385900] [moveit_robot_model.robot_model]: Loading robot model 'ur5'...
[cartesian-1] [INFO] [1725963972.010397927] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[cartesian-1] Plan!
[cartesian-1] PN6moveit16task_constructor11TaskPrivateE
[cartesian-1] Init!
[cartesian-1] 1
[cartesian-1] [ERROR] [1725963972.144757040] [moveit_task_constructor_executor_140733679541664]: Goal was aborted or canceled

```

rhaschke, 2024-09-10:
This is obviously a controller issue. Maybe googling the error message helps, e.g.
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/1072

minku1219, 2024-09-10:
Do you have any script which we can use to run any example of MTC with UR.
Since i am modifying the cartesian.cpp, i think there must be an script which can directly be used with UR

rhaschke, 2024-09-10:
I don't have a UR available and thus cannot provide a script that is tested on a UR.
MoveIt and MTC are essentially robot-agnostic. They run on any robot as long as the robot controller satisfies the required API. Check the `example_move.py` script suggested in the above link.

minku1219, 2024-09-10:
Sure! Thanks.

I just wanted to know how we can set the initial state as the current state in the scripts?

```
// start from a fixed robot state
	t.loadRobotModel(node);
	auto scene = std::make_shared<planning_scene::PlanningScene>(t.getRobotModel());
	{
		auto& state = scene->getCurrentState();
		// state.setToDefaultValues(state.getJointModelGroup(group), "ready");

		auto fixed = std::make_unique<stages::FixedState>("initial state");
		fixed->setState(scene);
		t.add(std::move(fixed));
	}
```

I want my robot to start execution from current state instead of the fixed state.
On debugging i came to know that it is starting from a fixed  robot state which is causing the path tolerance issue

minku1219, 2024-09-10:
As in moveit API we have a direct function i.e. setStartStateToCurrentState().

rhaschke, 2024-09-10:
See the docs or have a look at another example: 
https://github.com/moveit/moveit_task_constructor/blob/5a4480814628db4207d9734d63dd907619ced93b/demo/src/modular.cpp#L101

minku1219, 2024-09-11:
Hi, Nothing is available in the docs.
I tried this but it did'nt worked for me.

minku1219, 2024-09-11:
Error Log after using it.

```
[modular-1] [INFO] [1726036561.404413082] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0772726 seconds
[modular-1] [INFO] [1726036561.404535399] [moveit_robot_model.robot_model]: Loading robot model 'ur5'...
[modular-1] [INFO] [1726036561.404546615] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[modular-1] Plan!
[modular-1] PN6moveit16task_constructor11TaskPrivateE
[modular-1] Init!
[modular-1] [WARN] [1726036564.470796678] [CurrentState]: failed to acquire current PlanningScene
[modular-1]   0  - ←   0 →   -  0 / Reusable Containers
[modular-1]     0  - ←   0 →   -  0 / current
[modular-1]     -  0 →   0 →   -  0 / Cartesian Path
[modular-1]       -  0 →   0 →   -  0 / x +0.2
[modular-1] Failing stage(s):
[modular-1] -1
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[modular-1] [INFO] [1726036572.663836270] [rclcpp]: signal_handler(signum=2)
[INFO] [modular-1]: process has finished cleanly [pid 30914]
```

minku1219, 2024-09-11:
Another Log with cartesian example:-

```
[cartesian-1] [INFO] [1726036800.764109858] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0543219 seconds
[cartesian-1] [INFO] [1726036800.764200250] [moveit_robot_model.robot_model]: Loading robot model 'ur5'...
[cartesian-1] [INFO] [1726036800.764226708] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[cartesian-1] Plan!
[cartesian-1] PN6moveit16task_constructor11TaskPrivateE
[cartesian-1] Init!
[cartesian-1] planning failed with exception
[cartesian-1] Error initializing stage:
[cartesian-1] Cartesian Path: cannot connect end interface of 'current' (→) to start interface of 'initial state' (←)
[cartesian-1]   0  - ←   0 →   -  0 / Cartesian Path
[cartesian-1]     0  - ←   0 →   -  - / current
[cartesian-1]     -  - ←   0 →   -  0 / initial state
[cartesian-1]     -  0 →   0 →   -  0 / x +0.2
```

minku1219, 2024-09-11:
Cartesian Example code :- 

```
Task createTask(const rclcpp::Node::SharedPtr& node) {
	Task t;
	t.stages()->setName("Cartesian Path");

	const std::string group = "ur5_manipulator";
	const std::string eef = "ur5_tool0";

	// create Cartesian interpolation "planner" to be used in various stages
	auto cartesian_interpolation = std::make_shared<solvers::CartesianPath>();
	// create a joint-space interpolation "planner" to be used in various stages
	auto joint_interpolation = std::make_shared<solvers::JointInterpolationPlanner>();

	// start from a fixed robot state
	t.loadRobotModel(node);
	t.add(std::make_unique<stages::CurrentState>("current"));

	auto scene = std::make_shared<planning_scene::PlanningScene>(t.getRobotModel());
	{
		auto& state = scene->getCurrentStateNonConst();
		state.setToDefaultValues(state.getJointModelGroup(group), "ready");

		auto fixed = std::make_unique<stages::FixedState>("initial state");
		fixed->setState(scene);
		t.add(std::move(fixed));
	}

	{
		auto stage = std::make_unique<stages::MoveRelative>("x +0.2", cartesian_interpolation);
		stage->setGroup(group);
		stage->setIKFrame("ur5_tool0");
		geometry_msgs::msg::Vector3Stamped direction;
		direction.header.frame_id = "world";
		direction.vector.x = 0.02;
		stage->setDirection(direction);
		t.add(std::move(stage));
	}

	return t;
}

int main(int argc, char** argv) {
	rclcpp::init(argc, argv);
	auto node = rclcpp::Node::make_shared("mtc_tutorial");
	std::thread spinning_thread([node] { rclcpp::spin(node); });

	auto task = createTask(node);
	try {
		if (task.plan()){
			task.introspection().publishSolution(*task.solutions().front());
			task.execute(*task.solutions().front());
		}
	} catch (const InitStageException& ex) {
		std::cerr << "planning failed with exception" << std::endl << ex << task;
	}

	// keep alive for interactive inspection in rviz
	spinning_thread.join();
	return 0;
}
```

minku1219, 2024-09-11:
Modular.cpp code :-

```
std::unique_ptr<SerialContainer> createModule(const std::string& group) {
	auto c = std::make_unique<SerialContainer>("Cartesian Path");
	c->setProperty("group", group);

	// create Cartesian interpolation "planner" to be used in stages
	auto cartesian = std::make_shared<solvers::CartesianPath>();
	// create joint interpolation "planner"
	auto joint_interpolation = std::make_shared<solvers::JointInterpolationPlanner>();

	{
		auto stage = std::make_unique<stages::MoveRelative>("x +0.2", cartesian);
		stage->properties().configureInitFrom(Stage::PARENT, { "group" });
		stage->setIKFrame("ur5_tool0");
		geometry_msgs::msg::Vector3Stamped direction;
		direction.header.frame_id = "world";
		direction.vector.x = 0.2;
		stage->setDirection(direction);
		c->insert(std::move(stage));
	}

	
	return c;
}

Task createTask(const rclcpp::Node::SharedPtr& node) {
	Task t;
	t.loadRobotModel(node);
	t.stages()->setName("Reusable Containers");
	t.add(std::make_unique<stages::CurrentState>("current"));

	const std::string group = "ur5_manipulator";
	t.add(createModule(group));

	return t;
}

int main(int argc, char** argv) {
	rclcpp::init(argc, argv);
	auto node = rclcpp::Node::make_shared("mtc_tutorial");
	std::thread spinning_thread([node] { rclcpp::spin(node); });

	auto task = createTask(node);
	try {
		if (task.plan()){
			task.introspection().publishSolution(*task.solutions().front());
			task.execute(*task.solutions().front());
		}
	} catch (const InitStageException& ex) {
		std::cerr << "planning failed with exception" << std::endl << ex << task;
	}

	// keep alive for interactive inspection in rviz
	spinning_thread.join();
	return 0;
}
```

minku1219, 2024-09-11:
Hi @rhaschke,

Thank you for your support.
I managed to get this to work with my UR5.
I will be now working more precisely with Pick Place demo.

minku1219, 2024-09-12:
HI @rhaschke ,

While going through the demo examples, i encountered with an issue.

Code Snippet:-

```
{  // move back to ready pose
		auto stage = std::make_unique<stages::MoveTo>("moveTo ready", joint_interpolation);
		stage->setIKFrame("ur5_tool0");
		stage->properties().configureInitFrom(Stage::PARENT);
		stage->setGoal("ready");
		c->insert(std::move(stage));
	}
```

Error log:-

```
[ur_test-1] [INFO] [1726119849.361650285] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0595334 seconds
[ur_test-1] [INFO] [1726119849.361747493] [moveit_robot_model.robot_model]: Loading robot model 'ur5'...
[ur_test-1] [INFO] [1726119849.361767808] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ur_test-1] Plan!
[ur_test-1] PN6moveit16task_constructor11TaskPrivateE
[ur_test-1] Init!
[ur_test-1] planning failed with exception
[ur_test-1] Error initializing stage:
[ur_test-1] moveTo ready: Unknown joint pose: ready
```

It seems that the joint goal with name "ready" is not available in the API.
Could you please help me to how to assign these values in the code?

rhaschke, 2024-09-12:
Named poses, like `ready`, are stored in the [SRDF file](https://moveit.github.io/moveit_tutorials/doc/urdf_srdf/urdf_srdf_tutorial.html?highlight=srdf#robot-poses) of your robot.

minku1219, 2024-09-17:
Thanks @rhaschke for the response.
Sorry for the delay in response from my side, I was occupied with something else.

minku1219, 2024-09-18:
Hi @rhaschke
Now as i am heading forward towards using this MTC with my perception pipeline, I am encountering an issue on how to set the pose in it.

code snippet I am using :-

```
{  // rotate about TCP
		auto stage = std::make_unique<stages::GeneratePose>("generate pose");
		stage->properties().configureInitFrom(Stage::PARENT, { "group" });
		stage->setMonitoredStage(initial_state_ptr);
		// stage->setIKFrame("ur5_tool0");
		geometry_msgs::msg::PoseStamped pose;
		pose.header.frame_id = "world";
		pose.pose.position.x = -0.02;
		stage->setPose(pose);
		auto wrapper = std::make_unique<stages::ComputeIK>("grasp pose IK", std::move(stage));
		wrapper->setMaxIKSolutions(8);  // limit number of solutions
		wrapper->setMinSolutionDistance(1.0);
		// define virtual frame to reach the target_pose
		wrapper->setIKFrame("ur5_tool0");
		wrapper->properties().configureInitFrom(Stage::PARENT);  // inherit properties from parent
		c->insert(std::move(wrapper));
		// c->insert(std::move(stage));
	}
```

This is the snippet I am trying to make use of.

Log output for the reference :-
```
[INFO] [ur_test-1]: process started with pid [86044]
[ur_test-1] [INFO] [1726657893.407109381] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.063451 seconds
[ur_test-1] [INFO] [1726657893.407197821] [moveit_robot_model.robot_model]: Loading robot model 'ur5'...
[ur_test-1] [INFO] [1726657893.407210253] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ur_test-1] Plan!
[ur_test-1] PN6moveit16task_constructor11TaskPrivateE
[ur_test-1] Init!
[ur_test-1] planning failed with exception
[ur_test-1] Error initializing stage:
[ur_test-1] generate pose: no monitored stage defined
[ur_test-1]   0  - ←   0 →   -  0 / Reusable Containers
[ur_test-1]     -  - ←   0 →   -  - / current
[ur_test-1]     -  - ?   0 ?   -  - / Cartesian Path
[ur_test-1]       -  - ?   0 ?   -  - / grasp pose IK
[ur_test-1]         -  - ←   0 →   -  - / generate pose
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```

Hope to see your response ASAP.

rhaschke, 2024-09-18:
Please state a concrete question. Skimming through the log, I see the error `generate pose: no monitored stage defined`,
which means that you didn't set a monitored stage. This should be done here:
`stage->setMonitoredStage(initial_state_ptr);`
If that still yields the mentioned error, this obviously means that `initial_state_ptr` is `NULL`.

When reading the logs and your code, you should have been able to come to that conclusion yourself.
github issues are not intended to provide usage support, but to report real bugs!

minku1219, 2024-09-19:
I apologize for the oversight and will refrain from using GitHub issues for usage support in the future.

Thanks again for your help!