moveit/moveit_task_constructor #613: moveit task constructor execution results are inconsistent with the planned results in gazebo

LiangXin-gugu, 2024-09-23:
**Environment:**
ros2 version: humble
gazebo version: Fortress

**Problem Description:**
I used MTC to plan and execute a pick place task, and MTC planned an solution. 
[image]
[image]
But when I execute the solution, gazebo does not strictly follow the planned order. It did not wait until one subtask was completed before starting the next subtask. Instead, a transition stage seemed to be added between each sub-task, which resulted in each sub-task not being fully executed, ultimately leading to task failure.
For example, theoretically, the robot arm should move to the grasping position（i.e. complete move to pick stage）, then approach object, and then close gripper.
But in actual execution, the robot arm seems to start to approach the object before it reaches the grasping position, and closes the gripper at the same time.
But when I execute individual sub stage step by step, it is able to complete each sub stage, and then finish the pick place task as planned in gazebo.

**My Question:**
1. Why are the step-by-step execution results different from the total task execution results?
2. How to make the total task execution result the same as the planning result, so that the robotic arm can complete the pick place task the same as the plan.

rhaschke, 2024-09-23:
I cannot imagine any reason for the described behavior. Individual sub-stages are strictly executed in order.
It might be possible that your (virtual) robot controller in Gazebo reports success prematurely though, in which case the next stage will be executed before the previous one is correctly finished.

LiangXin-gugu, 2024-09-24:
> I cannot imagine any reason for the described behavior. Individual sub-stages are strictly executed in order. It might be possible that your (virtual) robot controller in Gazebo reports success [...]

Thanks for your reply. Did you mean that MTC will receive feedback from the (virtual) robot controller, and only when it receives feedback that current sub-stage has been executed completely, will MTC send the task instructions of the next sub-stage to the controller.

rhaschke, 2024-09-24:
Yes.

LiangXin-gugu, 2024-09-28:
Thanks rhaschke! I found the reason, it is as @rhaschke speculated:

> It might be possible that your (virtual) robot controller in Gazebo reports success prematurely, in which case the next stage will be executed before the previous one is correctly finished.

I used joint_trajectory_controller(JTC) provied by ros2_controllers. But I didn't set any constraints for JTC in **controllers.yaml**, which means there is only default constraint: **stopped_velocity_tolerance=0.01**.
So once the task execution time planned by MTC is arrived, as long as the joints speeds are less than 0.01rad/s at this time, the controller will report success. Since the "**constraints.joints.goal**" is not set, the controller will not check whether the actual joint angle reaches near the goal angle.
Besides, the **max_velocity** and the **max_acceleration** in **joint limits.yaml** is set too high, which exceeds the actual joint limits in gazebo, so the joints can't reach goal angle as fast as MTC planned.
Therefore, I did two things to sovle the problem:
1. Set JTC constraints in **controllers.yaml**
```
controller_manager:
  ros__parameters:
    update_rate: 10
    arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController
arm_controller:
  ros__parameters:
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
    command_interfaces:
      - position
    state_interfaces:
      - position 
    open_loop_control: false
    constraints:
      stopped_velocity_tolerance: 0.001
      goal_time: 2.0
      shoulder_pan_joint:
        goal: 0.05
        trajectory: 0.0
```
2. Lower the **max_velocity** and the **max_acceleration** in **joint limits.yaml**
```
joint_limits:
  shoulder_pan_joint:
    has_velocity_limits: true
    max_velocity: 1.57
    has_acceleration_limits: true
    max_acceleration: 0.26
    has_deceleration_limits: true
    max_deceleration: -0.26
```

_reference:_ 
1. [https://robotics.stackexchange.com/questions/104784/joint-trajectory-controller-with-velocity-command-interface-reports-success-to-e?rq=1](url)
4. [https://control.ros.org/humble/doc/ros2_controllers/joint_trajectory_controller/doc/parameters.html](url)

closed 2024-09-28 by LiangXin-gugu