moveit/moveit_task_constructor #627: isaac_ros_cumotion pipeline planner in Connect stage suddenly jerks and changes joint values at last waypoint of the trajectory

bornaparo, 2024-11-09:
Hi, I created pick and place task with moveit task constructor following the [tutorial](https://moveit.picknik.ai/main/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html), and it works. Then I wanted to try using it with NVIDIA's isaac ros cumotion planner but it doesn't work. Connect stage creates trajectory that at last waypoint suddenly changes joint values. I tried changing a lot of properties of the connect stages and of pipeline planner, also have tried changing a lot of parameters in `isaac_ros_cumotion/cumotion_planner.py`  but nothing helped. When I use isaac_ros_cumotion planner just for MoveTo stage then it works without the mentioned problems, so I guess the problem has to be with Connect stage but I cannot figure out where.

Here is a video of the problem:
https://github.com/user-attachments/assets/66b01b1b-e40b-4955-af3d-a0f2dd40a24d

[Here is an output of mtc node](https://pastebin.com/FneTWe7Z) where it says `Trajectory success!` so I guess it is not a problem with cumotion. Also cumotion works correctly when I tried using it in RViz to plan around obstacles.

I am using moveit2 from humble branch and ros2 humble in docker.

To reproduce this error:
- Create the same package with the same files as in [tutorial](https://moveit.picknik.ai/main/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html), or modify the already created files in `moveit2_tutorials/doc/tutorials/pick_and_place_with_moveit_task_constructor/`
- install isaac_ros_cumotion moveit plugin from [here](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_cumotion/isaac_ros_cumotion_moveit/index.html), I installed it from source in my own docker container
- copy `isaac_ros_cumotion_moveit/config/isaac_ros_cumotion_planning.yaml` to `moveit_resources/panda_moveit_config/config/`
- in `mtc_demo.launch.py`:
  - replace `pipelines=["ompl", "chomp", "pilz_industrial_motion_planner"]` with `pipelines=["ompl", "chomp", "pilz_industrial_motion_planner", "isaac_ros_cumotion"]`
  - add curobo planner node: 
  ```python
  curobo_planner_node = Node( 
        package="isaac_ros_cumotion",
        executable="cumotion_planner_node",
        output="screen",
        parameters=[
            {"robot": "franka.xrdf"},
            {"urdf_path": os.path.join(get_package_share_directory("moveit_resources_panda_description"), "urdf", "panda.urdf")},
        ]
    )
  ``` 
     and add it to the LaunchDescription
  - in `mtc_node.cpp` replace `auto sampling_planner = std::make_shared<mtc::solvers::PipelinePlanner>(node_);` with `auto sampling_planner = std::make_shared<mtc::solvers::PipelinePlanner>(node_, "isaac_ros_cumotion");`
  - build the package and run it:
    - `ros2 launch mtc_tutorial mtc_demo.launch.py` and wait for cumotion to warmup
    - `ros2 launch mtc_tutorial pick_place_demo.launch.py`

Any help is appreciated!

rhaschke, 2024-11-11:
Looks like cumotion only cares about the Cartesian end-effector pose, but ignores the joint pose:
As you can see from the rviz visualization, the generated paths reach the end-effector pose and then jump to a different joint-pose.
This also explains, why it works for MoveTo but not for Connect. The latter requires planning from a joint-pose to another joint-pose.
I suggest to file an issue with cumotion to check whether this mode of operation is supported somehow.

closed 2024-11-11 by rhaschke

bornaparo, 2024-11-18:
In `cumotion_planner.py` I replaced
```python
motion_gen_result = self.motion_gen.plan_single(
            start_state,
            goal_pose,
            MotionGenPlanConfig(max_attempts=5, enable_graph_attempt=1,
                                time_dilation_factor=time_dilation_factor),
        )
```
with
```python
motion_gen_result = self.motion_gen.plan_single_js(
            start_state,
            goal_state,
            MotionGenPlanConfig(max_attempts=5, enable_graph_attempt=1,
                                time_dilation_factor=time_dilation_factor),
        )
```
so it does joint state to joint state planning, but the problem is the same as before?

rhaschke, 2024-11-18:
Then, probably, your changed code is not used? Add a debug message before that line and check whether it pops up in the log.

bornaparo, 2024-11-23:
I changed from `print(...)` to `self.get_logger().info(...)` and now message is being printed and method is called as it should. So I guess the problem isn't with cumotion but somewhere with connect stage?

rhaschke, 2024-12-02:
Are you sure cumotion's  `plan_single_js` is doing what it should? Planning with other planners in the Connect stage works as expected. The problem is that the cumotion planner doesn't respect the given (joint-space) goal_state.

kirienkomaxym, 2024-12-05:
Facing the issue with setting Cumotion as a planner for MTC Tasks using ur5e robot and robotiq gripper, but not sure is this actually the MTC-related issue, however, I will append the thread with results of my research about setting up Cumotion with MTC

So the general, after setting the cumotion to plan with IK and changing the linear pose to joint pose: 
```
motion_gen_result = self.motion_gen.plan_single_js(
    start_state,
    goal_state,  # it is now a joint state, not the recalculated forward kin pose
    MotionGenPlanConfig(max_attempts=5, enable_graph_attempt=1,
                        time_dilation_factor=time_dilation_factor),
)
```
the planning is considered failed on the MTC side (without actual exception), while using rviz2 motion gui to plan with the same start and end joint positions succeeds, and the action response and request structure is almost identical.

I am using almost the same approach for a pick place task as it is described in MTC demos, with the same planners: cartesian for linear movement, sampling planner for the joint-space planning (move to target and move to pick) and interpolation planner for the gripper:
```
auto sampling_planner = std::make_shared<solvers::PipelinePlanner>(node);
sampling_planner->setProperty("planning_pipeline", "isaac_ros_cumotion");
sampling_planner->setPlannerId("cuMotion");

auto cartesian_planner = std::make_shared<solvers::CartesianPath>();
cartesian_planner->setProperty("max_velocity_scaling_factor", 0.1);
cartesian_planner->setProperty("max_acceleration_scaling_factor", 0.1);
cartesian_planner->setProperty("planning_attempts", 10);
cartesian_planner->setProperty("planning_time", 5.0);
cartesian_planner->setStepSize(.02);

auto interpolation_planner = std::make_shared<solvers::JointInterpolationPlanner>();
```
so when I do the motion planning with the MTC pipeline, it ends up with a following error:
```
[pick_place_demo-3]   0  - ←   0 →   -  0 / pick_place_task
[pick_place_demo-3]     1  - ←   1 →   -  0 / initialization
[pick_place_demo-3]       1  - ←   1 →   -  0 / applicability test
[pick_place_demo-3]         1  - ←   1 →   -  1 / current state
[pick_place_demo-3]       -  0 →   1 →   -  1 / allow collision (hand,object)
[pick_place_demo-3]     -  0 →   1 →   -  1 / open hand
[pick_place_demo-3]     -  1 →   0 ←  12  - / move to pick
[pick_place_demo-3]    12  - ←  12 →   - 12 / pick object
[pick_place_demo-3]      12  - ←  12 ←   0  - / approach object
[pick_place_demo-3]       0  - ←  14 →   -  0 / grasp alternatives
[pick_place_demo-3]        14  - ←   3 →   - 14 / grasp pose IK
[pick_place_demo-3]           3  - ←   3 →   -  3 / generate grasp pose
[pick_place_demo-3]        14  - ←   3 →   - 14 / grasp pose IK
[pick_place_demo-3]           3  - ←   3 →   -  3 / generate grasp pose
[pick_place_demo-3]        14  - ←   3 →   - 14 / grasp pose IK
[pick_place_demo-3]           3  - ←   3 →   -  3 / generate grasp pose
[pick_place_demo-3]        14  - ←   3 →   - 14 / grasp pose IK
[pick_place_demo-3]           3  - ←   3 →   -  3 / generate grasp pose
[pick_place_demo-3]        14  - ←   2 →   - 14 / grasp pose IK
[pick_place_demo-3]           3  - ←   3 →   -  3 / generate grasp pose
[pick_place_demo-3]       -  0 →  14 →   -  0 / close hand
[pick_place_demo-3]       -  0 →  14 →   -  0 / attach object
[pick_place_demo-3]       -  0 →  13 →   - 13 / lift object
[pick_place_demo-3]     - 12 →   0 ←  12  - / move to place
[pick_place_demo-3]    12  - ←  12 →   - 12 / place object
[pick_place_demo-3]      21  - ←  21 ←   0  - / lower object
[pick_place_demo-3]       0  - ←  24 →   -  0 / place pose IK
[pick_place_demo-3]        24  - ←  24 →   - 24 / generate place pose
[pick_place_demo-3]       -  0 →  24 →   -  0 / open hand
[pick_place_demo-3]       -  0 →  24 →   -  0 / detach object
[pick_place_demo-3]       -  0 →  12 →   -  0 / retreat after place
[pick_place_demo-3]       -  0 →  12 →   - 12 / forbid collision (hand,object)
[pick_place_demo-3] Failing stage(s):
[pick_place_demo-3] move to pick (0/8): SUCCESS
[pick_place_demo-3] [ERROR] [1733412203.662779097] [mercurio_motion]: Task planning failed.
[pick_place_demo-3] [ERROR] [1733412203.662793914] [mercurio_motion]: Number of failed stages: 0
[pick_place_demo-3] [ERROR] [1733412203.662796781] [pick_place_demo]: Failed to plan PickPlaceTask
```
I have the code that does iterate over stages, to check where the error is:
```
bool PickPlaceTask::plan(const std::size_t max_solutions) {
    RCLCPP_INFO(LOGGER, "Start searching for task solutions");

    try {
        if (!task_->plan(20.0)) {
            RCLCPP_ERROR(LOGGER, "Task planning failed.");
            const auto& failures = task_->failures();
            RCLCPP_ERROR(LOGGER, "Number of failed stages: %zu", failures.size());
            for (const auto& failure : failures) {
                RCLCPP_ERROR_STREAM(LOGGER, "Failure description: " << failure->comment());
            }
            return false;
        }
```
however, I can not see the error because `Number of failed stages: 0`.

So then, I have unpacked the request from the cumotion side, and tried to set these joint values via rviz2 motion planning gui, and it planned successfully. I have checked the request and response from both attemps, and they look the same to me, however, their structure could be discovered in cumotion logs.

[... cut ...] 492d-a2da-5ad9c276cf41) from cumotion when setting joint values via rviz, also adding the two videos – [attempt](https://github.com/user-attachments/assets/80e5d28e-a57d-42b2-af29-972abf33ad81
) to plan using MTC, and [successfull planning](https://github.com/user-attachments/assets/b7f2a2f5-c7b2-45fe-9c0a-5f5b1a33c7e7)
 by setting required joint values via rviz2.

If you have any ideas how I could investigate the errors deeper, or what should I check out to resolve the issue, it would be great!

rhaschke, 2024-12-06:
@kirienkomaxym: Your code to check where the error is, is not needed (as already provided by the function `explainFailures()` called in `Task::plan()`) and not working (as failures are stored only for leaf stages, not containers).
The failing stage returned `SUCCESS` for its failures, which is indeed not very helpful. You could use the rviz visualization of MTC to inspect individual stage's solutions and failures.
Looking into your MoveIt log, I see many `Timed out!` errors, which do not originate from MoveIt or MTC as far as I can tell.
The error seen in this thread before, was that cumotion doesn't correctly plan to joint-space poses.

rhaschke, 2025-05-14:
This is exactly what I proposed. Thanks for implementing this.

> That said, I'm still working on integrating MTC and Cumotion.

What exactly do you mean by integration here? As I understood, with your change, the integration as a MoveIt-planning-pipeline planner is working as expected. What else is missing in your point of view?

rhaschke, 2025-05-14:
These are all issues with the cumotion planner. Unfortunately, I will not have the time to look into that anytime soon.