moveit/moveit_task_constructor #631: Missing ik_frame when trying to connect two SerialContainers with MoveRelative

ralphieraccoon, 2024-11-26:
I have the following that generates a pick and place motion (without a gripper, this is for a tool changer):

```cpp
    std::unique_ptr<moveit::task_constructor::stages::CurrentState>
        current_state(new moveit::task_constructor::stages::CurrentState(
            "current state"));

    moveit::task_constructor::Stage *monitor_ptr = current_state.get();

    std::unique_ptr<moveit::task_constructor::stages::CurrentState> end_state(
        new moveit::task_constructor::stages::CurrentState("end state"));

    tool_change_task_->add(std::move(current_state));

    std::unique_ptr<moveit::task_constructor::stages::Connect>
        location_connector(new moveit::task_constructor::stages::Connect(
            "move to tool",
            moveit::task_constructor::stages::Connect::GroupPlannerVector{
                {this->get_parameter("group_name").as_string(),
                 sampling_planner}}));
    location_connector->setTimeout(5.0);
    location_connector->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT);

    tool_change_task_->add(std::move(location_connector));

    if (!active_tool_.empty()) { // Place existing tool.
      tool_change_task_->add(std::move(create_pick_place(
          active_tool_, cartesian_planner, monitor_ptr)));
    }

    if (!active_tool_.empty() &&
        !goal_handle->get_goal()->tool.empty()) { // Move to pick tool.

      std::unique_ptr<moveit::task_constructor::stages::Connect>
          pick_place_connector(new moveit::task_constructor::stages::Connect(
              "move to next tool",
              moveit::task_constructor::stages::Connect::GroupPlannerVector{
                  {this->get_parameter("group_name").as_string(),
                   cartesian_planner}}));
      pick_place_connector->setTimeout(5.0);
      pick_place_connector->properties().configureInitFrom(
          moveit::task_constructor::Stage::PARENT);

      tool_change_task_->add(std::move(pick_place_connector));
    }

    if (!goal_handle->get_goal()->tool.empty()) { // Pick up new tool.
      tool_change_task_->add(
          std::move(create_pick_place(goal_handle->get_goal()->tool,
                                      cartesian_planner, monitor_ptr)));
    }

    std::unique_ptr<moveit::task_constructor::stages::Connect> end_connector(
        new moveit::task_constructor::stages::Connect(
            "move to end",
            moveit::task_constructor::stages::Connect::GroupPlannerVector{
                {this->get_parameter("group_name").as_string(),
                 sampling_planner}}));
    end_connector->setTimeout(5.0);
    end_connector->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT);

    tool_change_task_->add(std::move(end_connector));
    tool_change_task_->add(std::move(end_state));
```

function `create_pick_place()`:

```cpp
std::unique_ptr<moveit::task_constructor::SerialContainer> create_pick_place(
      std::string tool,
      std::shared_ptr<moveit::task_constructor::solvers::CartesianPath>
          cartesian_planner,
      moveit::task_constructor::Stage *monitor_ptr) {

    geometry_msgs::msg::Vector3Stamped vec;
    vec.header.frame_id = this->get_parameter("link_name").as_string();

    double tool_approach_offset =
        this->get_parameter("tool_approach_offset").as_double();

    tf2::Quaternion orientation;
    orientation.setRPY(0.0f, M_PI, 0.0f);

    std::vector<double> tool_box_origin =
        this->get_parameter("tool_box_origin").as_double_array();

    std::unique_ptr<moveit::task_constructor::SerialContainer> pick_place(
        new moveit::task_constructor::SerialContainer("pick/place tool"));

    tool_change_task_->properties().exposeTo(pick_place->properties(),
                                             {"eef", "group"});
    pick_place->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT, {"eef", "group"});

    std::unique_ptr<moveit::task_constructor::stages::MoveRelative>
        tool_approach(new moveit::task_constructor::stages::MoveRelative(
            "approach tool", cartesian_planner));
    vec.vector.z = tool_approach_offset;
    tool_approach->setIKFrame(this->get_parameter("link_name").as_string());
    tool_approach->setForwardedProperties({"ik_frame"});
    tool_approach->setDirection(vec);
    tool_approach->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT);
    tool_approach->setMinMaxDistance(tool_approach_offset,
                                     tool_approach_offset + 0.05);

    std::unique_ptr<moveit::task_constructor::stages::MoveRelative>
        tool_retract(new moveit::task_constructor::stages::MoveRelative(
            "retract tool", cartesian_planner));
    vec.vector.z = -tool_approach_offset;
    tool_retract->setIKFrame(this->get_parameter("link_name").as_string());
    tool_retract->setDirection(vec);
    tool_retract->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT);
    tool_retract->setMinMaxDistance(tool_approach_offset - 0.05,
                                    tool_approach_offset);

    std::unique_ptr<moveit::task_constructor::stages::GeneratePose>
        tool_location(new moveit::task_constructor::stages::GeneratePose(
            "tool location"));
    geometry_msgs::msg::PoseStamped pose;

    pose.header.frame_id = "world";

    geometry_msgs::msg::Point tool_origin = tools_.at(tool).tool_origin;

    pose.pose.position.x = tool_box_origin[0] + tool_origin.x;
    pose.pose.position.y = tool_box_origin[1] + tool_origin.y;
    pose.pose.position.z =
        tool_box_origin[2] + tool_origin.z +
        this->get_parameter("tool_approach_offset").as_double();

    pose.pose.orientation = tf2::toMsg(orientation);

    tool_location->setPose(pose);
    tool_location->setMonitoredStage(monitor_ptr);

    std::unique_ptr<moveit::task_constructor::stages::ComputeIK>
        tool_location_ik(new moveit::task_constructor::stages::ComputeIK(
            "tool location IK", std::move(tool_location)));
    tool_location_ik->setMaxIKSolutions(8);
    tool_location_ik->setMinSolutionDistance(1.0);
    tool_location_ik->setIKFrame(Eigen::Isometry3d().Identity(),
                                 this->get_parameter("link_name").as_string());
    tool_location_ik->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT, {"eef", "group"});
    tool_location_ik->properties().configureInitFrom(
        moveit::task_constructor::Stage::INTERFACE, {"target_pose"});

    pick_place->add(std::move(tool_approach));
    pick_place->add(std::move(tool_location_ik));
    pick_place->add(std::move(tool_retract));

    monitor_ptr = pick_place.get();

    return pick_place;
  }
```

I get the following error messages when I try to plan:

```
[robot_arm_core-17]   0  - ←   0 →   -  0 / tool change
[robot_arm_core-17]     1  - ←   1 →   -  1 / current state
[robot_arm_core-17]     -  1 →   1 ←   1  - / move to tool
[robot_arm_core-17]     1  - ←   1 →   -  1 / pick/place tool
[robot_arm_core-17]       1  - ←   1 ←   0  - / approach tool
[robot_arm_core-17]       0  - ←   1 →   -  0 / tool location IK
[robot_arm_core-17]         1  - ←   1 →   -  1 / tool location
[robot_arm_core-17]       -  0 →   1 →   -  1 / retract tool
[robot_arm_core-17]     -  1 →   0 ←   1  - / move to next tool
[robot_arm_core-17]     1  - ←   1 →   -  1 / pick/place tool
[robot_arm_core-17]       1  - ←   1 ←   0  - / approach tool
[robot_arm_core-17]       0  - ←   1 →   -  0 / tool location IK
[robot_arm_core-17]         1  - ←   1 →   -  1 / tool location
[robot_arm_core-17]       -  0 →   1 →   -  1 / retract tool
[robot_arm_core-17]     -  1 →   1 ←   1  - / move to end
[robot_arm_core-17]     1  - ←   1 →   -  1 / end state
[robot_arm_core-17] Failing stage(s):
[robot_arm_core-17] move to next tool (0/1): missing ik_frame
```

Is this a bug or am I doing something wrong? Thanks!

rhaschke, 2024-12-02:
For your `cartesian_planner` you need to define the `ik_frame` property. Your robot probably doesn't have a unique tip and thus you need to explicitly define, which tip link should move along the Cartesian path. Note that the Cartesian planner does not plan around obstacles!

ralphieraccoon, 2024-12-04:
> For your `cartesian_planner` you need to define the `ik_frame` property. Your robot probably doesn't have a unique tip and thus you need to explicitly define, which tip link should move along the Cartesian path. Note that the Cartesian planner does not plan around obstacles!

Thanks, this seems to work.

closed 2024-12-04 by ralphieraccoon

rhaschke, 2024-12-04:
@ralphieraccoon: How can we improve the error message? It stated that it was missing the `ik_frame`...