moveit/moveit_task_constructor #632: MTC tutorial fails to execute on franka panda real hardware

PietroValerio, 2024-11-27:
I'm trying execute the mtc pick and place example that is copied as-is from the tutorial for ros2 humble and I get this error:
```[move_group-2] [WARN] [1732697278.834175150] [moveit_robot_state.conversions]: The transform for multi-dof joints was specified in frame 'world' but it was not possible to transform that to frame 'panda_link0'
[move_group-2] [WARN] [1732697278.834689542] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '3' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-2] [WARN] [1732697278.834703942] [moveit_robot_state.conversions]: The transform for multi-dof joints was specified in frame 'world' but it was not possible to transform that to frame 'panda_link0'
[move_group-2] [WARN] [1732697278.834922998] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '4' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-2] [WARN] [1732697278.834933785] [moveit_robot_state.conversions]: The transform for multi-dof joints was specified in frame 'world' but it was not possible to transform that to frame 'panda_link0'
[move_group-2] [WARN] [1732697278.835019571] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '6' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-2] [WARN] [1732697278.835027951] [moveit_robot_state.conversions]: The transform for multi-dof joints was specified in frame 'world' but it was not possible to transform that to frame 'panda_link0'
[move_group-2] [WARN] [1732697278.835036766] [moveit_robot_state.conversions]: The transform for multi-dof joints was specified in frame 'world' but it was not possible to transform that to frame 'panda_link0'
[move_group-2] [WARN] [1732697278.835140434] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '10' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-2] [WARN] [1732697278.835149997] [moveit_robot_state.conversions]: The transform for multi-dof joints was specified in frame 'world' but it was not possible to transform that to frame 'panda_link0'
[move_group-2] [WARN] [1732697278.835163008] [moveit_robot_state.conversions]: The transform for multi-dof joints was specified in frame 'world' but it was not possible to transform that to frame 'panda_link0'
[move_group-2] [WARN] [1732697278.835373624] [moveit_task_constructor_visualization.execute_task_solution]: The trajectory of stage '12' from task '' does not have any controllers specified for trajectory execution. This might lead to unexpected controller selection.
[move_group-2] [WARN] [1732697278.835383271] [moveit_robot_state.conversions]: The transform for multi-dof joints was specified in frame 'world' but it was not possible to transform that to frame 'panda_link0'
[move_group-2] [ERROR] [1732697278.835434773] [moveit_task_constructor_visualization.execute_task_solution]: Could not find JointModelGroup that actuates {panda_joint1, panda_joint2, panda_joint3, panda_joint4, panda_joint5, panda_joint6, panda_joint7, panda_finger_joint1} 
```
On rviz I can see the planning that is looped, so the code fails on execution. As reference i put also the code for the task
```cpp 
#include <moveit/planning_scene/planning_scene.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit/task_constructor/solvers.h>
#include <moveit/task_constructor/stages.h>
#include <moveit/task_constructor/task.h>

#include <rclcpp/rclcpp.hpp>
#if __has_include(<tf2_geometry_msgs/tf2_geometry_msgs.hpp>)
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
#else
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
#endif
#if __has_include(<tf2_eigen/tf2_eigen.hpp>)
#include <tf2_eigen/tf2_eigen.hpp>
#else
#include <tf2_eigen/tf2_eigen.h>
#endif

static const rclcpp::Logger LOGGER = rclcpp::get_logger("mtc_tutorial");
namespace mtc = moveit::task_constructor;

class MTCTaskNode {
 public:
  MTCTaskNode(const rclcpp::NodeOptions &options);

  rclcpp::node_interfaces::NodeBaseInterface::SharedPtr getNodeBaseInterface();

  void doTask();

  void setupPlanningScene();

 private:
  // Compose an MTC task from a series of stages.
  mtc::Task createTask();
  mtc::Task task_;
  rclcpp::Node::SharedPtr node_;
};

rclcpp::node_interfaces::NodeBaseInterface::SharedPtr
MTCTaskNode::getNodeBaseInterface() {
  return node_->get_node_base_interface();
}

MTCTaskNode::MTCTaskNode(const rclcpp::NodeOptions &options)
    : node_{std::make_shared<rclcpp::Node>("mtc_node", options)} {}

void MTCTaskNode::setupPlanningScene() {
  moveit_msgs::msg::CollisionObject object;
  object.id = "object";
  object.header.frame_id = "world";
  object.primitives.resize(1);
  object.primitives[0].type = shape_msgs::msg::SolidPrimitive::CYLINDER;
[... 312 lines cut ...]
int main(int argc, char **argv) {
  rclcpp::init(argc, argv);

  rclcpp::NodeOptions options;
  options.automatically_declare_parameters_from_overrides(true);

  auto mtc_task_node = std::make_shared<MTCTaskNode>(options);
  rclcpp::executors::MultiThreadedExecutor executor;

  auto spin_thread =
      std::make_unique<std::thread>([&executor, &mtc_task_node]() {
        executor.add_node(mtc_task_node->getNodeBaseInterface());
        executor.spin();
        executor.remove_node(mtc_task_node->getNodeBaseInterface());
      });

  mtc_task_node->setupPlanningScene();
  mtc_task_node->doTask();

  spin_thread->join();
  rclcpp::shutdown();
  return 0;
}
```
As note an example should work both in simulation and on real hardware without any effort required. If you need more info I am available.

rhaschke, 2024-11-29:
There are two issues, both unrelated to MTC:
1. `The transform for multi-dof joints was specified in frame 'world' but it was not possible to transform that to frame 'panda_link0'`
  This means that your Panda launch files don't declare a static TF2 transform from world to panda_link0.
2. `Could not find JointModelGroup that actuates {panda_joint1, panda_joint2, panda_joint3, panda_joint4, panda_joint5, panda_joint6, panda_joint7, panda_finger_joint1}`
  This means that you don't have any controllers defined or your joints are named differently.

robotics-qc, 2025-09-16:
Hello @rhaschke 

I have a question along this thread

I have a robot arm and a multi fingered gripper - both are seperate planning groups (left_arm, left_hand) and they have their controllers implemented. 

I can open rviz, select each planning group and send them to poses etc - and also use MTC's moveTo stage to send them each to pre-defined SRDF poses. 

However when I do pick and place, then I see that the Connect stage cannot seem to find a path  unless I had a planner for both groups to the definition

```
        auto stage_move_to_pick = std::make_unique<moveit::task_constructor::stages::Connect>("move to pick",moveit::task_constructor::stages::Connect::GroupPlannerVector{ { arm_group_name, sampling_planner }, {hand_group_name, sampling_planner}});

```

Once I add both the planners in the vector, then I can generate a successful plan as needed. But when I try to execute it, it does not call the controller for the individual planning group but seems to expect a combined controller. (I have controllers for the gripper - all the finger joints) and the arm - each of which definitely work, even with MTC tasks)

[image]

Only Connect seems to struggle. 

Any way I can keep the controllers seperate and still use connect to make my task planning pipeline?

Here is my task for reference 
```
 moveit::task_constructor::Task pickObjectTask(std::string object){
        // Relevant groups and frames
        const auto& arm_group_name = "left_arm";
        const auto& hand_frame = "left_hand_palm_link";
        const auto& arm_frame = "torso_link";
        const auto& hand_group_name = "left_hand"; 

        // Initialize the task
        moveit::task_constructor::Task task;
        task.setName("pick object task : " + object);
        
        task.setProperty("group", arm_group_name);
        task.setProperty("eef", "left_palm");
        task.setProperty("ik_frame", hand_frame);

        //Load Robot Model
        task.loadRobotModel(shared_from_this());

        // Initialize planners
        auto cartesian_planner = std::make_shared<moveit::task_constructor::solvers::CartesianPath>();
        cartesian_planner->setMaxVelocityScalingFactor(1);
        cartesian_planner->setMaxAccelerationScalingFactor(1);
        cartesian_planner->setStepSize(.01);

        auto sampling_planner = std::make_shared<moveit::task_constructor::solvers::PipelinePlanner>(shared_from_this());
        sampling_planner->setProperty("goal_joint_tolerance", 1e-5); // If this is large then we get an error - that planned trajectory is too far from the goal etc
        sampling_planner->setMaxVelocityScalingFactor(1);
        sampling_planner->setMaxAccelerationScalingFactor(1);

        //Stage pointers
        moveit::task_constructor::Stage* attach_object_stage_pointer = nullptr;
        moveit::task_constructor::Stage* current_state_ptr = nullptr; 

        // 1. Current stage
        auto current_stage = std::make_unique<moveit::task_constructor::stages::CurrentState>("current");
          current_state_ptr = current_stage.get();
        task.add(std::move(current_stage));

        // 2. MoveTo arm to ready position
        auto move_to_ready_stage = std::make_unique<moveit::task_constructor::stages::MoveTo>("MoveTo predefined position (ready pose)", sampling_planner);
            move_to_ready_stage->setGroup(arm_group_name);
            move_to_ready_stage->setGoal("left_home");
        task.add(std::move(move_to_ready_stage));
            

[... 58 lines cut ...]
                vec1.vector.z = 1.0;
                lift_object_stage->setDirection(vec1);
            grasp->insert(std::move(lift_object_stage));
            
        }
        task.add(std::move(grasp));

        move_to_ready_stage = std::make_unique<moveit::task_constructor::stages::MoveTo>("MoveTo predefined position (drop pose)", sampling_planner);
            move_to_ready_stage->setGroup(arm_group_name);
            move_to_ready_stage->setGoal("left_home");
        task.add(std::move(move_to_ready_stage));

        // 7e. ModifyPlanningScene stage to attach the object frame to the hand frame to show us the visualization as if the object is being picked up
        auto detach_object_stage = std::make_unique<moveit::task_constructor::stages::ModifyPlanningScene>("detach object");
            detach_object_stage->detachObject(object, hand_frame);
        task.add(std::move(detach_object_stage));
        
        // 7d. ModifyPlanningScene stage allows us to modify collisions, remove objects, move objects, attach frames etc - anything to do with the current planning scene
        auto remove_object = std::make_unique<moveit::task_constructor::stages::ModifyPlanningScene>("remove  object");
            remove_object->removeObject(object);
        task.add(std::move(remove_object));

        return task;
    }
```