moveit/moveit_task_constructor #637: MoveIt task constructor not buildind

Abhishek8857, 2024-12-08:
`6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/utils.h:47,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage.h:43,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/container.h:41,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/container_p.h:41,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/src/container.cpp:37:
6126.3 /colcon_ws/src/moveit2/moveit_core/macros/include/moveit/macros/class_forward.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/merge.h:40,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/src/container.cpp:39:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/robot_model/robot_model.h:51:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    51 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/merge.h:41,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/src/container.cpp:39:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/robot_trajectory/robot_trajectory.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/merge.h:42,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/src/container.cpp:39:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/trajectory_processing/time_parameterization.h:14:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    14 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/src/container.cpp:41:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/planning_scene/planning_scene.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/src/container.cpp:42:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/trajectory_processing/time_optimal_trajectory_generation.h:52:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    52 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/marker_tools.h:5,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/src/marker_tools.cpp:1:
6126.3 /colcon_ws/src/moveit2/moveit_core/macros/include/moveit/macros/class_forward.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/src/marker_tools.cpp:2:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/planning_scene/planning_scene.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/storage.h:42,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/cost_terms.h:41,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/src/cost_terms.cpp:37:
6126.3 /colcon_ws/src/moveit2/moveit_core/macros/include/moveit/macros/class_forward.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/utils.h:47,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage.h:43,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:41,
6126.3                  from /colcon_ws/src/moveit_task_constructor/core/src/introspection.cpp:37:
6126.3 /colcon_ws/src/moveit2/moveit_core/macros/include/moveit/macros/class_forward.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/src/cost_terms.cpp:41:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/collision_detection/collision_common.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/src/marker_tools.cpp:3:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/robot_state/robot_state.h:51:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    51 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/src/cost_terms.cpp:42:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/robot_trajectory/robot_trajectory.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/src/cost_terms.cpp:43:
6126.3 /colcon_ws/install/moveit_core/include/moveit_core/moveit/planning_scene/planning_scene.h:50:77: note: ‘#pragma message: .h header is obsolete. Please use the .hpp header instead.’
6126.3    50 | #pragma message(".h header is obsolete. Please use the .hpp header instead.")
6126.3       |                                                                             ^
6126.3 In file included from /colcon_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/task.h:50,
[... 276 lines cut ...]
6126.3   113 |   PlanningPipeline(const moveit::core::RobotModelConstPtr& model, const std::shared_ptr<rclcpp::Node>& node,
6126.3       |   ^~~~~~~~~~~~~~~~
[... cut ...]
 120 | >>> RUN source /opt/ros/humble/setup.bash && \
 121 | >>>     MAKEFLAGS="-j4 -l2" colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 3 --symlink-install --executor sequential
 122 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c source /opt/ros/humble/setup.bash &&     MAKEFLAGS=\"-j4 -l2\" colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 3 --symlink-install --executor sequential" did not complete successfully: exit code: 2
`

getting this error

rhaschke, 2024-12-09:
Looks like you try building incompatible versions of MoveIt and MTC. Ensure to use matching branches, e.g. Humble + Humble.

Abhishek8857, 2024-12-10:
Solved @rhaschke . thanks

Abhishek8857, 2024-12-10:
Solved @rhaschke . Thanks