moveit/moveit_task_constructor #640: Error when using catkin_make MTC tutorial

He11oworldd, 2024-12-20:
I want to try running the MoveIt Task Constructor tutorial.Here are my steps：
**1.cd ~/ws_moveit/src
2.git clone https://github.com/moveit/moveit_task_constructor.git
3.rosdep install --from-paths . --ignore-src --rosdistro noetic
4.cd ..
5.catkin_make****
When I run 5.catkin_make in the terminal, an error message appears. I will capture the error message below:
My system environment：**Ubuntu 20.04 + ROS Noetic**

[image]

He11oworldd, 2024-12-20:
Unfortunately, I re-downloaded the MTC feature package according to your suggestion (git clone --recursive https://github.com/moveit/moveit_task_constructor.git), and catkin_make, but the same error still occurs.
The instructions I'm getting these errors are from https://moveit.github.io/moveit_tutorials/doc/moveit_task_constructor/moveit_task_constructor_tutorial.html
**Error：**
[image]

rhaschke, 2024-12-21:
Looks like MTC cannot be build with `catkin_make`. Try [catkin tools](https://catkin-tools.readthedocs.io/en/latest/index.html) instead: `catkin build`

He11oworldd, 2024-12-22:
> Looks like MTC cannot be build with `catkin_make`. Try [catkin tools](https://catkin-tools.readthedocs.io/en/latest/index.html) instead: `catkin build`

Thank you very much for your advice, the MTC tutorial ran successfully!But I am confused about the difference between catkin_make and catkin build. Why does catkin_make fail?

rhaschke, 2024-12-22:
> I am confused about the difference between catkin_make and catkin build. 

These are simply different build tools. catkin_make is very much outdated.

> Why does catkin_make fail?

I don't yet know.

closed 2024-12-22 by rhaschke