moveit/moveit_task_constructor #642: rviz_marker_tools CMake Error at CMakeLists.txt:6 (find_package):

Nanquaidi, 2024-12-29:
System Environment:
Ubuntu 22.04
ROS2 Humble

Problem Description:
When following the tutorial to install and build MoveIt2 from source, I encountered some issues that caused the build to fail. Upon checking the terminal output, the following errors were displayed:

garyhuang@garyhuang-ubuntu:~/Documents/cobot_ws$ colcon build --mixin release
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_pr2_description
Starting >>> srdfdom
Starting >>> moveit_common
Starting >>> launch_param_builder
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_prbt_support
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Starting >>> yolo_msgs
--- stderr: moveit_task_constructor_msgs
CMake Error at CMakeLists.txt:6 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< moveit_task_constructor_msgs [0.28s, exited with code 1]
--- stderr: rviz_marker_tools
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< rviz_marker_tools [0.29s, exited with code 1]
Aborted  <<< srdfdom [0.31s]
Aborted  <<< moveit_resources_prbt_support [0.30s]
Aborted  <<< moveit_resources_panda_description [0.31s]
Aborted  <<< moveit_resources_fanuc_description [0.30s]
Aborted  <<< moveit_common [0.31s]
Aborted  <<< moveit_resources_pr2_description [0.31s]
Aborted  <<< yolo_msgs [0.44s]
Aborted  <<< launch_param_builder [0.47s]

Summary: 0 packages finished [0.78s]
  2 packages failed: moveit_task_constructor_msgs rviz_marker_tools
  8 packages aborted: launch_param_builder moveit_common moveit_resources_fanuc_description moveit_resources_panda_description moveit_resources_pr2_description moveit_resources_prbt_support srdfdom yolo_msgs
  2 packages had stderr output: moveit_task_constructor_msgs rviz_marker_tools
  68 packages not processed

The problem seems to be related to rviz_marker_tools, which requires Catkin to build as it is a ROS1 package. However, I am using Ament to build in ROS2. How can I resolve this issue? Looking forward to your response.

JafarAbdi, 2024-12-29:
Looks like you checked the wrong branch, make sure you are using [humble](https://github.com/moveit/moveit_task_constructor/tree/humble) for this repo

Nanquaidi, 2024-12-29:
> Looks like you checked the wrong branch, make sure you are using [humble](https://github.com/moveit/moveit_task_constructor/tree/humble) for this repo

Thank you for your reply. However, I tried rebuilding using the Humble, Main, and ROS2 branches, but the following error occurred every time:

CMake Error at CMakeLists.txt:6 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

JafarAbdi, 2024-12-29:
Can you share the output of `vcs status .`?

closed 2024-12-29 by JafarAbdi