moveit/moveit_task_constructor #646: Unable to allow collisions between a link and a collision object if their name share the same prefix

DaniGarciaLopez, 2025-01-27:
I have a link with a collision geometry defined in the URDF named `holder` and a collision object named `tool`, which spawns inside the collision geometry at the start of the task  (allowing their collision using a `ModifyPlanningScene` stage). Then, I pick up the `tool` object, disallow the collision between `tool` and `holder`, and later place it back, allowing their collision again in the `ComputeIK stage` (passing a monitored stage with the ACM modified, as explained [here](https://github.com/moveit/moveit_task_constructor/issues/565#issuecomment-2099178457)). This works as expected, and I can correctly place the tool inside the collision geometry.

However, if I rename the link to `tool_holder`, the task fails in the `ComputeIK` stage. On the other hand, if the word "tool" appears at the end of the name (e.g., `holder_tool`), it works correctly. It seems there is an issue when allowing or disallowing collisions between a pair of objects that share the same prefix in their names (`tool` and `tool_holder`).

I’m not sure if this is a moveit or MTC issue, I couldn't find the root cause of it. I'm using ROS2 Jazzy with the latest commits from the moveit2 main branch and moveit_task_constructor ros2 branch.

Does anyone have an idea why this could be happening? Otherwise, I’ll try to prepare a small example in the next few days to reproduce the issue.

Thanks!

rhaschke, 2025-01-27:
I'm not aware that name matching uses a prefix only. Please provide a minimal example. Thanks.

DaniGarciaLopez, 2025-02-03:
Looking deeper into the problem, we realized that it was actually on our end. No issue here. Thanks anyway!

closed 2025-02-03 by DaniGarciaLopez