moveit/moveit_task_constructor #650: ComputeIK fails when any collision objects are in the scene.

ralphieraccoon, 2025-01-30:
I've got a tool swapping function using moveit task constructor:

```cpp
bool swap_tool(std::string tool)
{

    std::shared_ptr<moveit::task_constructor::solvers::CartesianPath>
        cartesian_planner(
            new moveit::task_constructor::solvers::CartesianPath());

    cartesian_planner->setMaxVelocityScalingFactor(1.0);
    cartesian_planner->setMaxAccelerationScalingFactor(1.0);
    cartesian_planner->setStepSize(.01);
    cartesian_planner->setIKFrame(this->get_parameter("link_name").as_string());

    std::shared_ptr<moveit::task_constructor::solvers::PipelinePlanner>
        sampling_planner(new moveit::task_constructor::solvers::PipelinePlanner(
            std::make_shared<rclcpp::Node>(this->get_name())));
    sampling_planner->setProperty("goal_joint_tolerance", 1e-5);
    sampling_planner->setMaxVelocityScalingFactor(1.0);
    sampling_planner->setMaxAccelerationScalingFactor(1.0);

    moveit::task_constructor::TaskPtr task(
        new moveit::task_constructor::Task());

    task->stages()->setName("tool change");
    task->loadRobotModel(std::make_shared<rclcpp::Node>(this->get_name()));
    task->setProperty("group", this->get_parameter("group_name").as_string());
    task->setProperty("eef", this->get_parameter("link_name").as_string());
    task->setProperty("ik_frame", this->get_parameter("link_name").as_string());

    std::unique_ptr<moveit::task_constructor::stages::CurrentState>
        current_state(new moveit::task_constructor::stages::CurrentState(
            "current state"));

    moveit::task_constructor::Stage *monitor_ptr = current_state.get();

    geometry_msgs::msg::Vector3Stamped vec;
    vec.header.frame_id = this->get_parameter("link_name").as_string();

    double tool_approach_offset =
        this->get_parameter("tool_approach_offset").as_double();

    tf2::Quaternion orientation;
    orientation.setRPY(0.0f, M_PI, M_PI);

    std::vector<double> tool_box_origin =
        this->get_parameter("tool_box_origin").as_double_array();

    std::unique_ptr<moveit::task_constructor::stages::MoveRelative>
        tool_approach(new moveit::task_constructor::stages::MoveRelative(
            "approach tool", cartesian_planner));
    vec.vector.z = tool_approach_offset;
    tool_approach->setIKFrame(this->get_parameter("link_name").as_string());
    tool_approach->setForwardedProperties({"ik_frame"});
    tool_approach->setDirection(vec);
    tool_approach->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT);
    tool_approach->setMinMaxDistance(tool_approach_offset,
                                     tool_approach_offset + 0.05);

    std::unique_ptr<moveit::task_constructor::stages::MoveRelative>
        tool_retract(new moveit::task_constructor::stages::MoveRelative(
            "retract tool", cartesian_planner));
    vec.vector.z = -tool_approach_offset;
    tool_retract->setIKFrame(this->get_parameter("link_name").as_string());
    tool_retract->setForwardedProperties({"ik_frame"});
    tool_retract->setDirection(vec);
    tool_retract->properties().configureInitFrom(
        moveit::task_constructor::Stage::PARENT);
[... 236 lines cut ...]
rail_box.dimensions[0] = 1.355;
rail_box.dimensions[1] = 0.640;
rail_box.dimensions[2] = 0.01;

boundaries.primitives.push_back(rail_box);
boundaries.primitive_poses.push_back(rail_box_pose);

planning_scene.world.collision_objects.push_back(boundaries);

planning_scene_diff_publisher->publish(planning_scene);
```

ComputeIK fails every time, regardless of where the collision object is, even if it's nowhere near the end effector or the arm and no IK plan could possibly collide with it. Here's the relevant logs:

```
[robot_arm_core-17]   0  - ←   0 →   -  0 / tool change
[robot_arm_core-17]     1  - ←   1 →   -  1 / current state
[robot_arm_core-17]     -  1 →   0 ←   0  - / move to tool
[robot_arm_core-17]     0  - ←   0 ←   0  - / approach tool
[robot_arm_core-17]     0  - ←   0 →   -  0 / tool location IK
[robot_arm_core-17]       1  - ←   1 →   -  1 / tool location
[robot_arm_core-17] Failing stage(s):
[robot_arm_core-17] tool location IK (0/1)
[robot_arm_core-17] [ERROR] [1738260711.215896384] [robot_arm_core]: Task constructor plan failed: PLANNING_FAILED
```

rhaschke, 2025-01-30:
Probably this is related to your setup. Is there any error message from `ComputeIK` which could explain the failure? Did you look at the failure comment reported in the rviz GUI?

ralphieraccoon, 2025-01-31:
No other error message on the log as far as I can see, no comments in the rviz GUI (screenshot below): 

[image]

Here are the robot arm URDF and SRDF if they are of any help:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<!--This does not replace URDF, and is not an extension of URDF.
    This is a format for representing semantic information about the robot structure.
    A URDF file must exist for this robot as well, where the joints and the links that are referenced are defined
-->
<robot name="fairino3_v6_robot">
    <!--GROUPS: Representation of a set of joints and links. This can be useful for specifying DOF to plan for, defining arms, end effectors, etc-->
    <!--LINKS: When a link is specified, the parent joint of that link (if it exists) is automatically included-->
    <!--JOINTS: When a joint is specified, the child link of that joint (which will always exist) is automatically included-->
    <!--CHAINS: When a chain is specified, all the links along the chain (including endpoints) are included in the group. Additionally, all the joints that are parents to included links are also included. This means that joints along the chain and the parent joint of the base link are included in the group-->
    <!--SUBGROUPS: Groups can also be formed by referencing to already defined group names-->
    <group name="fairino3_v6_group">
        <joint name="j1"/>
        <joint name="j2"/>
        <joint name="j3"/>
        <joint name="j4"/>
        <joint name="j5"/>
        <joint name="j6"/>
        <joint name="eef"/>
        <chain base_link="base_link" tip_link="end_effector"/>
    </group>
    <!--GROUP STATES: Purpose: Define a named state for a particular group, in terms of joint values. This is useful to define states like 'folded arms'-->
    <group_state name="pos1" group="fairino3_v6_group">
        <joint name="j1" value="2.4468"/>
        <joint name="j2" value="-1.6214"/>
        <joint name="j3" value="1.5465"/>
        <joint name="j4" value="-1.5877"/>
        <joint name="j5" value="-1.6368"/>
        <joint name="j6" value="0"/>
    </group_state>
    <group_state name="pos2" group="fairino3_v6_group">
        <joint name="j1" value="0.8606"/>
        <joint name="j2" value="-1.4189"/>
        <joint name="j3" value="1.5465"/>
        <joint name="j4" value="-1.7227"/>
        <joint name="j5" value="-1.5018"/>
        <joint name="j6" value="0"/>
    </group_state>
    <end_effector name="end_effector" parent_link="end_effector" group="fairino3_v6_group" />
    <!--DISABLE COLLISIONS: By default it is assumed that any link of the robot could potentially come into collision with any other link in the robot. This tag disables collision checking between a specified pair of links. -->
    <disable_collisions link1="base_link" link2="shoulder_link" reason="Adjacent"/>
    <disable_collisions link1="base_link" link2="upperarm_link" reason="Never"/>
    <disable_collisions link1="forearm_link" link2="upperarm_link" reason="Adjacent"/>
    <disable_collisions link1="forearm_link" link2="wrist1_link" reason="Adjacent"/>
    <disable_collisions link1="shoulder_link" link2="upperarm_link" reason="Adjacent"/>
    <disable_collisions link1="shoulder_link" link2="wrist1_link" reason="Never"/>
    <disable_collisions link1="wrist1_link" link2="wrist2_link" reason="Adjacent"/>
    <disable_collisions link1="wrist1_link" link2="wrist3_link" reason="Never"/>
    <disable_collisions link1="wrist2_link" link2="wrist3_link" reason="Adjacent"/>
</robot>
```

```xml
<?xml version="1.0" encoding="utf-8"?>
<!-- This URDF was automatically created by SolidWorks to URDF Exporter! Originally created by Stephen Brawner (brawner@gmail.com) 
     Commit Version: 1.5.1-0-g916b5db  Build Version: 1.5.7152.31018
     For more information, please see http://wiki.ros.org/sw_urdf_exporter -->
<robot
  name="fairino3_v6_robot">
  <link
    name="base_link">
    <inertial>
      <origin
[... 437 lines cut ...]
      velocity="3.2" />
    <calibration
      rising="0"
      falling="0" />
    <dynamics
      damping="0"
      friction="0" />
    <safety_controller
      soft_lower_limit="-3.0543"
      soft_upper_limit="3.0543"
      k_position="15"
      k_velocity="10" />
  </joint>
  <link name="end_effector" />
  <joint name="eef" type="fixed">
      <origin
      xyz="0 0 0.164"
      rpy="0 0 0" />
    <parent
      link="wrist3_link" />
    <child
      link="end_effector" />
  </joint>
</robot>
```

ralphieraccoon, 2025-01-31:
Occasionally, for a reason I can't fathom (some kind of race condition?), the first task does execute, but then the second task fails with the following log:

```
[robot_arm_core-17] [ERROR] [1738282299.600385473] [robot_arm_core]: Task constructor plan failed: PLANNING_FAILED
[robot_arm_core-17]   0  - ←   0 →   -  0 / tool change
[robot_arm_core-17]     1  - ←   1 →   -  0 / current state
[robot_arm_core-17]     -  0 →   1 →   -  1 / retract tool
[robot_arm_core-17]     -  1 →   0 ←   1  - / move to end
[robot_arm_core-17]     1  - ←   1 →   -  1 / end state
[robot_arm_core-17] Failing stage(s):

```
Similar screenshot of rviz GUI:

[image]

rhaschke, 2025-01-31:
Please provide a .zip file comprising a fully self-contained colcon or catkin workspace (you didn't even specify the ROS version you are using). With only a few files provided, one cannot attempt reproducing an issue.

ralphieraccoon, 2025-01-31:
My apologies, I'm using ROS Jazzy on Ubuntu 24.04 (Noble Numbat). The `/src` directory of my colcon workspace is over 800MB so I can't upload it here. You can download a copy from [here](https://www.dropbox.com/scl/fi/zy0s4mfycmnohn7r4x43m/src.zip?rlkey=gwrf21cevfra0c3fbz022ahie&st=bwoncbya&dl=0). Is that OK of do you want the `\build`,`\install` and `\log` directories as well?

DaniGarciaLopez, 2025-02-01:
Jumping into the conversation to comment that I recently experienced something similar, also in Jazzy. The task fails without providing any meaningful message about the stage at which it occurs (no stage is marked as failed, and the logging "Failing stage(s):" doesn't specify any message).

I haven't had time to investigate deeply, but I was able to replicate the issue by simply changing the `ik_frame` in the `ComputeIK` stage to one that doesn't exist in the scene. Like @ralphieraccoon, I am also using `trac_ik`, so maybe there is a problem with this kinematics solver.

I don't think this is related to the former issue, but I hope this helps narrow down the problem when `ComputeIK` fails silently.

rhaschke, 2025-02-02:
@ralphieraccoon: The sources are sufficient. However, please narrow down your workspace to the minimum required to reproduce your issue. Also, I need to know, which launch files to start.

@DaniGarciaLopez: Indeed, the `failing stage(s)` error message is empty in that case. However, a warning is issued:
`WARN 11:15:08 ros.moveit_task_constructor_core.ComputeIK: ik frame unknown in robot: 'foo'`

DaniGarciaLopez, 2025-02-02:
> [@DaniGarciaLopez](https://github.com/DaniGarciaLopez): Indeed, the `failing stage(s)` error message is empty in that case. However, a warning is issued:
> `WARN 11:15:08 ros.moveit_task_constructor_core.ComputeIK: ik frame unknown in robot: 'foo'`

That's right, I also get the warning. However, when you have a big task with several `ComputeIK` stages, it is difficult to determine which one failed just by looking at the log. I think it would be much easier to debug if the solution were directly marked as a failure with a meaningful message so you could check in rviz exactly which stage failed.

I just took a look at the code and it should be pretty straightforward to make the changes. I might have time in the following days to prepare a PR if you agree @rhaschke.

rhaschke, 2025-02-02:
@DaniGarciaLopez, thanks for offering help. I already implemented the change quickly: #651.

rhaschke, 2025-02-02:
@ralphieraccoon: Looking at https://github.com/moveit/moveit_task_constructor/issues/650#issuecomment-2626000813, a possible reason for your failure might be incompatible states feeding into both ends of the `Connect` stage. These are silently filtered out. You may enable debug-level logging and see some warnings on the command-line.

ralphieraccoon, 2025-02-02:
> [@ralphieraccoon](https://github.com/ralphieraccoon): The sources are sufficient. However, please narrow down your workspace to the minimum required to reproduce your issue. Also, I need to know, [...]

Hi, I've updated the zip file to only include packages needed. To reproduce the problem run `ros2 launch ampi_project launch_webots.py` in the terminal, and in a new window run `ros2 action send_goal /swap_tool ampi_project_interfaces/action/SwapTool  "{tool: shaft_chuck_2_0mm}"

rhaschke, 2025-02-02:
> Bear in mind `webots` is needed as a dependency, though you could probably comment out parts of the launch file to disable it.

Please do so.

ralphieraccoon, 2025-02-03:
OK, should work without `webots` now.

ralphieraccoon, 2025-02-03:
Thanks for the tip about the debug. I think there are two issues at play here:

- `trac_ik` isn't always finding non-colliding solutions, which causes `ComputeIK` to fail. This causes this debug message: `[introspection]: new solution #3 (tool location IK): inf Collision between 'upperarm_link' and 'boundary'` in my case. This is a problem with `trac_ik` and not the task constructor as @DaniGarciaLopez found out.
- When `trac_ik` does find a non-colliding solution, after the task is cleared for the retract operation this debug message occurs: `[Connecting]: move to end: different number of collision objects`. I wonder if I need to allow collisions with a `ModifyPlanningScene` stage, but the tutorial files suggested that wasn't necessary.

ralphieraccoon, 2025-02-03:
I think I've fixed the problem, it seems to be an issue with the `end_state` stage, which is a `FixedState` that captures the `CurrentState` at the start, to allow the robot to return to the position it started from. When creating the stage a `PlanningScene` is created from `task->getRobotModel()` which **doesn't** include the collision objects:

```cpp
std::unique_ptr<moveit::task_constructor::stages::FixedState> end_state(
    new moveit::task_constructor::stages::FixedState("end state"));

planning_scene::PlanningScenePtr current_plan(
    new planning_scene::PlanningScene(task->getRobotModel()));
current_plan->setCurrentState(solution.start_scene.robot_state);

end_state->setState(current_plan);
```

The solution in my case was to add the `world` argument to the constructor of `PlanningScene` from a `PlanningSceneMonitor` I was already using for something else:

```cpp
planning_scene_monitor::LockedPlanningSceneRW ls(servo_planning_scene_monitor_);
planning_scene::PlanningScenePtr current_plan(
    new planning_scene::PlanningScene(task->getRobotModel(), ls->getWorldNonConst()));
current_plan->setCurrentState(solution.start_scene.robot_state);
```

This doesn't feel like the best way to do things for me, perhaps there is a more elegant solution. It would be nice if there was a function to create a `FixedState` that is a "capture" of a `CurrentState` so the robot can return to this saved state later on. Perhaps as an additional constructor. I'll look into it when I have time. Thanks @rhaschke and @DaniGarciaLopez for your help, I know it was a struggle at times!

closed 2025-02-03 by ralphieraccoon

rhaschke, 2025-02-04:
You could explicitly fetch the current scene using the PlanningSceneInterface (or the corresponding getPlanningScene service).