moveit/moveit_task_constructor #654: is it mandatory to use rclpp Node?

adamanov, 2025-02-10:
Hello, first of all, thank you for an amazing framework.
 
We usually use a Python node initializing it 
```
from rclpy.node import Node
rclpy.init()
node= Node("my_node")
```

however, in all examples or similar application where MTC was used with python bindings, you use

```
import rclcpp
rclcpp.init()
node = rclcpp.Node("mtc_tutorial")
```
where I am using a **rclpy version Node** then I am not able to use some critical functions (sofar which I needed) from MTC such as `loadRobotModel`, `core.PipelinePlanner` since they are expecting a rclcpp Node, 
```
task = core.Task()
task.name = "pick + place"
task.loadRobotModel(node)

# Connect the two stages
pipeline = core.PipelinePlanner(node, pipeline="ompl", planner_id="RRTConnectkConfigDefault")
planners = [(ARM_GROUP, pipeline)]

move_to_pick = stages.Connect("move_to_pick", planners)
```
```
[manipulation_node-1] TypeError: loadRobotModel(): incompatible function arguments. The following argument types are supported:
[manipulation_node-1]     1. (self: pymoveit_mtc.core.Task, node: rclcpp::Node, robot_description: str = 'robot_description') -> None
```
```
[manipulation_node-1]     pipeline = core.PipelinePlanner()
[manipulation_node-1] TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
[manipulation_node-1]     1. pymoveit_mtc.core.PipelinePlanner(node: rclcpp::Node, pipeline: str = 'ompl', planner_id: str = '')
```

Is there a way to go around it? and use still a Python Node i.e. `rclpy.init()` ?

rhaschke, 2025-02-11:
You can just use both, the rclpy and rclcpp Node.

closed 2025-02-11 by rhaschke

adamanov, 2025-02-11:
> You can just use both, the rclpy and rclcpp Node. And, yes, MTC requires an rclcpp Node as the python API wraps C++ code.

Hello

I have been using `rclpy.node` for most MTC functions without issues. However, I encounter compatibility errors when a function explicitly requires node: rclcpp::Node, such as with `PipelinePlanner` or `core.Task().loadRobotModel`.

I couldn't find any references or methods to convert or adapt rclpy.node for use as rclcpp::Node arguments in these cases.

we want to explicitly use `rclpy.node` overall, how can it be done?

rhaschke, 2025-02-11:
> > You can just use both, the rclpy and rclcpp Node.
> 
> do you mean in the same python script?

Exactly.
`rclpy.Node` and `rclcpp.Node` are *not* interchangeable and cannot be converted into each other - unfortunately.