moveit/moveit_task_constructor #656: Iron branch fails to build

MarqRazz, 2025-02-12:
@rhaschke I see that you pushed several changes to the Iron branch the other day but it now fails to build. Are other changes coming to MoveIt2 to support this?

```
--- stderr: moveit_task_constructor_core                                                         
In file included from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/merge.h:39,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/src/merge.cpp:38:
/root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/storage.h:42:10: fatal error: moveit/macros/class_forward.hpp: No such file or directory
   42 | #include <moveit/macros/class_forward.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/cost_terms.h:41,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/src/cost_terms.cpp:37:
/root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/storage.h:42:10: fatal error: moveit/macros/class_forward.hpp: No such file or directory
   42 | #include <moveit/macros/class_forward.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:132: src/CMakeFiles/moveit_task_constructor_core.dir/merge.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:90: src/CMakeFiles/moveit_task_constructor_core.dir/cost_terms.cpp.o] Error 1
In file included from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage.h:43,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/container.h:41,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/container_p.h:41,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/src/task.cpp:38:
/root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/utils.h:47:10: fatal error: moveit/macros/class_forward.hpp: No such file or directory
   47 | #include <moveit/macros/class_forward.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:188: src/CMakeFiles/moveit_task_constructor_core.dir/task.cpp.o] Error 1
In file included from /root/2_apg_ws/src/moveit_task_constructor/core/src/marker_tools.cpp:1:
/root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/marker_tools.h:5:10: fatal error: moveit/macros/class_forward.hpp: No such file or directory
    5 | #include <moveit/macros/class_forward.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:118: src/CMakeFiles/moveit_task_constructor_core.dir/marker_tools.cpp.o] Error 1
In file included from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage.h:43,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:41,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/src/stage.cpp:38:
/root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/utils.h:47:10: fatal error: moveit/macros/class_forward.hpp: No such file or directory
   47 | #include <moveit/macros/class_forward.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:160: src/CMakeFiles/moveit_task_constructor_core.dir/stage.cpp.o] Error 1
In file included from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage.h:43,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:41,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/src/storage.cpp:38:
/root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/utils.h:47:10: fatal error: moveit/macros/class_forward.hpp: No such file or directory
   47 | #include <moveit/macros/class_forward.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:174: src/CMakeFiles/moveit_task_constructor_core.dir/storage.cpp.o] Error 1
In file included from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage.h:43,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:41,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/src/introspection.cpp:37:
/root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/utils.h:47:10: fatal error: moveit/macros/class_forward.hpp: No such file or directory
   47 | #include <moveit/macros/class_forward.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/stage.h:43,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/container.h:41,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/container_p.h:41,
                 from /root/2_apg_ws/src/moveit_task_constructor/core/src/container.cpp:37:
/root/2_apg_ws/src/moveit_task_constructor/core/include/moveit/task_constructor/utils.h:47:10: fatal error: moveit/macros/class_forward.hpp: No such file or directory
   47 | #include <moveit/macros/class_forward.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:76: src/CMakeFiles/moveit_task_constructor_core.dir/container.cpp.o] Error 1
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:104: src/CMakeFiles/moveit_task_constructor_core.dir/introspection.cpp.o] Error 1
/root/2_apg_ws/src/moveit_task_constructor/core/src/utils.cpp:44:10: fatal error: moveit/robot_state/robot_state.hpp: No such file or directory
   44 | #include <moveit/robot_state/robot_state.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:202: src/CMakeFiles/moveit_task_constructor_core.dir/utils.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:362: src/CMakeFiles/moveit_task_constructor_core.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_task_constructor_core [0.67s, exited with code 2]
```

rhaschke, 2025-02-13:
I'm surprised that this fails now when it worked before. When I update ros2 branch, I usually just rebase the few reversion commits needed for humble, jazzy, and iron.
Unfortunately, due do https://github.com/moveit/moveit2/issues/3333, I don't see build issues on Iron yet.
Could you please validate whether https://github.com/ubi-agni/moveit_task_constructor/tree/iron works for you?

rhaschke, 2025-02-13:
Ok. I pushed a revert commit for that one too...

closed 2025-02-13 by rhaschke